seemoo-lab/nexmon #235: Nexus 6 internal wireless card

Nexmonissues, 2018-06-23:
I have tried many different builds, originally bought this phone to run nethunter on however I couldn't get an external wireless card working on it (TL-WN722N) 

On the page for Nexmon it says 

bcm4356 | 7_35_101_5_sta | Nexus 6 | Android 7.1.2

The only build I could find for Android 7.1.2 was PureNexus - I popped that on my phone using TWRP, ran 
sudo apt-get install git gawk qpdf adb flex bison
sudo dpkg --add-architecture i386
sudo apt-get update
sudo apt-get install libc6:i386 libncurses5:i386 libstdc++6:i386
git clone https://github.com/seemoo-lab/nexmon.git
cd nexmon
source setup_env.sh
make
cd patches/bcm4356/7_35_101_5_sta/nexmon/
make
make backup-firmware
make install-firmware 

It all seemed to run fine, I then downloaded the APK for nexmon - installed it, Granted root for the app, installed the tools required - headed to firmware, then tried to install the firmware it asked for. 

I am completely lost at this point, I have tried different builds/kernels and still nothing after around 100 hours. I can't find anything online featuring how to set this up.

Nexmonissues, 2018-06-23:
I am happy to do a full writeup of the process with screenshots and all once I have figured it out, I know a lot of people will be after this.

JuniorJPDJ, 2018-06-24:
You already installed firmware, installation in app was pointless

Nexmonissues, 2018-06-24:
How am I to install and get the app to run if I don't install the firmware? The app wont allow me to do anything till I have done that.