seemoo-lab/nexmon #237: airmon-ng/airodump-ng doesn't work in Rpi-3b:  brcmfmac: nexmon _n1_ioct1_handler  Errrors

alimp5, 2018-07-02:
root@Hero: airmon-ng start wlan0
```
PHY	Interface	Driver		Chipset
phy0	wlan0		brcmfmac	Broadcom 43430
           .................. see attachment (errors skipped here) ..................
	Interface wlan0 is already in monitor mode
```
https://imgur.com/a/6uCHAsQ

root@Hero: airodump-ng wlan0
or
root@Hero: airodump-ng wlan0mon
```
ARP linktype is set to 1 (Ethernet) - expected ARPHRD_IEEE80211,
ARPHRD_IEEE80211_FULL or ARPHRD_IEEE80211_PRISM instead.  Make
sure RFMON is enabled: run 'airmon-ng start wlan0 <#>'
Sysfs injection support was not found either.

Failed initializing wireless card(s): wlan0
```

root@Hero: uname -a
Linux Hero 4.9.59-v7_Re4son-Kali-Pi+ #1 SMP Wed Feb 14 20:50:28 CST 2018 armv7l GNU/Linux
aircrack-ng is already the newest version (1:1.2.0-6).

steev, 2018-07-04:
You probably want to use aircrack-ng from git instead of the packaged version.  It has better handling of nexmon.

aircrack-ng, 2018-07-04:
Regarding airmon-ng, as @steev said, using the one from the repo should fix it. If it doesn't, then please open an issue in aircrack-ng repository.

About the second issue, check out aircrack-ng/aircrack-ng#1882

3x-M4ch1nA, 2018-07-08:
Re4son (the main contributor of Kali ARM), said that nexmon no longer supports aircrack-ng. 
When running airmon-ng start wlan0, I got the warning "monitor mode not enabled, this is not a raspberry pi 3" (running on rpi3).
The solution I was told, was that to put into monitor, the new command for nexmon is wlan0up.havent test ed airodump after after running command yet.
But not supporting aircrack is a real hassle for so many repos that start your monitor mode using it (fluxion).

3x-M4ch1nA, 2018-07-08:
@Steve you are correct. however I didn't post the issue, I was responding to the op with a solution for his exact distro (Re4son's Kali-Pi)

aircrack-ng, 2018-07-16:
It's working just fine without it on 4.14.50-kali-v7+ without any hacks so this ticket can be closed. See aircrack-ng/aircrack-ng#1882 for more details