seemoo-lab/nexmon #238: Can't compile in Rpi3b 4.14.54-v7+

atek23, 2018-07-13:
Sorry if it was aswered, I can't find it.
```
root@raspberrypi:/home/pi/Tools/nexmon/patches/bcm43430a1/7_45_41_46/nexmon# make

COLLECTING STATISTICS read /home/pi/Tools/nexmon/STATISTICS.md for more information
  COMPILING src/version.c => obj/version.o (details: log/compiler.log)
  GENERATING LINKER FILE gen/nexmon.pre => gen/nexmon.ld
  GENERATING LINKER FILE gen/nexmon.pre => gen/flashpatches.ld
  LINKING OBJECTS => gen/patch.elf (details: log/linker.log, log/linker.err)
  GENERATING MAKE FILE gen/nexmon.pre => gen/nexmon.mk
  GENERATING MAKE FILE gen/nexmon.pre => gen/flashpatches.mk
  APPLYING FLASHPATCHES gen/flashpatches.mk => brcmfmac43430-sdio.bin (details: log/flashpatches.log)
  APPLYING PATCHES gen/nexmon.mk => brcmfmac43430-sdio.bin (details: log/patches.log)
  BUILDING DRIVER for kernel 4.14 brcmfmac_4.14.y-nexmon/brcmfmac.ko (details: log/driver.log)
make[1]: *** /lib/modules/4.14.54-v7+/build: No such file or directory.  Stop.
Makefile:45: recipe for target 'brcmfmac.ko' failed
make: *** [brcmfmac.ko] Error 2
```

coderbaka, 2018-07-13:
Hi atek23,
I had the same problem with you. Because raspi-kernel package has been updated,so the old modules has been deleted,but "uname -r" still shows the old version. Just reboot and source set-env.sh,then you can compile. Also,I discover if you don't kill wpa_supplicant progress,you won't be able to scan WiFi use   
 monitor mode.

cbeytas, 2018-11-05:
I wasted 3 hours on this same problem. Maybe add a blurb to readme about rebooting?