seemoo-lab/nexmon #240: Can't restore firmware 6p

S1CAR1US, 2018-07-20:
The documentation on using the app is scarce.  I read that using the app is the easy way to get monitor mode working with a 6p.  I install the app, follow the prompts to install the tools.  Then firmware, make a backup, install nexmon firmware,  WiFi stops working and can no longer turn it on at all.  Go to firmware section in the nexmon app to restore firmware using the backup I made, and now it tells me that the device I have is not supported... I have a 6p and just a minute ago it was supported.  I have had to re flash Nexus image and nethunter twice now because of this.  How do I fix without starting over every time?

matthiasseemoo, 2018-07-20:
Manually make a backup of the firmware file and restore it when something
fails. The app can now be considered depricated as we did not update it for
quite a while. If you have a firmware version in your android rom that is
supported by nexmon, compile your own firmware from the repository and
install it.

-- 
Matthias Schulz
Secure Mobile Networking Lab - SEEMOO

Email: matthias.schulz@seemoo.tu-darmstadt.de
Web: http://www.seemoo.de/mschulz
Phone (new): +49 6151 16-25478    Fax: +49 6151 16-25471

Department of Computer Science
Center for Advanced Security Research Darmstadt
Technische Universität Darmstadt
Mornewegstr. 32 (Office 4.2.10, Building S4/14)
D-64293 Darmstadt, Germany

S1CAR1US, 2018-07-20:
That's a pretty cheap cop-out, the info on your own page refers the app.  Now your telling me as if I should have known.  At least offer some direction as to how to go about restoring the correct fw

matthiasseemoo, 2018-07-20:
This should be the location of the firmware file:
/vendor/firmware/fw_bcmdhd.bin
Where exactly do we refer to the app on the webpage?

In any case, you use nexmon on your own risk:
Our software may damage your hardware and may void your hardware’s
warranty! You use our tools at your own risk and responsibility! If you
don't like these terms, don't use nexmon!

-- 
Matthias Schulz
Secure Mobile Networking Lab - SEEMOO

Email: matthias.schulz@seemoo.tu-darmstadt.de
Web: http://www.seemoo.de/mschulz
Phone (new): +49 6151 16-25478    Fax: +49 6151 16-25471

Department of Computer Science
Center for Advanced Security Research Darmstadt
Technische Universität Darmstadt
Mornewegstr. 32 (Office 4.2.10, Building S4/14)
D-64293 Darmstadt, Germany

LarsLundkvist, 2019-01-19:
Hi,

I am sorry to bother with trivials, but i am a total beginner in the linux-environment. Could you please give me som directions on how to make use of my backup-file. I have the file backup file brcmfmac.ki.orig. But how shall I use it to actually restore the firmware?

Again - sorry to bother. This was probably not a good project for a beginner :-)

I am running a Raspberry Pi3b+ with Raspian