seemoo-lab/nexmon #241: Adding Monitor Mode Interface (Raspberry Pi 3 B+)

tdwoods, 2018-07-23:
Following the readme, I run all of the commands as root but when I try to execute the command: 

> iw phy `iw dev wlan0 info | gawk '/wiphy/ {printf "phy" $2}'` interface add mon0 type monitor

 it gives me the error: `command Failed: Operation not supported (-95)`

aiokswhatever, 2018-11-13:
You got to install firmware each time after reboot before doing this

Install the patched firmware on your RPI3: make install-firmware

MaximusBaton, 2018-11-15:
> You got to install firmware each time after reboot before doing this
> 
> `Install the patched firmware on your RPI3: make install-firmware`

The manual describes how to make the patch working after reboot

spadejack014, 2019-07-02:
I met this problemt when I try to execute the command, too.
The only different thing is I run all commmands on my raspberry 3B.
How did you deal with that?