seemoo-lab/nexmon #242: Rpi3 build of libisl.so.10 needs libgmp3-dev on Raspbian Stretch

daver-gs, 2018-07-24:
Building nexmon for the RPi3 under Raspbian, I didn't have `libisl.so.10`.  Attempting to build showed a dependency on `gmp.h`, found in libgmp3-dev.  Installing that, then a `make clean,`rerun of `./configure` and another `make` got a successful build.

Maybe something about this could be mentioned in the blurb?

**$ cd buildtools/isl-0.10
$ ./configure**

```
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking for style of include used by make... GNU
checking dependency style of gcc... gcc3
checking for g++... g++
checking whether we are using the GNU C++ compiler... yes
checking whether g++ accepts -g... yes
checking dependency style of g++... gcc3
checking for C compiler vendor... gnu
checking build system type... armv7l-unknown-linux-gnueabihf
checking host system type... armv7l-unknown-linux-gnueabihf
checking whether C compiler accepts -malign-double... no
checking whether C compiler accepts -fstrict-aliasing... yes
checking whether C compiler accepts -ffast-math... yes
checking for gcc architecture flag... 
checking for gcc architecture flag... unknown
checking whether C compiler accepts -O3 -fomit-frame-pointer -fstrict-aliasing -ffast-math... yes
checking whether the compiler supports function __attribute__((__warn_unused_result__))... yes
checking for __attribute__... yes
checking how to print strings... printf
checking for a sed that does not truncate output... /bin/sed
checking for grep that handles long lines and -e... /bin/grep
checking for egrep... /bin/grep -E
checking for fgrep... /bin/grep -F
checking for ld used by gcc... /usr/bin/ld
checking if the linker (/usr/bin/ld) is GNU ld... yes
checking for BSD- or MS-compatible name lister (nm)... /usr/bin/nm -B
checking the name lister (/usr/bin/nm -B) interface... BSD nm
checking whether ln -s works... yes
checking the maximum length of command line arguments... 1572864
checking whether the shell understands some XSI constructs... yes
checking whether the shell understands "+="... yes
checking how to convert armv7l-unknown-linux-gnueabihf file names to armv7l-unknown-linux-gnueabihf format... func_convert_file_noop
checking how to convert armv7l-unknown-linux-gnueabihf file names to toolchain format... func_convert_file_noop
checking for /usr/bin/ld option to reload object files... -r
checking for objdump... objdump
checking how to recognize dependent libraries... pass_all
checking for dlltool... no
checking how to associate runtime and link libraries... printf %s\n
checking for ar... ar
checking for archiver @FILE support... @
checking for strip... strip
checking for ranlib... ranlib
checking command to parse /usr/bin/nm -B output from gcc object... ok
checking for sysroot... no
checking for mt... mt
checking if mt is a manifest tool... no
checking how to run the C preprocessor... gcc -E
checking for ANSI C header files... yes
checking for sys/types.h... yes
[... 564 lines cut ...]
  CC     isl_test-isl_test.o
  CCLD   isl_test
  CC     isl_polyhedron_sample-polyhedron_sample.o
  CCLD   isl_polyhedron_sample
  CC     isl_pip-pip.o
  CCLD   isl_pip
  CC     isl_polyhedron_minimize-polyhedron_minimize.o
  CCLD   isl_polyhedron_minimize
  CC     isl_polytope_scan-polytope_scan.o
  CCLD   isl_polytope_scan
  CC     isl_polyhedron_detect_equalities-polyhedron_detect_equalities.o
  CCLD   isl_polyhedron_detect_equalities
  CC     isl_cat-cat.o
  CCLD   isl_cat
  CC     isl_closure-closure.o
  CCLD   isl_closure
  CC     isl_bound-bound.o
  CCLD   isl_bound
make[2]: Leaving directory '/home/pi/src/nexmon/buildtools/isl-0.10'
Making all in doc
make[2]: Entering directory '/home/pi/src/nexmon/buildtools/isl-0.10/doc'
make[2]: Nothing to be done for 'all'.
make[2]: Leaving directory '/home/pi/src/nexmon/buildtools/isl-0.10/doc'
make[1]: Leaving directory '/home/pi/src/nexmon/buildtools/isl-0.10'
```

matthiasseemoo, 2018-07-24:
Just read the readme carefully:

   - Install the kernel headers to build the driver and some dependencies: sudo
   apt install raspberrypi-kernel-headers git *libgmp3-dev* gawk qpdf bison
   flex make

-- 
Matthias Schulz
Secure Mobile Networking Lab - SEEMOO

Email: matthias.schulz@seemoo.tu-darmstadt.de
Web: http://www.seemoo.de/mschulz
Phone (new): +49 6151 16-25478    Fax: +49 6151 16-25471

Department of Computer Science
Center for Advanced Security Research Darmstadt
Technische Universität Darmstadt
Mornewegstr. 32 (Office 4.2.10, Building S4/14)
D-64293 Darmstadt, Germany

daver-gs, 2018-07-24:
Oops! Donkey's ears for me!

Thanks.

closed 2018-07-24 by daver-gs

MaximusBaton, 2018-11-10:
For the pi's clean debian stretch also needed:
```
apt-get install autoconf automake
```