seemoo-lab/nexmon #247: Samsung Galaxy S8 ROM Extraction

tech5drive, 2018-09-05:
I am trying to get nexmon to work on Samsung Galaxy S8 and have trouble with the ROM extraction.

Initially I used the original ROM rooted and tried to do `dhdutil membytes 0x0 128` as root which gave me
`dhdutil: __dhd_driver_io: Permission denied`

so I figured out it was SELinux. Unfortunately I could not set it to permissive (according to https://android.stackexchange.com/questions/142909/how-to-set-selinux-to-permissive) as it was compiled with config_always_enforce=true

Then I flashed Lineage 14.1 and set SELinux to permissive. This time running `dhdutil membytes 0x0 128` gave me
`dhdutil: __dhd_driver_io: Operation not supported on transport endpoint`

Running some dhdutil commands worked fine while running others failed as you can see below.

```
dreamlte:/ # dhdutil ramstart                                              
1507328 (0x170000)

dreamlte:/ # dhdutil ramsize                                                   
1376256 (0x150000)

dhdutil bcmerrorstr                                               
Unsupported

dreamlte:/ # dhdutil wdtick
0

dreamlte:/ # dhdutil ioctl_timeout                                             
4000 (0xfa0)

dreamlte:/ # dhdutil upload /sdcard/test.txt                               
dhdutil: __dhd_driver_io: Operation not supported on transport endpoint
```

I also tried to use flashpatch (similar to bcm4358) however I kept getting the same error when dhdutil was used.

```
# make dump-rom

          ###########   ###########   ##########    ##########           
         ############  ############  ############  ############          
         ##            ##            ##   ##   ##  ##        ##          
         ##            ##            ##   ##   ##  ##        ##          
         ###########   ####  ######  ##   ##   ##  ##    ######          
          ###########  ####  #       ##   ##   ##  ##    #    #          
                   ##  ##    ######  ##   ##   ##  ##    #    #          
                   ##  ##    #       ##   ##   ##  ##    #    #          
         ############  ##### ######  ##   ##   ##  ##### ######          
         ###########    ###########  ##   ##   ##   ##########           

            S E C U R E   M O B I L E   N E T W O R K I N G               

                               presents:                                  

              # ###   ###  #   # # ###  ###   ###  # ###                  
              ##   # #   #  # #  ##   ##   # #   # ##   #                 
              #    # #####   #   #    #    # #   # #    #                 
              #    # #      # #  #    #    # #   # #    #                 
              #    #  #### #   # #    #    #  ###  #    #                 

                The C-based Firmware Patching Framework                   

                           !!! WARNING !!!                                
    Our software may damage your hardware and may void your hardware’s    
     warranty! You use our tools at your own risk and responsibility      

  COLLECTING STATISTICS read /home/alex/github/nexmon/STATISTICS.md for more information
  COMPILING src/version.c => obj/version.o (details: log/compiler.log)
  PREPARING gen/nexmon.pre => gen/nexmon2.pre
  GENERATING LINKER FILE gen/nexmon.pre => gen/nexmon.ld
  GENERATING LINKER FILE gen/nexmon.pre => gen/flashpatches.ld
  LINKING OBJECTS => gen/patch.elf (details: log/linker.log, log/linker.err)
  GENERATING MAKE FILE gen/nexmon.pre => gen/nexmon.mk
/home/alex/github/nexmon/buildtools/gcc-arm-none-eabi-5_4-2016q2-linux-x86/bin/arm-none-eabi-objdump: section '.text.patch' mentioned in a -j option, but not found in any input file
/home/alex/github/nexmon/buildtools/gcc-arm-none-eabi-5_4-2016q2-linux-x86/bin/arm-none-eabi-objdump: section '.text.ucode' mentioned in a -j option, but not found in any input file
  GENERATING MAKE FILE gen/nexmon.pre => gen/flashpatches.mk
  APPLYING FLASHPATCHES gen/flashpatches.mk => fw_bcmdhd.bin (details: log/flashpatches.log)
  APPLYING PATCHES gen/nexmon.mk => fw_bcmdhd.bin (details: log/patches.log)
  COPYING TO PHONE fw_bcmdhd.bin => /sdcard/fw_bcmdhd.bin
  LOADING FIRMWARE /sdcard/fw_bcmdhd.bin
dhdutil: __dhd_driver_io: Operation not supported on transport endpoint
  DUMPING ROM ROM => /sdcard/rom.bin
dhdutil: __dhd_driver_io: Operation not supported on transport endpoint
  PULLING /sdcard/rom.bin => rom.bin
  RELOADING PREVIOUS FIRMWARE
  COPYING ROM rom.bin => /home/alex/github/nexmon/firmwares/bcm4358/rom.bin
  ```

What causes this issue and how can I bypass it?

LoloBee, 2018-12-11:
Hi! I'm also working on S8 and I faced the same issue.

After checking this [paper](https://pdfs.semanticscholar.org/078e/bf9f4c7916e59db4711e726ee94e2e802d54.pdf) you can read that

> The firmware that resides in ROM can simply  be  extracted  using  the  dhdutil that  is  distributed  as part of the Android platform and allows to send specific ioctls to the BCMDHD driver in case the driver was compiled with active DHD_DEBUG flag.

I can't find the flag DHD_DEBUG, so, the more look-a-like ones are
`CONFIG_B43_DEBUG=y
CONFIG_BRCMDBG=y`
So, I'm building a kernel for S8 with these flags and I'll try to dump ROM.

Another symptom which leads me to think about debug flags is that nexutil can't set adapter in monitor mode, which Matthias says that is available on all Broadcom chipsets.

`nexutil -m1                                                                                                                                                                                                                                                      
__nex_driver_io: error`

Did you make any progress?

LoloBee, 2018-12-11:
Ok DHD_DEBUG flag is set in Makefile by hand, but after building, no successwith dhdutil neither nexutil.

matthiasseemoo, 2018-12-11:
Did you check how the wifi interface is called? If it is not called wlan0,
you need to define it with the -I argument when calling nexutil.
Additionally, you might not be able to directly access the ROM contents
anymore on newer chips, hence, I normally extract clean ROM images (without
flashpatches) by using ioctls. Can you check if you get a meaningful
result, when you call nexutil with argument -g0? The result shall contain
the two letters wl.

Patching this firmware will be also more complicated as I suppose that it
contains four different ucodes: with and without mu mimo for both the d11
and the d11x core. In the second half of december I might find some time to
provide a rom extraction patch for the galaxy s8 that you could test.

LoloBee, 2018-12-11:
Hi Matthias, great to read you again.

The result is meaningful or I think so

`dreamlte:/ # nexutil -g0`
`0x000000: 77 6c e4 14                                     wl..`

I'll take a look on how to extract clean roms using ioctls :)

Btw, thanks for looking for time for S8!

matthiasseemoo, 2018-12-12:
It is quite complex as you need to integrate ucode compression first before
you can extend the ioctl handler with a custom ROM extraction command.

LoloBee, 2018-12-12:
UCODE compression is used when patching firmware, I can extract code used to do this, but I don't think got enough luck.

Anyway, if you need SSH access to an S8, I can provide it.

Kind regards!

tech5drive, 2018-12-19:
Hello from me again, 

I'm happy there are finally more people working with this device!

@matthiasseemoo 
> Additionally, you might not be able to directly access the ROM contents anymore on newer chips, hence, I normally extract clean ROM images (without flashpatches) by using ioctls. 

what do you mean that you might not be able to access the ROM contents on newer chips? There will always be a need to push firmware updates so I guess there needs to be some communication going on.

Also how do you go about extracting them using ioctls? Are ioctls not just used to communicated with the driver? If the chip is locked down how would that work?

@LoloBee 
I thought I had to patch the kernel. Any progress from your side?

matthiasseemoo, 2018-12-19:
The ROM cannot be changed, as it is a read only memory. You can only apply
temporary patches that we call flashpatches, which are used for patching.
The driver loads one part of the firmware into the RAM and starts execution
from there, but some functions are still called from ROM, hence, it is
important to extract it for analysis. Ioctls are used to communicate with
the firmware, hence, the simplest way, in my opinion, is dumping the ROM
using a custom ioctl.

If you want to know more about how the firmware works, read my thesis.

-- 
Matthias Schulz
Secure Mobile Networking Lab - SEEMOO

Email: matthias.schulz@seemoo.tu-darmstadt.de
Web: http://www.seemoo.de/mschulz
Phone (new): +49 6151 16-25478    Fax: +49 6151 16-25471

Department of Computer Science
Center for Advanced Security Research Darmstadt
Technische Universität Darmstadt
Mornewegstr. 32 (Office 4.2.10, Building S4/14)
D-64293 Darmstadt, Germany

LoloBee, 2018-12-20:
Hi @tech5drive 
From my side, bcm4361 driver seems to have DHD_DEBUG enabled in MakeFile. I'm using [TGPKernel-s8](https://github.com/TheGalaxyProject/tgpkernel-s8-o) kernel, because I can build original Samsung S8 kernel, but Oreo's boot.img is a pain in the a**, so TGPKernel resolves this problem with a really nice batch packaging script.
As I said, TGPKernel, in _drivers/net/wireless/bcm4361/Makefile_ @line 1641
```
EXTRA_CFLAGS += $(DHDCFLAGS) -DDHD_DEBUG
```
So, it's enabled.

I think that the real problems comes from dhdutil, which cannot interface properly with the driver because some commands works and others dont't.
As @matthiasseemoo told me, running
```
dreamlte:/ # nexutil -g0
0x000000: 77 6c e4 14 wl..
```
calls driver's ioctls and will return driver's magic number, defined in _drivers/net/wireless/bcm4361/include/wlioctl_defs.h_ @line 542

```
/* common ioctl definitions */
#define WLC_GET_MAGIC                           0
```

But if I call some ioctls defined in this file, for example
```
#define WLC_GET_CHANNEL                         29
```
won't return real channel, as _nexutil -k_ returns
```
130|dreamlte:/ # nexutil -k
chanspec: 0xe02a, 36/80
dreamlte:/ # nexutil -g29
0x000000: 24 00 00 00                                     $... 
```
Trying another ioctl, for example
```
#define WLC_GET_SSID                            25
```
Should return current SSID, but nexutils starts to report errors, as dhdutil does:
```
dreamlte:/ # nexutil -g25                                                                                                                                                                      
__nex_driver_io: error
0x000000: 00 00 00 00                                     ....
```
One of the main functions to get monitor mode working on J7 was _consoledump_ parameter, but, as other dhdutil's parameters is not working:
```
dreamlte:/ # dhdutil consoledump
dhdutil: __dhd_driver_io: Operation not supported on transport endpoint
```
Some parameters works, and others don't:
```
95|dreamlte:/ # dhdutil version                                                                                                                                                                
(null): 1.88 RC5.0
Dongle Host Driver, version 1.77.72.1 (r)
Compiled in drivers/net/wireless/bcmdhd4361 on Dec 13 2018 at 16:14:32
Bus API revisions:(FW rev7)(DHD rev7)
```
```
dreamlte:/ # dhdutil -V
get not defined for var
```
So, I think that these ioctls are not working properly, maybe limited by driver, firmware or rom. If _consoledump_ does not work, it will be impossible to debug patches and if ioctls are not handled properly, it also be hard to copy a clean ROM to RAM and dump it.

As a side note, this chip uses NVRAM (J7 did not)
```
dhd_bus_download_firmware: firmware path=/vendor/etc/wifi/bcmdhd_sta.bin_b0, nvram path=/vendor/etc/wifi/nvram.txt_r02j_b0
<4>[177311.869414]  [3:wifi@1.0-servic: 3384] dhdpcie_download_code_file: download firmware /vendor/etc/wifi/bcmdhd_sta.bin_b0
```

matthiasseemoo, 2018-12-20:
To be exact, nexutil tunnels ioctls through the driver to the firmware,
where those ioctls are answered. As long as some ioctls work, we can add
custon ioctl handlers. Regarding consoledump, on some chips the console
output is visible in the kernel log, so try calling dmesg. Otherwise, we
can always defibe an ioctl to return the console content. I still need to
finish another project, but after that I coulf take a look at the bcm4361
firmware.

matthiasseemoo, 2018-12-27:
I now reverse engineered the RAM firmware and created the necessary files to build a firmware patch that activates flashpatches and ucode compression. As the BCM4361b0 is a bit different than the other chips, we first need to check whether the created firmware file is operational. Please pull the nexmon repo and build the rom_extraction patch from [1]. It currently does not extract anything, it just checks, whether the firmware runs or crashes on load. Then, copy the patched bcmdhd_sta.bin_b0 to your Galaxy S8 smartphone. First, make a backup of the existing firmware file and check whether the existing firmware has version number 13.38.55.1 (e.g., by running strings on the firmware file). Then overwrite the /vendor/etc/wifi/bcmdhd_sta.bin_b0 file to activate the patched nexmon firmware and tell me whether it runs. Regular Wi-Fi operation shall be possible. You can also try nexutil -g0, the expected output shall start with wl. If it does not run, please send me the dmesg output. And, please, also send me the output of nexutil -V and send me the output so that I can update our REVINFO.md file.

Keep in mind, that this is only a preliminary testing version. It will probably not work.

[1] https://github.com/seemoo-lab/nexmon/tree/master/patches/bcm4361b0/13_38_55_1_sta/rom_extraction

LoloBee, 2018-12-28:
Great news Matthias.
I'll test it this afternoon and I'll report results.

Thx!

LoloBee, 2018-12-28:
Hi Matthias!
I got exactly the same firmware version
```
4361b0-roml/config_pcie_release_ipa Version: 13.38.55.1 (B0 Network/rsdb) CRC: 5c68b38e Date: Fri 2018-06-15 19:16:56 KST Ucode Ver: 1181.2019 FWID: 01-b97cda56
```
Rom extraction builds without any error, so I won't paste any log.

After copying patched firmware to /vendor/etc/wifi and set right permissions, normal wifi operation seems not to be working. If I perform a 

> ifconfig wlan0 down
> ifconfig wlan0 up

The console hungs and I need to press Ctrl+C to be able to continue using it and if I enable Wifi through android UI, it does not show any wireless network.
The device seems to enter in a 4 o 5 iterations loop trying to set the device up logging this on dmesg:

```
[  490.272671] dhd_bus_devreset: == Power ON ==
[  490.285092] DHD: dongle ram size is set to 1376256(orig 1376256) at 0x170000
[  490.285422] dhdpcie_irq_enabled: depth:0
[  490.285702] dhd_read_file: Couldn't read the file /data/misc/conn/.cid.info, ret=-4
[  490.285754] dhd_check_module_bcm4361: failed to get module infomaion from .cid.info
[  490.286682] dhd_set_blob_support: ----- blob file exist -----
[  490.286729] dhd_bus_download_firmware: firmware path=/vendor/etc/wifi/bcmdhd_sta.bin_b0, nvram path=/vendor/etc/wifi/nvram.txt_r02j_b0
[  490.286783] dhdpcie_dump_resource: BAR0(VA): 0x0000000000000000, BAR0(PA): 0x0000000011b00000, SIZE: 32768
[  490.286834] dhdpcie_dump_resource: BAR1(VA): 0x0000000000000000, BAR1(PA): 0x0000000011800000, SIZE: 4194304
[  490.287758] dhdpcie_download_code_file: download firmware /vendor/etc/wifi/bcmdhd_sta.bin_b0
[  490.288196] dhdpcie_download_code_file: dhd_os_get_image_block failed (-4)
[  490.288246] _dhdpcie_download_firmware: dongle image file download failed
[  490.288294] dhd_bus_start: failed to download firmware /vendor/etc/wifi/bcmdhd_sta.bin_b0
[  490.288463] dhd_bus_devreset: dhd_bus_start: -1
[  490.288509] dhd_net_bus_devreset: dhd_bus_devreset: -1
[  490.288649] dhd_wl_ioctl: returning as busstate=0
[  490.288728] dhd_bus_devreset: == Power OFF ==
```

There's one failing function
```
[  490.288196] dhdpcie_download_code_file: dhd_os_get_image_block failed (-4)
```

And as device firmware won't load, I cant use nexutil to try to send -g0.

Here're full dmesg logs

Full DMESG: https://pastebin.com/L99DF82M
dhd grepped: https://pastebin.com/qf0KCsaU

Hope this helps

matthiasseemoo, 2019-01-01:
> UCODE compression is used when patching firmware, I can extract code used to do this, but I don't think got enough luck.
> 
> Anyway, if you need SSH access to an S8, I can provide it.
> 
> Kind regards!

Hi, I would be happy, if you could give me SSH access to your S8 so that I could debug the firmware patch.

LoloBee, 2019-01-01:
Absolutely yes! Let me set a machine up and I'll give you access.

Thanks!

LoloBee, 2019-01-02:
I've send you an email @matthiasseemoo

matthiasseemoo, 2019-01-03:
ROM extraction works now, simply run "make dump-rom". I made a mistake when
handling flashpatches before.

So far I did not find a wlc_monitor or wl_monitor function in the firmware,
which is wired as it was always available in any other firmware I saw so
far. Anyways, it will be possible to reimplement them.

LoloBee, 2019-01-04:
Great Matthias!
I don't know if wl_monitor has been implemented in all firmwares but, do
you think manufacturers are trying to get monitor mode working more
difficult?

I've never implemented a new function into firmware but I'll try.

If I have any problem I'll post it.

Kind regards.

matthiasseemoo, 2019-01-06:
We can now capture frames using wlc_recv, however, I only observed Beacon Frames so far. There is no radiotap header so far. you have to run nexutil -m first and then activate monitor mode using nexutil -m1, it it fails, try it again. To dump frames, simply run tcpdump: LD_PRELOAD=libnexmon.so tcpdump -i wlan0 -xxx

m4stersh4d0w, 2019-02-05:
Hello,

I am having the same issue, I can't use dhdutil membytes to dump the actual ROM contents on my GS8? I am getting the same error as already described above:
`dhdutil: __dhd_driver_io: Operation not supported on transport endpoint` 

dhdutil ramsize and ramstart works perfectly fine. What was the solution to this problem?

Thank you.

Steve

hy-l, 2019-03-22:
Hi, @matthiasseemoo 

I'm a bit confused by the difference between "ROM" and "firmware" described in README.
> The Wi-Fi firmware consists of a read-only part stored in the ROM of every Wi-Fi chip and another part that is loaded by the driver into the RAM.

In particular to Galaxy S8, the firmware is "bcmdhd_sta.bin_b0", isn't it? Then what should the ROM be? There are also some other files like "bcmdhd_clm.blob", "bcmdhd_mfg.bin_b0", "bcmdhd_mfg.bin_b2", and "bcmdhd_sta.bin_b2", do they have anything to do with this?

Also, I saw some different files in nougat /system/etc/wifi directory (with different sha1 sums), is it worth to do some research on them? 

And by the way, which address should I use with `dhdutil`? Just like OP said `dhdutil membytes 0x0 128`, or how can I figure it out? 

Sorry too many questions... ;)

Thanks in advance.

hy-l, 2019-03-25:
Hi - again,

I've done some diggings on my phone these days, here are some things I've found out:

- There are some dot files in `/data/misc/conn` that are created automatically while loading firmware. They contain (cached?) some info (including version numbers) about the firmware just loaded. If you try to load another firmware without deleting them, the old info may be retained in some cases, which might be misleading.
- There are two binaries `/vendor/bin/hw/macloader` and `/vendor/bin/hw/mfgloader` that are used to do all the initialization works for wifi device. In particular, the former loads `bcmdhd_sta.bin*` and the later loads `bcmdhd_mfg.bin*` (test mode). The `/init.wifi.rc` script can proof this.
```
dreamqlte:/data/ssh/root # cat /init.wifi.rc
 . . . 
service wpa_supplicant /system/vendor/bin/hw/wpa_supplicant -iwlan0 -Dnl80211 -c/data/misc/wifi/wpa_supplicant.conf -I/system/vendor/etc/wifi/wpa_supplicant_overlay.conf -O/data/misc/wifi/sockets -N -ip2p0 -Dnl80211 -c/data/misc/wifi/p2p_supplicant.conf -I/system/vendor/etc/wifi/p2p_supplicant_overlay.conf -puse_p2p_group_interface=1 -puse_multi_chan_concurrent=1 -e/data/misc/wifi/entropy.bin -g@android:wpa_wlan0
    class main
    socket wpa_wlan0 dgram 660 wifi wifi
    disabled
    oneshot

# tiburona
# start for WiFi MFG (TestMode)
service mfgloader /system/vendor/bin/hw/mfgloader
    class main
    disabled
    oneshot

service wlandutservice /system/bin/wlandutservice
    class main
    user system
    group wifi net_raw net_admin system inet
    disabled
    oneshot

service macloader /system/vendor/bin/hw/macloader
    class main
    oneshot

on property:init.svc.macloader=stopped
    chown system root /data/misc/conn/.cid.info
    chmod 0664 /data/misc/conn/.cid.info
    chown system root /data/.rev
    chmod 0664 /data/.rev

# for wlan FTM app
service ftm_ptt /system/bin/ftm_ptt -d
    user root
    group radio system
    disabled

# end of WiFi MFG (TestMode)
# It is present in hostapd.android.rc
#service hostapd /system/vendor/bin/hw/hostapd -dddd /data/hostapd/hostapd.conf
#    class late_start
#    user root
#    group root
#    oneshot
#    disabled
dreamqlte:/data/ssh/root # 
```
- To see the differences:
```
dreamqlte:/storage/emulated/0 # rm -f /data/misc/conn/.[!.]*
dreamqlte:/storage/emulated/0 # /vendor/bin/hw/macloader
dreamqlte:/storage/emulated/0 # grep -R . /sys/module/dhd/parameters/
/sys/module/dhd/parameters/st_str_file_path:/data/misc/conn/rtecdc.bin
/sys/module/dhd/parameters/dhd_console_ms:0
/sys/module/dhd/parameters/nvram_path:/vendor/etc/wifi/nvram_mfg.txt
/sys/module/dhd/parameters/passive_channel_skip:0
/sys/module/dhd/parameters/rom_map_file_path:/data/misc/conn/roml.map
/sys/module/dhd/parameters/logstrs_path:/data/misc/conn/logstrs.bin
/sys/module/dhd/parameters/info_string: Driver: 1.77.71 (r)
/sys/module/dhd/parameters/info_string: Firmware: wl0: Jun 15 2018 19:14:58 version 13.38.55.1 (B0 Network/rsdb) FWID 01-b97cda56
/sys/module/dhd/parameters/info_string: Chip: 4347 Rev 3 Pkg 0
/sys/module/dhd/parameters/disable_proptx: 0
/sys/module/dhd/parameters/rom_st_str_file_path:/data/misc/conn/roml.bin
/sys/module/dhd/parameters/h2d_max_txpost:512
/sys/module/dhd/parameters/dhd_napi__weight:32
/sys/module/dhd/parameters/op_mode:0
/sys/module/dhd/parameters/instance_base:0
/sys/module/dhd/parameters/map_file_path:/data/misc/conn/rtecdc.map
/sys/module/dhd/parameters/firmware_path:/vendor/etc/wifi/bcmdhd_sta.bin
dreamqlte:/storage/emulated/0 # 
dreamqlte:/storage/emulated/0 # rm -f /data/misc/conn/.[!.]*
dreamqlte:/storage/emulated/0 # /vendor/bin/hw/mfgloader
dreamqlte:/storage/emulated/0 # grep -R . /sys/module/dhd/parameters/
/sys/module/dhd/parameters/st_str_file_path:/data/misc/conn/rtecdc.bin
/sys/module/dhd/parameters/dhd_console_ms:0
/sys/module/dhd/parameters/nvram_path:/vendor/etc/wifi/nvram_mfg.txt
/sys/module/dhd/parameters/passive_channel_skip:0
/sys/module/dhd/parameters/rom_map_file_path:/data/misc/conn/roml.map
/sys/module/dhd/parameters/logstrs_path:/data/misc/conn/logstrs.bin
/sys/module/dhd/parameters/info_string: Driver: 1.77.71 (r)
/sys/module/dhd/parameters/info_string: Firmware: wl0: Jun 15 2018 19:19:35 version 13.38.55.1 (B0 WLTEST) FWID 01-f50cc893
/sys/module/dhd/parameters/info_string: Chip: 4347 Rev 3 Pkg 0
/sys/module/dhd/parameters/disable_proptx:0
/sys/module/dhd/parameters/rom_st_str_file_path: /data/misc/conn/roml.bin
/sys/module/dhd/parameters/h2d_max_txpost:512
/sys/module/dhd/parameters/dhd_napi_weight:32
/sys/module/dhd/parameters/op_mode:0
/sys/module/dhd/parameters/instance_base:0
/sys/module/dhd/parameters/map_file_path:/data/misc/conn/rtecdc.map
/sys/module/dhd/parameters/firmware_path:/vendor/etc/wifi/bcmdhd_mfg.bin
dreamqlte:/storage/emulated/0 # 
```
I can then enable wifi through command line with `ifconfig wlan0 up` and scan for APs with `iw wlan0 scan`, though GUI wlan settings are no longer working at this point.
- There seems to be a function called `wlc_monitor_attach` in `bcmdhd_mfg.bin_b0`
```
dreamqlte:/storage/emulated/0 # strings /vendor/etc/wifi/bcmdhd_mfg.bin_b0 | grep monitor
wlc_monitor_attach
monitor_promisc_level
wl%d: radio_disabled %x radio_monitor %d delay_off = %dlast_radio_disabled = %d
wlc_monitor_attach
```
is it the function matthiasseemoo mentioned above?

Talked so much, hopefully something is useful. 

As always, appreciate for you guy's works on this. :)

Alexxdal, 2021-11-18:
> Hi - again,
>
> I've done some diggings on my phone these days, here are some things I've found out:
>
>     * There are some dot files in `/data/misc/conn` that are created automatically while loading [...]

Yes, there are the function needed to activate monitor mode i think, im stuck after extracting the rom i merged it with the ram firmware and reversing it with ghidra i was able to find functions but there is no monitor fun, instead in the other files bcmdhd_mon.bin2 and mfg there is the monitor function.
Now how can i import the function? It need to be rewritten? And how?