seemoo-lab/nexmon #249: What would it take to compile it on Arch on RPi3?

splithor1zon, 2018-09-15:
I found compiling this driver on arch linux quite troublesome, but I finished it. The driver and firmware loaded and were ready to go. I could successfully connect on wifi using wlan0 but the trouble begun when I tried to use monitor mode interface mon0. I have set up the mon0 interface using iw command as noted in readme, but when I tried to use for example wifite (airocrack-ng) it just started to throw errors "Set Channel Failed" and then if I tried to use the wireless driver for anything else, it just froze and threw at me more errors "mcast failed and such... The only way out was to reload the driver. I got so frustrated I needed to check whether or not nexmon patch actually works, so I installed Kali-Pi with reason kernel (4.9) and everything is running smoothly... But that doesn't change the fact I want to use Arch with kernel 4.14 (4.14.69-2). I got the Arch back and tried to compile it freshly again, still no success.. Is there anything I can try to fix this problem? Or is there anybody who can please help me with this problem? I can supply logs, just tell me which ones.

matthiasseemoo, 2018-09-17:
check that the driver works with your kernel version. if you need another
driver, you have to integrate the changes we have in our repo. you should
also make sure that the firmware version matches the version expected by
the driver.

-- 
Matthias Schulz
Secure Mobile Networking Lab - SEEMOO

Email: matthias.schulz@seemoo.tu-darmstadt.de
Web: http://www.seemoo.de/mschulz
Phone (new): +49 6151 16-25478    Fax: +49 6151 16-25471

Department of Computer Science
Center for Advanced Security Research Darmstadt
Technische Universität Darmstadt
Mornewegstr. 32 (Office 4.2.10, Building S4/14)
D-64293 Darmstadt, Germany

splithor1zon, 2018-09-17:
Well so basically I need to use your patch on the driver and firmware supplied by Arch and check whether or not the firmware isn't different version that is expected by the driver? Or am I misunderstanding something? This project is oriented on firmware patching but when I install customized firmware I also need to install customized driver, am I right? It is obvious, that fresh build on Arch doesn't work well... Can you please point me to some sort of manual which can explain the process of patching firmware/driver and what to do when we need to customize it for our platform? It will speed up the process of spreading nexmon driver to much more people with various platforms and distributions.

Eugenio-Liso, 2018-09-24:
I didn't even came up with an installation of Nexmon on my RPI3...Using arch armv7 (last versione), i'm stuck on an error on the first ./configure (in path nexmon/buildtools/isl-0.10). I cannot proceed as my config.log says:

```
configure:2762: checking for a BSD-compatible install
configure:2830: result: /usr/bin/install -c
configure:2841: checking whether build environment is sane
configure:2891: result: yes
configure:3032: checking for a thread-safe mkdir -p
configure:3071: result: /usr/bin/mkdir -p
configure:3084: checking for gawk
configure:3100: found /usr/bin/gawk
configure:3111: result: gawk
configure:3122: checking whether make sets $(MAKE)
configure:3144: result: yes
configure:3287: checking for gcc
configure:3314: result: /home/alarm/nexmon/buildtools/gcc-arm-none-eabi-5_4-2016q2-linux-armv7l/bin/arm-none-eabi-
configure:3543: checking for C compiler version
configure:3552: /home/alarm/nexmon/buildtools/gcc-arm-none-eabi-5_4-2016q2-linux-armv7l/bin/arm-none-eabi- --version >&5
./configure: line 3554: /home/alarm/nexmon/buildtools/gcc-arm-none-eabi-5_4-2016q2-linux-armv7l/bin/arm-none-eabi-: No such file or directory
configure:3563: $? = 127
configure:3552: /home/alarm/nexmon/buildtools/gcc-arm-none-eabi-5_4-2016q2-linux-armv7l/bin/arm-none-eabi- -v >&5
./configure: line 3554: /home/alarm/nexmon/buildtools/gcc-arm-none-eabi-5_4-2016q2-linux-armv7l/bin/arm-none-eabi-: No such file or directory
configure:3563: $? = 127
configure:3552: /home/alarm/nexmon/buildtools/gcc-arm-none-eabi-5_4-2016q2-linux-armv7l/bin/arm-none-eabi- -V >&5
./configure: line 3554: /home/alarm/nexmon/buildtools/gcc-arm-none-eabi-5_4-2016q2-linux-armv7l/bin/arm-none-eabi-: No such file or directory
configure:3563: $? = 127
configure:3552: /home/alarm/nexmon/buildtools/gcc-arm-none-eabi-5_4-2016q2-linux-armv7l/bin/arm-none-eabi- -qversion >&5
./configure: line 3554: /home/alarm/nexmon/buildtools/gcc-arm-none-eabi-5_4-2016q2-linux-armv7l/bin/arm-none-eabi-: No such file or directory
configure:3563: $? = 127
configure:3583: checking whether the C compiler works
configure:3605: /home/alarm/nexmon/buildtools/gcc-arm-none-eabi-5_4-2016q2-linux-armv7l/bin/arm-none-eabi-    conftest.c  >&5
./configure: line 3607: /home/alarm/nexmon/buildtools/gcc-arm-none-eabi-5_4-2016q2-linux-armv7l/bin/arm-none-eabi-: No such file or directory
configure:3609: $? = 127
configure:3647: result: no
configure: failed program was:
| /* confdefs.h */
```

How did you pass this step?

matthiasseemoo, 2018-09-24:
You have to build libisl before running source setup_env.sh

splithor1zon, 2018-09-25:
> 
> 
> I didn't even came up with an installation of Nexmon on my RPI3...Using arch armv7 (last version), i'm stuck on an error on the first ./configure (in path nexmon/buildtools/isl-0.10). I cannot proceed as my config.log says:
> 
> ```
> configure:2762: checking for a BSD-compatible install
> configure:2830: result: /usr/bin/install -c
> configure:2841: checking whether build environment is sane
> configure:2891: result: yes
> configure:3032: checking for a thread-safe mkdir -p
> configure:3071: result: /usr/bin/mkdir -p
> configure:3084: checking for gawk
> configure:3100: found /usr/bin/gawk
> configure:3111: result: gawk
> configure:3122: checking whether make sets $(MAKE)
> configure:3144: result: yes
> configure:3287: checking for gcc
> configure:3314: result: /home/alarm/nexmon/buildtools/gcc-arm-none-eabi-5_4-2016q2-linux-armv7l/bin/arm-none-eabi-
> configure:3543: checking for C compiler version
> configure:3552: /home/alarm/nexmon/buildtools/gcc-arm-none-eabi-5_4-2016q2-linux-armv7l/bin/arm-none-eabi- --version >&5
> ./configure: line 3554: /home/alarm/nexmon/buildtools/gcc-arm-none-eabi-5_4-2016q2-linux-armv7l/bin/arm-none-eabi-: No such file or directory
> configure:3563: $? = 127
> configure:3552: /home/alarm/nexmon/buildtools/gcc-arm-none-eabi-5_4-2016q2-linux-armv7l/bin/arm-none-eabi- -v >&5
> ./configure: line 3554: /home/alarm/nexmon/buildtools/gcc-arm-none-eabi-5_4-2016q2-linux-armv7l/bin/arm-none-eabi-: No such file or directory
> configure:3563: $? = 127
> configure:3552: /home/alarm/nexmon/buildtools/gcc-arm-none-eabi-5_4-2016q2-linux-armv7l/bin/arm-none-eabi- -V >&5
> ./configure: line 3554: /home/alarm/nexmon/buildtools/gcc-arm-none-eabi-5_4-2016q2-linux-armv7l/bin/arm-none-eabi-: No such file or directory
> configure:3563: $? = 127
> configure:3552: /home/alarm/nexmon/buildtools/gcc-arm-none-eabi-5_4-2016q2-linux-armv7l/bin/arm-none-eabi- -qversion >&5
> ./configure: line 3554: /home/alarm/nexmon/buildtools/gcc-arm-none-eabi-5_4-2016q2-linux-armv7l/bin/arm-none-eabi-: No such file or directory
> configure:3563: $? = 127
> configure:3583: checking whether the C compiler works
> configure:3605: /home/alarm/nexmon/buildtools/gcc-arm-none-eabi-5_4-2016q2-linux-armv7l/bin/arm-none-eabi-    conftest.c  >&5
> ./configure: line 3607: /home/alarm/nexmon/buildtools/gcc-arm-none-eabi-5_4-2016q2-linux-armv7l/bin/arm-none-eabi-: No such file or directory
> configure:3609: $? = 127
> configure:3647: result: no
> configure: failed program was:
> | /* confdefs.h */
> ```
> 
> How did you pass this step?

I installed libisl, created links in the right directories but it didn't find the lib... So I just created link to libisl.so in /usr/local/lib named libisl.so.10 and then I set variable "export LD_LIBRARY_PATH=/usr/local/lib" and everything went through.

bag-man, 2018-12-18:
libmpfr.so.4 is missing on Arch too.

mirzaatifbaig, 2018-12-19:
2why nkt0 6f 6e 20 4c 4c  56 4d 20 33 2e 38 2e 32  |ed on LLVM 3.8.2|
00003060  35 36 32 32 39 29 00 00  04 00 00 00 09 00 00 00
|56229)..........|
00003070  04 00 00 00 47 4e 55 00  67 6f 6c 64 20 31 2e 31  |....GNU.gold
1.1|
00003080  31 00 00 00 41 3b 00 00  00 61 65 61 62 69 00 01
|1...A;...aeabi..|
00003090  31 00 00 00 05 41 52 4d  20 76 37 00 06 0a 07 41  |1....ARM
v7....A|
000030a0  08 01 09 02 0a 05 0c 02  11 02 12 04 14 01 15 01
|................|
000030b0  17 03 18 01 1a 02 1e 06  22 01 24 01 26 01 2c 02
|........".$.&.,.|
000030c0  6c 69 62 64 6c 2e 73 6f  00 00 00 00 09 3a 06 c3
|libdl.so.....:..|
000030d0  00 2e 73 68 73 74 72 74  61 62 00 2e 69 6e 74 65
|..shstrtab..inte|
000030e0  72 70 00 2e 6e 6f 74 65  2e 61 6e 64 72 6f 69 64
|rp..note.android|
000030f0  2e 69 64 65 6e 74 00 2e  6e 6f 74 65 2e 67 6e 75
|.ident..note.gnu|
00003100  2e 62 75 69 6c 64 2d 69  64 00 2e 64 79 6e 73 79
|.build-id..dynsy|
00003110  6d 00 2e 64 79 6e 73 74  72 00 2e 67 6e 75 2e 68
|m..dynstr..gnu.h|
00003120  61 73 68 00 2e 67 6e 75  2e 76 65 72 73 69 6f 6e
|ash..gnu.version|
00003130  00 2e 67 6e 75 2e 76 65  72 73 69 6f 6e 5f 64 00
|..gnu.version_d.|
00003140  2e 72 65 6c 2e 64 79 6e  00 2e 72 65 6c 2e 70 6c  |.rel.dyn..
rel.pl|
00003150  74 00 2e 74 65 78 74 00  2e 41 52 4d 2e 65 78 69
|t..text..ARM.exi|
00003160  64 78 00 2e 41 52 4d 2e  65 78 74 61 62 00 2e 66
|dx..ARM.extab..f|
00003170  69 6e 69 5f 61 72 72 61  79 00 2e 64 79 6e 61 6d
|ini_array..dynam|
00003180  69 63 00 2e 67 6f 74 00  2e 64 61 74 61 00 2e 63
|ic..got..data..c|
00003190  6f 6d 6d 65 6e 74 00 2e  6e 6f 74 65 2e 67 6e 75
|omment..note.gnu|
000031a0  2e 67 6f 6c 64 2d 76 65  72 73 69 6f 6e 00 2e 41
|.gold-version..A|
000031b0  52 4d 2e 61 74 74 72 69  62 75 74 65 73 00 2e 67
|RM.attributes..g|
20 6f 6e 20 4c 4c  56 4d 20 33 2e 38 2e 32  |ed on LLVM 3.8.2|
00003060  35 36 32 32 39 29 00 00  04 00 00 00 09 00 00 00
|56229)..........|
00003070  04 00 00 00 47 4e 55 00  67 6f 6c 64 20 31 2e 31  |....GNU.gold
1.1|
00003080  31 00 00 00 41 3b 00 00  00 61 65 61 62 69 00 01
|1...A;...aeabi..|
00003090  31 00 00 00 05 41 52 4d  20 76 37 00 06 0a 07 41  |1....ARM
v7....A|
000030a0  08 01 09 02 0a 05 0c 02  11 02 12 04 14 01 15 01
|................|
000030b0  17 03 18 01 1a 02 1e 06  22 01 24 01 26 01 2c 02
|........".$.&.,.|
000030c0  6c 69 62 64 6c 2e 73 6f  00 00 00 00 09 3a 06 c3
|libdl.so.....:..|
000030d0  00 2e 73 68 73 74 72 74  61 62 00 2e 69 6e 74 65
|..shstrtab..inte|
000030e0  72 70 00 2e 6e 6f 74 65  2e 61 6e 64 72 6f 69 64
|rp..note.android|
000030f0  2e 69 64 65 6e 74 00 2e  6e 6f 74 65 2e 67 6e 75
|.ident..note.gnu|
00003100  2e 62 75 69 6c 64 2d 69  64 00 2e 64 79 6e 73 79
|.build-id..dynsy|
00003110  6d 00 2e 64 79 6e 73 74  72 00 2e 67 6e 75 2e 68
|m..dynstr..gnu.h|
00003120  61 73 68 00 2e 67 6e 75  2e 76 65 72 73 69 6f 6e
|ash..gnu.version|
00003130  00 2e 67 6e 75 2e 76 65  72 73 69 6f 6e 5f 64 00
|..gnu.version_d.|
00003140  2e 72 65 6c 2e 64 79 6e  00 2e 72 65 6c 2e 70 6c  |.rel.dyn..
rel.pl|
00003150  74 00 2e 74 65 78 74 00  2e 41 52 4d 2e 65 78 69
|t..text..ARM.exi|
00003160  64 78 00 2e 41 52 4d 2e  65 78 74 61 62 00 2e 66
|dx..ARM.extab..f|
00003170  69 6e 69 5f 61 72 72 61  79 00 2e 64 79 6e 61 6d
|ini_array..dynam|
00003180  69 63 00 2e 67 6f 74 00  2e 64 61 74 61 00 2e 63
|ic..got..data..c|
00003190  6f 6d 6d 65 6e 74 00 2e  6e 6f 74 65 2e 67 6e 75
|omment..note.gnu|
000031a0  2e 67 6f 6c 64 2d 76 65  72 73 69 6f 6e 00 2e 41
|.gold-version..A|
000031b0  52 4d 2e 61 74 74 72 69  62 75 74 65 73 00 2e 67
|RM.attributes..g|