seemoo-lab/nexmon #251: libfakeioctlkali Chroot

mirzaatifbaig, 2018-09-16:
I compiled and placed libfakeioctlkali and monstart-nh in a Note 3 and successfully enable monitor mode in chroot but When I run airodump-ng wlan0.I get a few second freeze and a reboot.The kernel is from nethunter.What exactly am I doing wrong.I built all tools from this repo and they work like a charm.

pixeldublu, 2018-10-31:
can you share steps how to compile monstart stop nh ? Thanks

pixeldublu, 2018-11-01:
Nevermind.. found somewhere else.

Thanks for the insult.

JuniorJPDJ, 2019-02-18:
How did you compile libfakeioctl?

46736f6369657479, 2019-07-20:
@mirzaatifbaig From my understanding you have to rebuild the kernel for your device with the nexmon drivers and patches for nexmon to work inside the kali chroot. I am currently trying to figure out how to properly do this as there seems to be no proper documentation on how to build a kernel for nexmon to work within the kali chroot. I hope to make my own kernels for the Galaxy s7 and nexus 5 eventually.

mirzaatifbaig, 2019-07-20:
> @mirzaatifbaig From my understanding you have to rebuild the kernel for your device with the nexmon drivers and patches for nexmon to work inside the kali chroot. I am currently trying to figure out [...]

Just keep me updated, I gave up a while ago but am interested nonetheless, Good Luck.

mirzaatifbaig, 2019-09-13:
> @mirzaatifbaig From my understanding you have to rebuild the kernel for your device with the nexmon drivers and patches for nexmon to work inside the kali chroot. I am currently trying to figure out [...]

So, Did you find anything, I eventually figured it out but I don't think I should share, After someone posted about this project on xda, it got a lot of heat and people were pretty excited into becoming leet haxor1337 and the only obstacle was not being able to turn monitor mode inside chroot, So I think It should stay that way.

46736f6369657479, 2019-09-22:
I have not been able to find anything still. I tried talking to one of the devs of the nexmon project but they did not seem to know the complete answer either. I would, as I am sure many others would appreciate it if you would share any info you have but I can understand why you wouldn't as well. A hint would even suffice for me as I am willing to work for it.

mirzaatifbaig, 2019-09-26:
Here it goes, modify the bcmdhd driver makefile and the driver files for your kernel in the source and add the Monitor Mode Support such that it doesn't need the access to the lib anymore, The files you would need to modify are the Makefile, dhd.h, wl_cfg80211.c, wl_cfg80211.h. Sorry man I can't help you or anyone more than that.

closed 2019-09-26 by mirzaatifbaig