seemoo-lab/nexmon #256: RPI 3B+ not capturing anything on Raspbian

marcelser, 2018-10-03:
Hi,

I installed the nexutil driver and it seems it installed just fine. I have mon0 available, however when running airodump-ng mon0 or tcpdump -i mon0 even after 30 mins not a single output.

I'm using the driver & firmware on latest raspbian stretch, Anybody knows what's wrong here?

ifconfig shows:
`eth0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.10.66  netmask 255.255.255.0  broadcast 192.168.10.255
        inet6 fda6:7fef:b222::869  prefixlen 128  scopeid 0x0<global>
        inet6 fda6:7fef:b222:0:e80d:32fb:bc4:7645  prefixlen 64  scopeid 0x0<global>
        inet6 fe80::d58e:a994:80a9:fce0  prefixlen 64  scopeid 0x20<link>
        ether b8:27:eb:de:60:76  txqueuelen 1000  (Ethernet)
        RX packets 5738  bytes 789981 (771.4 KiB)
        RX errors 0  dropped 2  overruns 0  frame 0
        TX packets 1282  bytes 309340 (302.0 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

mon0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        unspec B8-27-EB-8B-35-23-30-30-00-00-00-00-00-00-00-00  txqueuelen 1000  (UNSPEC)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

wlan0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 169.254.162.163  netmask 255.255.0.0  broadcast 169.254.255.255
        inet6 fe80::41e3:66ac:42e2:2616  prefixlen 64  scopeid 0x20<link>
        ether b8:27:eb:8b:35:23  txqueuelen 1000  (Ethernet)
        RX packets 50  bytes 8859 (8.6 KiB)
        RX errors 0  dropped 50  overruns 0  frame 0
        TX packets 55  bytes 14943 (14.5 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
`

DrSchottky, 2018-10-08:
Happened to me too.
The only way to make it working was starting mon0 with airmon-ng, that basically does
`sudo ip link set dev wlan0 down`
``sudo iw phy `iw dev wlan0 info | gawk '/wiphy/ {printf "phy" $2}'` interface add mon0 type monitor``
`sleep 1`
`sudo ip link set dev mon0 up`
`sudo iwconfig mon0 channel X`

But can't reproduce it now 

BTW monitor mode on 3B+ is very unstable (fw is prone to crashing)

BennyE, 2018-10-19:
I don't see actual "QOS DATA" frames with encrypted content, does that work for you?

MaximusBaton, 2018-11-15:
I had the same problem and my mistake was the wrong chip definition.
Try this: https://github.com/seemoo-lab/nexmon/issues/259#issuecomment-438768001

zrion, 2018-12-10:
I'm having the same issues on RPi3+. Have checked everything and installation was fine, without errors. Still, using either tcpdump or airodump-ng provides nothing. I use the patch for bcm43455c0.

matthiasseemoo, 2018-12-10:
Could you check, whether the following version works?
https://github.com/seemoo-lab/nexmon/tree/d64fcef31d715b2f074bea10e9e6d93d0a289b5d

-- 
Matthias Schulz
Secure Mobile Networking Lab - SEEMOO

Email: matthias.schulz@seemoo.tu-darmstadt.de
Web: http://www.seemoo.de/mschulz
Phone (new): +49 6151 16-25478    Fax: +49 6151 16-25471

Department of Computer Science
Center for Advanced Security Research Darmstadt
Technische Universität Darmstadt
Mornewegstr. 32 (Office 4.2.10, Building S4/14)
D-64293 Darmstadt, Germany

zrion, 2018-12-11:
This one doesn't work on my side either. Sniffing on mon0 doesn't get anything. I'm using RPiB3+ with Raspbian 9 Kernel 4.14.
Also, I have the same problem with the SDR patch at https://github.com/seemoo-lab/mobisys2018_nexmon_software_defined_radio, on the same RPi3+. Having tried different channels but couldn't see anything on spectrum analyzer. Maybe these two are related.

matthiasseemoo, 2018-12-11:
Then, try an older Raspbian version.

zrion, 2018-12-12:
Thanks. Reverting to Raspbian build 2018-03-13 makes it work for monitor mode patch. Also for SDR patch it can transmit something, though I haven't been able to inspect the packets with radiotap headers yet. However, I couldn't stop the transmission with ioctl 428, as I need to load the firmware again to make it stop.