seemoo-lab/nexmon #259: RPI 3B+ (Stretch, Kernel 4.14.71, bcm43430a1) Build

MaximusBaton, 2018-11-12:
Hello everyone!
First of all **thank you** guys for making this product!

I'm stuck on the successful build. (actually trying make it work for a week)
If you have some time, your help would be very appreciated!

I'm sure my Pi has the possibility to be in a monitor mode as I tried Kali+re5son build

So here are the step:

- Install latest Stretch Lite [2018-10-09-raspbian-stretch-lite.img](https://www.raspberrypi.org/downloads/raspbian/)

- sudo su

- uname -a
```
Linux raspberry 4.14.71-v7+ #1145 SMP Fri Sep 21 15:38:35 BST 2018 armv7l 
```

- apt update && apt upgrade && apt-get update && apt-get upgrade

- apt install -y raspberrypi-kernel-headers git libgmp3-dev gawk qpdf bison flex make

- apt-get install -y autoconf automake build-essential libtool

- ifconfig wlan0
```
wlan0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.1.75  netmask 255.255.255.0  broadcast 192.168.1.255
        inet6 xxxx::xxxx:xxxx:xxxx:xxxx  prefixlen 64  scopeid 0x20<link>
        ether xx:xx:xx:xx:xx:xx  txqueuelen 1000  (Ethernet)
        RX packets 80  bytes 12913 (12.6 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 44  bytes 6803 (6.6 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
```

- cd ~ && mkdir -p bcm43430_driverHack && cd bcm43430_driverHack

- git clone https://github.com/seemoo-lab/nexmon.git

- cd nexmon

- ls /usr/lib/arm-linux-gnueabihf | grep libisl
```
libisl.so.15
libisl.so.15.3.0
```

- cd buildtools/isl-0.10

- **autoreconf -f -i**
```
libtoolize: putting auxiliary files in AC_CONFIG_AUX_DIR, '.'.
libtoolize: copying file './ltmain.sh'
libtoolize: putting macros in AC_CONFIG_MACRO_DIRS, 'm4'.
libtoolize: copying file 'm4/libtool.m4'
libtoolize: copying file 'm4/ltoptions.m4'
libtoolize: copying file 'm4/ltsugar.m4'
libtoolize: copying file 'm4/ltversion.m4'
libtoolize: copying file 'm4/lt~obsolete.m4'
configure.ac:9: installing './compile'
configure.ac:4: installing './missing'
Makefile.am:35: warning: 'INCLUDES' is the old name for 'AM_CPPFLAGS' (or '*_CPPFLAGS')
Makefile.am: installing './depcomp'
parallel-tests: installing './test-driver'
interface/Makefile.am:8: warning: 'INCLUDES' is the old name for 'AM_CPPFLAGS' (or '*_CPPFLAGS')
```

- ./configure
[... 624 lines cut ...]
wlan0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.1.75  netmask 255.255.255.0  broadcast 192.168.1.255
        inet6 xxxx::xxxx:xxxx:xxxx:xxxx  prefixlen 64  scopeid 0x20<link>
        ether xx:xx:xx:xx:xx:xx  txqueuelen 1000  (Ethernet)
        RX packets 11  bytes 1634 (1.5 KiB)
        RX errors 0  dropped 1  overruns 0  frame 0
        TX packets 19  bytes 2827 (2.7 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
```

- ifconfig mon0
```
mon0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        unspec XX-XX-XX-XX-XX-XX-00-00-00-00-00-00-00-00-00-00  txqueuelen 1000  (UNSPEC)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
```

----

Seems like not every step went smoothly.
Am I missing smth?
Thank you in advance!

MaximusBaton, 2018-11-13:
I fixed warnings on /root/bcm43430_driverHack/nexmon/make, still no luck (

MaximusBaton, 2018-11-13:
Look like the same steps worked fine on Pi Zero W

FibreFoX, 2018-11-14:
Coming from the readme, maybe you are using a wrong patch?
https://github.com/seemoo-lab/nexmon#supported-devices
Raspberry Pi B3+ = bcm43455c0

Now I'm confused if the documentation is wrong, as I'm having issues too.

MaximusBaton, 2018-11-14:
Damn! You might be right!
I will check that and get back to you

MaximusBaton, 2018-11-14:
FibreFoX, you were absolutely right! With bcm43455c0 works fine!

So, Raspberry 3B+ has **bcm43455c0** and **not** bcm43430 wifi chip.

Here is the script for clean install:
```
#!/bin/bash

set -e

#### Hack RaspberryPi (Zero W bcm43430's) or (3B+ bcm43455c0's) driver for WiFi
##

is_pizero() {
   grep -q "^Revision\s*:\s*[ 123][0-9a-fA-F][0-9a-fA-F][0-9a-fA-F]0[9cC][0-9a-fA-F]$" /proc/cpuinfo
   return $?
}

echo -e "== [$(date '+%Y-%m-%d %H:%M:%S')] ============= HACK brcmfmac === START ="

# Install the kernel headers to build the driver and some dependencies
apt-get update -y
apt install -y raspberrypi-kernel-headers git libgmp3-dev gawk qpdf bison flex make
apt-get install -y autoconf automake build-essential libtool

cd ~ && rm -f -r brcmfmac_driverHack
cd ~ && mkdir -p brcmfmac_driverHack && cd brcmfmac_driverHack

# I use my repo as made some mutes&fixes on warnings
git clone https://github.com/MaximusBaton/nexmon.git

# Go into the root directory of our repository
cd nexmon

# Check if /usr/lib/arm-linux-gnueabihf/libisl.so.10 exists
if [[ -e "/usr/lib/arm-linux-gnueabihf/libisl.so.10" ]]; then
    echo "libisl.so.10 exists. skipping"
else 
    cd buildtools/isl-0.10
    autoreconf -f -i
    ./configure
    make
    make install
    ln -s /usr/local/lib/libisl.so /usr/lib/arm-linux-gnueabihf/libisl.so.10
    cd ../..
fi

# Compile some build tools and extract the ucode and flashpatches from the original firmware files
make

# Go to the patches folder for the bcm43430a1/bcm43455c0 chipset
if is_pizero ; then
    cd patches/bcm43430a1/7_45_41_46/nexmon/
else
    cd patches/bcm43455c0/7_45_154/nexmon/
fi

# Compile a patched firmware
make

# Generate a backup of your original firmware file (e.g. will run: cp /lib/firmware/brcm/brcmfmac43430-sdio.bin brcmfmac43430-sdio.bin.orig)
make backup-firmware

# Install the patched firmware on your RPI3
make install-firmware

# Install nexutil
#   from the root directory of our repository switch to the nexutil folder
cd ../../../../utilities/nexutil/
#   Compile and install nexutil
make && make install

# Return back to root nexmon directory
cd ../..

# Optional: remove wpa_supplicant for better control over the WiFi interface
# apt-get remove -y wpasupplicant

# Setup a new 'monitor mode' interface 'mon0'
iw phy `iw dev wlan0 info | gawk '/wiphy/ {printf "phy" $2}'` interface add mon0 type monitor
# Set the interface up to activate monitor mode in the firmware
ifconfig mon0 up
#   At this point, monitor mode is active. There is no need to call airmon-ng.
#   The interface already set the Radiotap header, therefore, tools like tcpdump or airodump-ng can be used out of the box: tcpdump -i mon0

## To make the RPi load the modified driver after reboot
# Find the path of the default driver at reboot
# e.g. '/lib/modules/4.14.71-v7+/kernel/drivers/net/wireless/broadcom/brcm80211/brcmfmac/brcmfmac.ko'
PATH_OF_DEFAULT_DRIVER_AT_REBOOT=$(modinfo brcmfmac | grep -m 1 -oP "^filename:(\s*?)(.*)$" | sed -e 's/^filename:\(\s*\)\(.*\)$/\2/g')
# Backup the original driver
mv $PATH_OF_DEFAULT_DRIVER_AT_REBOOT "$PATH_OF_DEFAULT_DRIVER_AT_REBOOT.orig"
# Copy the modified driver (Kernel 4.14)
if is_pizero ; then
    cp ./patches/bcm43430a1/7_45_41_46/nexmon/brcmfmac_4.14.y-nexmon/brcmfmac.ko $PATH_OF_DEFAULT_DRIVER_AT_REBOOT
else
    cp ./patches/bcm43455c0/7_45_154/nexmon/brcmfmac_4.14.y-nexmon/brcmfmac.ko $PATH_OF_DEFAULT_DRIVER_AT_REBOOT
fi
# Probe all modules and generate new dependency
depmod -a

sleep 1

cd ~ && rm -r brcmfmac_driverHack

echo -e "== [$(date '+%Y-%m-%d %H:%M:%S')] ============= HACK brcmfmac === END ==="

```

closed 2018-11-14 by MaximusBaton

spadejack014, 2019-07-03:
> 
> 
> Both RPi 3B+ and Zero W have bcm43430a1 WiFi chip.
> But RPi 3B is armv7l, and ZW is armv6l
so the RPi 3B cannot run it ?

spadejack014, 2019-07-03:
> 
> 
> Both RPi 3B+ and Zero W have bcm43430a1 WiFi chip.
> But RPi 3B is armv7l, and ZW is armv6l

Can RPi 3B run this?
if it could, which bcm should I choose?

FibreFoX, 2019-07-03:
@spadejack014 you might want to reread this thread again, RPi 3B does work. This is even stated in the README: https://github.com/seemoo-lab/nexmon#build-patches-for-bcm43430a1-on-the-rpi3zero-w-or-bcm434355c0-on-the-rpi3-using-raspbian-recommended

spadejack014, 2019-07-03:
@FibreFoX
Sir, I still cannot work in on my RPi 3B (not 3B+).
I cannot seem to get the wlan0 interface to load the driver correctly even though all seems to load wuithout errors ( I am following the guide step by step ), no matter which driver I load. Both 43430a1 or 43455c have the same error.

spadejack014, 2019-07-03:
@FibreFoX 
Sir，thank you.
I find the reason.
just use the image 4.14.79.
Don't update and upgrade it.
then, it will work.