seemoo-lab/nexmon #260: I cant add type  monitor, and iw list is empy after installing the  firmware  on my Raspbian Stretch

KhurramPirov, 2018-11-13:


aiokswhatever, 2018-11-13:
You got to install firmware each time after reload before doing this

`Install the patched firmware on your RPI3: make install-firmware`

learnedcat, 2018-11-14:
After execution make install-firmware there is no wlan0 and phy0 interfaces. 'iw list' output is empty. Also there are no errors in var/log/messages.

MaximusBaton, 2018-11-15:
You didn't mention your actual device.
If you talking about Raspi 3B+ or Zero W, I made it working: https://github.com/seemoo-lab/nexmon/issues/259#issuecomment-438768001

spadejack014, 2019-07-02:
> 
> 
> After execution make install-firmware there is no wlan0 and phy0 interfaces. 'iw list' output is empty. Also there are no errors in var/log/messages.

I have same problems.
How did you repair it?

mrkskwsnck, 2020-05-02:
I ended up confronting the same problem on Raspbian 10 Buster:

* Succesfully compiled patch 7_45_41_46 for bcm43430a1 (Zero W)
* After make `install-firmware` the output of `iw list` is empty

I'm looking forward for any suggestions. Thanks in advcance

matthiasseemoo, 2020-05-02:
Des the firmware version of the patch fit to the original version? If the patched version is incompatible with your driver, it might not work.

mrkskwsnck, 2020-05-02:
OK, I just gave the patch 7_45_189 for bcm43455c0 (which shall be for 3B+/4 as stated in the README) a try… and now it works! I don't know why, but it works just fine.

BTW: My system runs the Linux kernel v4.19.97+