seemoo-lab/nexmon #261: External adapter support

Modeo147, 2018-12-09:
Is support for external adapters planned? ( Example tp-link wn722n)

matthiasseemoo, 2018-12-09:
No, wn722n does not have a broadcom chip.

matthiasseemoo, 2018-12-31:
But you can try to find an external adapter with Broadcom Chip an tell me. It shall at least support 802.11ac. You can also use a Raspberry Pi as external adapter.

theoneandonly-vector, 2019-01-03:
why is this only for broadcom / cypress?

matthiasseemoo, 2019-01-03:
Because every chipvendor has a different firmware. The firmwares for
broadcom and cypress chips are very similar and hence quite easy to
support. But of you want, you can use our framework to write patches in C
for any other chip that has a suitable processor, you just need to reverse
engineere the firmware first. Addirionally, there are some WiFi usb dongles
with broadcom chips that we could support. Additionally, you could use a
raspberry pi as external wifi card.