seemoo-lab/nexmon #265: 'Make' failed with error 'brcmfmac.ko'

BlueSun99, 2019-01-04:
> root@raspberrypi:/home/pi/Desktop/nexmon/patches/bcm43455c0/7_45_154/nexmon# make
> 
> 
>           ###########   ###########   ##########    ##########           
>          ############  ############  ############  ############          
>          ##            ##            ##   ##   ##  ##        ##          
>          ##            ##            ##   ##   ##  ##        ##          
>          ###########   ####  ######  ##   ##   ##  ##    ######          
>           ###########  ####  #       ##   ##   ##  ##    #    #          
>                    ##  ##    ######  ##   ##   ##  ##    #    #          
>                    ##  ##    #       ##   ##   ##  ##    #    #          
>          ############  ##### ######  ##   ##   ##  ##### ######          
>          ###########    ###########  ##   ##   ##   ##########           
> 
>             S E C U R E   M O B I L E   N E T W O R K I N G               
> 
> 
>                                presents:                                  
> 
>               # ###   ###  #   # # ###  ###   ###  # ###                  
>               ##   # #   #  # #  ##   ##   # #   # ##   #                 
>               #    # #####   #   #    #    # #   # #    #                 
>               #    # #      # #  #    #    # #   # #    #                 
>               #    #  #### #   # #    #    #  ###  #    #                 
> 
>                 The C-based Firmware Patching Framework                   
> 
> 
>                            !!! WARNING !!!                                
>     Our software may damage your hardware and may void your hardware’s    
>      warranty! You use our tools at your own risk and responsibility      
> 
> 
>   COLLECTING STATISTICS read /home/pi/Desktop/nexmon/STATISTICS.md for more information
>   COMPILING src/version.c => obj/version.o (details: log/compiler.log)
>   GENERATING LINKER FILE gen/nexmon.pre => gen/nexmon.ld
>   GENERATING LINKER FILE gen/nexmon.pre => gen/flashpatches.ld
>   LINKING OBJECTS => gen/patch.elf (details: log/linker.log, log/linker.err)
>   GENERATING MAKE FILE gen/nexmon.pre => gen/nexmon.mk
>   GENERATING MAKE FILE gen/nexmon.pre => gen/flashpatches.mk
>   APPLYING FLASHPATCHES gen/flashpatches.mk => brcmfmac43455-sdio.bin (details: log/flashpatches.log)
>   APPLYING PATCHES gen/nexmon.mk => brcmfmac43455-sdio.bin (details: log/patches.log)
>   BUILDING DRIVER for kernel 4.9 brcmfmac_4.9.y-nexmon/brcmfmac.ko (details: log/driver.log)
> make[1]: *** /lib/modules/4.14.90-v7+/build: No such file or directory. Stop.
> Makefile:45: recipe for target 'brcmfmac.ko' failed
> make: *** [brcmfmac.ko] Error 2

I compiled 'libisl' well, make 'ln' with root shell (sudo su), did 'source setup_env.sh', 'make'.
Also did 'apt update', 'apt upgrade', 'apt dist-upgrade', 'reboot', 'rm -rf ./nexmon', 'git clone' again.
Everything not works :(

What did I need to do?

+Info
RPI 3B+

Have 4 directories in /lib/modules
> root@raspberrypi:/lib/modules# ls
> 4.14.79+     4.14.90+
> 4.14.79-v7+  4.14.90-v7+

DrSchottky, 2019-01-10:
Seems that you haven't installed raspberrypi-kernel-headers

BlueSun99, 2019-01-15:
@DrSchottky Of course I installed ALL dependencies in README file :-|

marrobHD, 2019-01-16:
same problem

lennycampino, 2019-01-26:
Any solution for this?

lennycampino, 2019-01-26:
https://github.com/seemoo-lab/nexmon/issues/274 this helped me.

marrobHD, 2019-01-29:
Not working for me

DrSchottky, 2019-01-30:
- Make sure you're compiling brcmfmac for your kernel (see #274)
- Make sure /lib/modules/<your kernel>/build isn't empty
And I don't know if the supplied brcmfmac compiles against 4.14.9x. It's fine on .79