seemoo-lab/nexmon #271: rasberry pi 3 B+ make fails

irweazel, 2019-01-13:
hopefully someone can help as i would really like to get the monitor mode working.

after reading through a few of the other issues i used the script from

https://github.com/seemoo-lab/nexmon/issues/259#issuecomment-438768001

but i still get stuck with the error 

make[1]: *** /lib/modules/4.14.92-v7+/build: No such file or directory.  Stop.
Makefile:45: recipe for target 'brcmfmac.ko' failed
make: *** [brcmfmac.ko] Error 2

4.14.92-v7+ #1187 SMP Wed Jan 9 20:01:54 GMT 2019 armv7l GNU/Linux
PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
 
thanks

ajlennon, 2019-01-14:
I did and put some notes here - 

https://github.com/dynamicdevices/nexmon-binaries

It's quite temperamental and doen't seem to like Txing packets over the mon interface

irweazel, 2019-01-16:
thanks for the reply, thats what i am noticing. I have skipped out the install process but using Kali-pi which has it already on there but i am finding it stops working after a short period - (tcpdump monitoring for SSID on a specific channel) and i end up having to reboot it to sort it out. 
Be good to try and get to the bottom of it on the Pi 3 b+