seemoo-lab/nexmon #272: bcm43430a1 vs bcm43438

ThatWeirdAndrew, 2019-01-13:
I have a raspberry pi 3 b and I'm trying to build nexmon. I noticed that the chart listed the wifi chip as the bcm43430a1, but every other website says it has the bcm43438. What's the difference?

matthiasseemoo, 2019-01-14:
If you run strings on the original firmware file, you will see that it is a
firmware for the bc43430, hence, I renamed the chip to reflect this chip
number.

-- 
Matthias Schulz
Secure Mobile Networking Lab - SEEMOO

Email: matthias.schulz@seemoo.tu-darmstadt.de
Web: http://www.seemoo.de/mschulz
Phone (new): +49 6151 16-25478    Fax: +49 6151 16-25471

Department of Computer Science
Center for Advanced Security Research Darmstadt
Technische Universität Darmstadt
Mornewegstr. 32 (Office 4.2.10, Building S4/14)
D-64293 Darmstadt, Germany

ThatWeirdAndrew, 2019-01-14:
Why is every other website listing it as the bcm43438?