seemoo-lab/nexmon #273: Find definitions.mk offsets for new FW release

DrSchottky, 2019-01-14:
Hi,
I'm trying to add nexmon support for a new release of bcm43455c0 (RPi 3B+) fw, hoping it fixes the stability issues of 7.45.154.
I dumped the clean ROM and I'm working on definitions.mk, but I'm struggling to understand where to get the variuos offsets. 
I managed to find FP_CONFIG_ORIGBASE/END but, for example, i don't know where too look for ROMSTART and ROMSIZE (or UCODE/TEMPLATE ones). Is there a procedure (i.e. structures to parse in ramfile) or I have to guess based on already existing definition/ramfile?

Thank you

DrSchottky, 2019-01-28:
Found them and ported nexmon. 
Unfortunately it didn't fix crash issues.

closed 2019-01-28 by DrSchottky

joshuarubin, 2019-10-29:
Can you explain how you found these? I'm trying to get a firmware and module built for rpi zero w buster with kernel `4.19.75+` and firmware version `7.45.98.83` (device `bcm43430a1`)

DrSchottky, 2019-10-29:
Sone values are per-chip, others per-fw.
For the latter, load your target ramfile and a ramfile for an already reversed fw into a disassembler and try to find the new values.

Btw despite having ported all the offsets I haven't managed to get a working monitor mode on 7.45.98.83