seemoo-lab/nexmon #278: How to analyse bcm43455c0-sdio.bin and rom.bin

sommeilboy, 2019-01-24:
I've extracted rom.bin from the WIFI chip, but I don't kown whether it is so-called clean. I have tried to applied flashpatches to the ROM file using the fpext utility but failed. So I use IDApro to decompile ROM file, it seems to be not always right. Could you tell me some details about the analysis process?(BTW. How to get the start address of ucode? I tried to disassemble the Ucode file by b43-dasm but it didn't work. )

Thank you very much.

DrSchottky, 2019-01-25:
According to what I read the rom.bin dumped with the extractor FW is the clean one.
Once patched with fpext it should be merged with ramfile to get the complete FW that can be reversed with IDA.
Same questions about ucode, I'm still struggling to figure out offsets...
I'm working on 43455c0 too, if you want you can reach out me to share info and do tests

sommeilboy, 2019-01-26:
I have found the start address of ucode about 43455c0. But it is not same for 43430a1. As for 43455c0, ucode: 0x21d0b0-0x198000=0x850b0. Then you can find it in bcm43455c0-sdio.bin, just like this. 
 
[image]

But I don't understand how he get the definitions.mk and how to merge by fpext. 

[image]

I tried but probably used the wrong parameters, so it didn't work.
 
Would you mind telling me  more details about the merging process? 

Thank you very much.

DrSchottky, 2019-01-26:
AFAIK some addresses are fixed for a chipset, others depends on fw version.
If you are on 7.45.154 all the offset you need are in the definitions.mk file provided, including the ones related to ucode. If, like me,  you are porting nexmon (#273 ) some offsets should be the same, some are in the ramfile and some others should be guessed in ways I still don't know :)

Flashpatches are already extracted when you run make in nexmon root folder, btw the command for extraction is
`fpext -r bcm43455c0-sdio.bin -s RAMSTART -b FP_CONFIG_ORIGBASE FP_CONFIG_ORIGEND`

and 
`fpext -r bcm43455c0-sdio.bin -s RAMSTART -b FP_CONFIG_ORIGBASE -e FP_CONFIG_ORIGEND -i rom.clean.bin -o rom.bin -t ROMSTART`
to patch a clean ROM file

Merging is just a file made of RAM+ROM with  bcm43455c0-sdio.bin starting at  RAMSTART and rom.bin starting at ROMSTART

Anyway, what's your goal?

sommeilboy, 2019-02-14:
Thank you very much.

What I am concerned about is how to reverse engineer the firmware to analyze the running process and functions of the firmware so that I can apply this method to other chips or firmware.

DrSchottky, 2019-02-16:
Porting a higher firmware (there're plenty of them form 45455) could be a good exercise

Jessman101, 2019-04-30:
Hi,
It looks like you've been able to dump the ROM for the RPI3B+ if I'm not mistaken? 
Are you able to give me insight on how you've been able to do this?
Thanks,
Jessie.

DrSchottky, 2019-04-30:
> Hi,
> It looks like you've been able to dump the ROM for the RPI3B+ if I'm not mistaken?
> Are you able to give me insight on how you've been able to do this?
> Thanks,
> Jessie.

Hi, there's a rom-extraction driver in 7_45_154 folder. 
Compile and load it then run `make dump` to create a dump file.

Jessman101, 2019-04-30:
> > Hi,
> > It looks like you've been able to dump the ROM for the RPI3B+ if I'm not mistaken?
> > Are you able to give me insight on how you've been able to do this?
> > Thanks,
> > Jessie.
> 
> Hi, there's a rom-extraction driver in 7_45_154 folder.
> Compile and load it then run `make dump` to create a dump file.

Thanks for you reply - I did see that directory and when compiling, realized that it compiles for the kernel version 4.9 (currently compiling Nexmon with 4.14.98-v7+). I'm assuming that I'll need to rework the Makefile to incorporate 4.14.y or downgrade the kernel?
Thanks,
Jessie.

DrSchottky, 2019-04-30:
> Thanks for you reply - I did see that directory and when compiling, realized that it compiles for the kernel version 4.9 (currently compiling Nexmon with 4.14.98-v7+). I'm assuming that I'll need to [...]
I did it with a 4.9 kernel

Jessman101, 2019-05-01:
> I did it with a 4.9 kernel

Did the kernel version you happen to use be 4.9.80-v7+? That's what I downgraded too.

I seem to be getting "/lib/modules/4.9.80-v7+: no such file or directory" error when I run make in the 7_45_154/rom_extraction folder. I'm stumped why I would be getting this error.

AsmitaJha08, 2020-06-04:
Hello @DrSchottky 
I am too working on using rpi 3B+. I extracted the rom the same way from 7_45_154/rom_extraction folder, by doing some modification in Makefile for Linux Kernel 4.19. There is no error while rom extraction. But the strings on the extracted rom.bin file is giving something like the screenshot attached below.

[image]

Any suggestions on where the issue could be? Thank you.

Regards
Asmita

AsmitaJha08, 2020-06-04:
Hi ,
Issue resolved,[this solution]( https://github.com/seemoo-lab/nexmon/issues/67#issuecomment-290686658) helped