seemoo-lab/nexmon #282: Get raw frames Raspberry Pi 3B+

Felipe31, 2019-02-07:
How would be possible to collect raw frames arriving at wlan0 working as an AP? 

As I understand, since monitor mode turns on the promiscuous mode, I would only need that if I wanted to get frames with destinations to other devices. But I want to get raw frames that are destined to my actual device.

Is it possible to make a patch to get that information and to send it to the user environment somehow? Maybe using ftrace kernel debugger options or send the info to the driver and use ftrace then.

I know [issue #92](https://github.com/seemoo-lab/nexmon/issues/92) is quite related. That's why I don't want to use AP and monitor mode at the same time.

Felipe31, 2019-02-08:
@matthiasseemoo could you, please, be more specific? Do you mean I should write a different patch using nexmon? Could you give me more details?

I did patch the Raspberry Pi 3B+ with nexmon and I added the monitor interface, but right now I'm being able to connect and use properly the RPi as an AP.

matthiasseemoo, 2019-02-08:
A different firmware file is normally loaded to operate the wifi chip in ap
mode. So if you need monitor mode and ap operation, you need to reverse
engineer and patch the ap firmware.