seemoo-lab/nexmon #293: S8+ help

Coding-Cyanide, 2019-03-14:
Im sorry, im short on time right now... here is what i have for you:

If I try to run source setup_env.sh I get a error: "Platform not supported!"

Any help please!!! I really want monitor mode!
Thanks so much, I really appreciate you!
Cyanide

matthiasseemoo, 2019-03-15:
What is your buildsystem? Did you follow all instructions from the readme?

Coding-Cyanide, 2019-03-15:
What do you mean by buildsystem and yes, I followed the read me carefully

matthiasseemoo, 2019-03-15:
Which ubuntu version on which processor? Did you grep for the erreo message
in the build files to figure out where it is generated?

Coding-Cyanide, 2019-03-15:
Linux version 4.4.16-12542406
Processor msm8998

No I didn't I'm new to this so bear with me. :)

Thanks so much!!

matthiasseemoo, 2019-03-15:
I meant the processor of your buildsystem. However, it will not be a quick
solution and monitor mode support for the s8 is currently only
experimental. So you will need some time to get it running and understand
how it works.

Coding-Cyanide, 2019-03-15:
Can I get some help through the process? I really appreciate your help!

Coding-Cyanide, 2019-03-15:
Bump (sorry I'm very impatient)

closed 2020-09-17 by Coding-Cyanide