seemoo-lab/nexmon #294: nexutil command option list

pengchengNJ, 2019-03-19:
I'm wondering how to find a comprehensive list of nexutil command. I think the nexutil is a powerful tool to communicate with chip firmware, but it's only a blackbox without clear and comprehensive documentation. If you have summarized the option list, could you share it? Thanks.

davidrozen76, 2019-03-21:
nexutil --help?

Coding-Cyanide, 2019-04-29:
I second @davidrozen76's answer.... nexutil --help is the command to use... If you can't do that for whatever reason, here is the commands from the nexutil.c file:

    {"interface-name", 'I', "CHAR", 0, "Set interface name (default: wlan0)"},
    {"monitor", 'm', "INT", OPTION_ARG_OPTIONAL, "Set/Get monitor mode"},
    {"promisc", 'p', "INT", OPTION_ARG_OPTIONAL, "Set/Get promiscuous mode"},
    {"scansuppress", 'c', "INT", OPTION_ARG_OPTIONAL, "Set/Get scan suppress setting to avoid scanning"},
    {"disassociate", 'd', 0, 0, "Disassociate from access point"},
    {"get-custom-cmd", 'g', "INT", 0, "Get custom command, e.g. 107 for WLC_GET_VAR"},
    {"set-custom-cmd", 's', "INT", 0, "Set custom command, e.g. 108 for WLC_SET_VAR"},
    {"custom-cmd-buf-len", 'l', "INT", 0, "Custom command buffer length (default: 4)"},
    {"custom-cmd-value", 'v', "CHAR/INT", 0, "Initialization value for the buffer used by custom command"},
    {"custom-cmd-value-int", 'i', 0, 0, "Define that custom-cmd-value should be interpreted as integer"},
    {"custom-cmd-value-base64", 'b', 0, 0, "Define that custom-cmd-value should be interpreted as base64 string"},
    {"base64-output", 'R', 0, 0, "Write base64 encoded strings to stdout instead of hex dumping"},
    {"raw-output", 'r', 0, 0, "Write raw output to stdout instead of hex dumping"},
    {"dump-wl_cnt", 'w', 0, 0, "Dump WL counters"},
    {"dump-objmem", 'o', "INT", 0, "Dumps objmem at addr INT"},
    {"chanspec", 'k', "CHAR/INT", OPTION_ARG_OPTIONAL, "Set chanspec either as integer (e.g., 0x1001, set -i) or as string (e.g., 64/80)."},
    {"security-cookie", 'x', "INT", OPTION_ARG_OPTIONAL, "Set/Get security cookie"},
    {"use-udp-tunneling", 'X', "INT", 0, "Use UDP tunneling with security cookie INT"},
    {"broadcast-ip", 'B', "CHAR", 0, "Broadcast IP to use for UDP tunneling (default: 192.168.222.255)"},
    {"revinfo", 'V', 0, 0, "Dump revision information of the Wi-Fi chip"},

kiciek, 2021-09-28:
I have stumbled on this issue looking for description of command codes used with `g` and `s` parameters. If anyone next come here in search of it, they are located in `nexmon\patches\include\nexioctls.h` file. Here is an excerpt :
```
#define IOCTL_ERROR                     -23
#define IOCTL_SUCCESS                     0

// IOCTLs used by Nexmon
#define NEX_GET_CAPABILITIES            400
#define NEX_WRITE_TO_CONSOLE            401
#define NEX_CT_EXPERIMENTS              402
#define NEX_GET_CONSOLE                 403
#define NEX_GET_PHYREG                  404
#define NEX_SET_PHYREG                  405
#define NEX_READ_OBJMEM                 406
#define NEX_WRITE_OBJMEM                407
#define NEX_INJECT_FRAME                408
#define NEX_PRINT_TIMERS                409
#define NEX_GET_SECURITYCOOKIE          410
#define NEX_SET_SECURITYCOOKIE          411
#define NEX_GET_WL_CNT                  412
#define NEX_GET_VERSION_STRING          413
#define NEX_TEST_ARGPRINTF              414
#define NEX_GET_RSPEC_OVERRIDE          415
#define NEX_SET_RSPEC_OVERRIDE          416
#define NEX_CLEAR_CONSOLE               417
#define NEX_GET_CHANSPEC_OVERRIDE       418
#define NEX_SET_CHANSPEC_OVERRIDE       419
#define NEX_GET_AMPDU_TX				420
#define NEX_SET_AMPDU_TX				421
#define NEX_TRIGGER_EVENT               422
#define NEX_TRIGGER_TDLS_DISCOVER       423
#define NEX_TRIGGER_TDLS_SETUP          424
#define NEX_TRIGGER_TDLS_TEARDOWN       425
#define NEX_WRITE_TEMPLATE_RAM          426
#define NEX_SDR_START_TRANSMISSION      427
#define NEX_SDR_STOP_TRANSMISSION       428

// IOCTLs used in original firmware
#define WLC_GET_MAGIC                     0
#define WLC_GET_VERSION                   1
#define WLC_UP                            2
#define WLC_DOWN                          3
#define WLC_GET_LOOP                      4
#define WLC_SET_LOOP                      5
#define WLC_DUMP                          6
```
```
#define WLC_GET_VAR                     262     /* get value of named variable */
#define WLC_SET_VAR                     263     /* set named variable to value */
```

```