seemoo-lab/nexmon #295: Pi 3b scanner is stopped suddenly after a while

sun9133, 2019-03-26:
My Wi-Fi scanner was developed by Nexmon (https://github.com/seemoo-lab/nexmon) based on Raspberry Pi 3B. During capturing the Wi-Fi probes, the scanner sometimes is stopped. The stop time is Mar 26 12:32:15 because we haven't had any data since then. 

In fact, when I see the log file of the Pi, there are many warning like this: Could you understand that kinds of warning..? Or my python code makes that the wlan0 of Pi is turned off and on repeatly for changing wifi channel (I want to get the packet with hopping channel). Is it too burden for the Pi??....

> Mar 26 12:32:15 monitoring-pi kernel: [126742.472373] python3: page allocation stalls for 10060ms, order:1,
>  mode:0x15080c0(GFP_KERNEL_ACCOUNT|__GFP_ZERO), nodemask=(null)
> Mar 26 12:32:15 monitoring-pi kernel: [126742.472401] python3 cpuset=/ mems_allowed=0
> Mar 26 12:32:15 monitoring-pi kernel: [126742.472422] CPU: 0 PID: 530 Comm: python3 Tainted: G         C O
>    4.14.79-v7+ #1159
> Mar 26 12:32:15 monitoring-pi kernel: [126742.472425] Hardware name: BCM2835
> Mar 26 12:32:15 monitoring-pi kernel: [126742.472448] [<8010ff30>] (unwind_backtrace) from [<8010c174>] (sh
> ow_stack+0x20/0x24)
> Mar 26 12:32:15 monitoring-pi kernel: [126742.472460] [<8010c174>] (show_stack) from [<80788fa4>] (dump_sta
> ck+0xd4/0x118)
> Mar 26 12:32:15 monitoring-pi kernel: [126742.472473] [<80788fa4>] (dump_stack) from [<80228fc8>] (warn_all
> oc+0xcc/0x17c)
> Mar 26 12:32:15 monitoring-pi kernel: [126742.472483] [<80228fc8>] (warn_alloc) from [<80229dbc>] (__alloc_
> pages_nodemask+0xc90/0x11d8)
> Mar 26 12:32:15 monitoring-pi kernel: [126742.472493] [<80229dbc>] (__alloc_pages_nodemask) from [<8011b398
> >] (copy_process.part.5+0xec/0x1858)
> Mar 26 12:32:15 monitoring-pi kernel: [126742.472500] [<8011b398>] (copy_process.part.5) from [<8011cc94>]
> (_do_fork+0xc8/0x408)
> Mar 26 12:32:15 monitoring-pi kernel: [126742.472507] [<8011cc94>] (_do_fork) from [<8011d0f8>] (SyS_clone+
> 0x34/0x3c)
> Mar 26 12:32:15 monitoring-pi kernel: [126742.472516] [<8011d0f8>] (SyS_clone) from [<80108000>] (ret_fast_
> syscall+0x0/0x28)
> Mar 26 12:32:15 monitoring-pi kernel: [126742.472520] Mem-Info:
> Mar 26 12:32:15 monitoring-pi kernel: [126742.472532] active_anon:112178 inactive_anon:112151 isolated_anon
> :32
> Mar 26 12:32:15 monitoring-pi kernel: [126742.472532]  active_file:458 inactive_file:504 isolated_file:0
> Mar 26 12:32:15 monitoring-pi kernel: [126742.472532]  unevictable:0 dirty:4 writeback:0 unstable:0
> Mar 26 12:32:15 monitoring-pi kernel: [126742.472532]  slab_reclaimable:1571 slab_unreclaimable:2173
> Mar 26 12:32:15 monitoring-pi kernel: [126742.472532]  mapped:1269 shmem:193 pagetables:626 bounce:0
> Mar 26 12:32:15 monitoring-pi kernel: [126742.472532]  free:5003 free_pcp:0 free_cma:1696
> Mar 26 12:32:15 monitoring-pi kernel: [126742.472540] Node 0 active_anon:448712kB inactive_anon:448604kB ac
> tive_file:1832kB inactive_file:2016kB unevictable:0kB isolated(anon):128kB isolated(file):0kB mapped:5076kB
>  dirty:16kB writeback:0kB shmem:772kB writeback_tmp:0kB unstable:0kB all_unreclaimable? no
> Mar 26 12:32:15 monitoring-pi kernel: [126742.472552] Normal free:20012kB min:16384kB low:20480kB high:2457
> 6kB active_anon:448664kB inactive_anon:448608kB active_file:1832kB inactive_file:2016kB unevictable:0kB wri
> tepending:16kB present:970752kB managed:949448kB mlocked:0kB kernel_stack:800kB pagetables:2504kB bounce:0k
> B free_pcp:0kB local_pcp:0kB free_cma:6784kB
> Mar 26 12:32:15 monitoring-pi kernel: [126742.472554] lowmem_reserve[]: 0 0
> Mar 26 12:32:15 monitoring-pi kernel: [126742.472562] Normal: 281*4kB (UMEC) 307*8kB (UMEC) 169*16kB (UMEC)
>  35*32kB (UMEC) 19*64kB (UMEC) 7*128kB (UEC) 3*256kB (U) 1*512kB (E) 1*1024kB (E) 2*2048kB (EC) 1*4096kB (C
> ) = 20012kB
> Mar 26 12:32:15 monitoring-pi kernel: [126742.472611] 1176 total pagecache pages
> Mar 26 12:32:15 monitoring-pi kernel: [126742.472615] 0 pages in swap cache
> Mar 26 12:32:15 monitoring-pi kernel: [126742.472618] Swap cache stats: add 25725, delete 25725, find 223/2
> 97
> Mar 26 12:32:15 monitoring-pi kernel: [126742.472622] Free swap  = 4kB
> Mar 26 12:32:15 monitoring-pi kernel: [126742.472625] Total swap = 102396kB
> Mar 26 12:32:15 monitoring-pi kernel: [126742.472627] 242688 pages RAM
> Mar 26 12:32:15 monitoring-pi kernel: [126742.472630] 0 pages HighMem/MovableOnly
> Mar 26 12:32:15 monitoring-pi kernel: [126742.472632] 5326 pages reserved
> Mar 26 12:32:15 monitoring-pi kernel: [126742.472634] 2048 pages cma reserved
> Mar 26 12:32:25 monitoring-pi kernel: [126752.433790] python3: page allocation stalls for 20020ms, order:1,
>  mode:0x15080c0(GFP_KERNEL_ACCOUNT|__GFP_ZERO), nodemask=(null)
> Mar 26 12:32:25 monitoring-pi kernel: [126752.433815] python3 cpuset=/ mems_allowed=0
> Mar 26 12:32:25 monitoring-pi kernel: [126752.433829] CPU: 3 PID: 530 Comm: python3 Tainted: G         C O
>    4.14.79-v7+ #1159
> Mar 26 12:32:25 monitoring-pi kernel: [126752.433832] Hardware name: BCM2835
> Mar 26 12:32:25 monitoring-pi kernel: [126752.433854] [<8010ff30>] (unwind_backtrace) from [<8010c174>] (sh
> ow_stack+0x20/0x24)
> Mar 26 12:32:25 monitoring-pi kernel: [126752.433864] [<8010c174>] (show_stack) from [<80788fa4>] (dump_sta
> ck+0xd4/0x118)
> Mar 26 12:32:25 monitoring-pi kernel: [126752.433877] [<80788fa4>] (dump_stack) from [<80228fc8>] (warn_all
> oc+0xcc/0x17c)
> Mar 26 12:32:25 monitoring-pi kernel: [126752.433888] [<80228fc8>] (warn_alloc) from [<80229dbc>] (__alloc_
> pages_nodemask+0xc90/0x11d8)
> Mar 26 12:32:25 monitoring-pi kernel: [126752.433898] [<80229dbc>] (__alloc_pages_nodemask) from [<8011b398
[... 170 lines cut ...]
> Mar 26 12:32:37 monitoring-pi kernel: [126764.422293] [  460]     0   460     1769        0       8       0
>        89             0 bluetoothd
> Mar 26 12:32:37 monitoring-pi kernel: [126764.422299] [  483]     0   483      737        0       6       0
>       100             0 dhcpcd
> Mar 26 12:32:37 monitoring-pi kernel: [126764.422305] [  484]     0   484   254830   223717     496       0
>     22873             0 python3
> Mar 26 12:32:37 monitoring-pi kernel: [126764.422310] [  496]     0   496     1049        0       6       0
>        35             0 agetty
> Mar 26 12:32:37 monitoring-pi kernel: [126764.422316] [  499]     0   499     2553        0       8       0
>       138         -1000 sshd
> Mar 26 12:32:37 monitoring-pi kernel: [126764.422322] Out of memory: Kill process 484 (python3) score 911 o
> r sacrifice child
> Mar 26 12:32:37 monitoring-pi kernel: [126764.422338] Killed process 484 (python3) total-vm:1019320kB, anon
> -rss:894868kB, file-rss:0kB, shmem-rss:0kB
> Mar 26 12:32:37 monitoring-pi kernel: [126764.642532] oom_reaper: reaped process 484 (python3), now anon-rs
> s:0kB, file-rss:2048kB, shmem-rss:0kB
> Mar 26 12:32:37 monitoring-pi kernel: [126764.646377] device mon0 left promiscuous mode
> Mar 26 12:32:37 monitoring-pi systemd[1]: wifi_monitor.service: Main process exited, code=killed, status=9/
> KILL
> Mar 26 12:32:37 monitoring-pi systemd[1]: wifi_monitor.service: Unit entered failed state.
> Mar 26 12:32:37 monitoring-pi systemd[1]: wifi_monitor.service: Failed with result 'signal'.
> Mar 26 12:32:51 monitoring-pi kernel: [126778.151918] Under-voltage detected! (0x00050005)
> Mar 26 12:32:57 monitoring-pi kernel: [126784.391911] Voltage normalised (0x00000000)
> Mar 26 12:34:51 monitoring-pi kernel: [126898.791909] Under-voltage detected! (0x00050005)
> Mar 26 12:34:55 monitoring-pi kernel: [126902.951937] Voltage normalised (0x00000000)

DrSchottky, 2019-03-27:
You're running out of memory and OOM killer kicks in.
Not related to nexmon, check your code

spadejack014, 2019-07-01:
Sir, I cannot run it on my B3. My device is Raspberry Pi B3 , and my system is Raspbian Kernel 4.19.
Terminal always displays operation not supported, when I try to run 'sudo iw phy `iw dev wlan0 info | gawk '/wiphy/ {printf "phy" $2}'` interface add mon0 type monitor'.
I'm sure I followed the steps in the README.
Could you told me how you did that?

maryannhorton, 2019-11-06:
spadejack014. did you resolve the problem? I'm having the same issue.
I cross-compiled 43430a1 on a Pi 3B+ and copied it over to the Pi 0, installing in /lib/firmware/brcm/brcmfmac43430-sdio.bin, then I rebooted. I'm not sure about the insmod - I copied the brcmfmac.ko file over to the pi0 but insmod says it's in the wrong format.

DrSchottky, 2019-11-06:
@maryannhorton to cross-compile the driver you have to change headers path in Makefile to point at the ones for your target