seemoo-lab/nexmon #299: Problem building libisl.so.10

seyo-IV, 2019-04-11:
My error

pi@raspberrypi:~/nexmon/buildtools/isl-0.10 $ sudo make                          make  all-recursive                                                              make[1]: Entering directory '/home/pi/nexmon/buildtools/isl-0.10'                Making all in .                                                                  make[2]: Entering directory '/home/pi/nexmon/buildtools/isl-0.10'                  CC     libisl_la-mp_get_memory_functions.lo                                    mp_get_memory_functions.c: In function '__gmp_get_memory_functions':             mp_get_memory_functions.c:9:21: error: '__gmp_allocate_func' undeclared (first use in this function)                                                                 *alloc_func_ptr = __gmp_allocate_func;                                                             ^~~~~~~~~~~~~~~~~~~                                         mp_get_memory_functions.c:9:21: note: each undeclared identifier is reported only once for each function it appears in                                            mp_get_memory_functions.c:11:23: error: '__gmp_reallocate_func' undeclared (first use in this function)                                                              *realloc_func_ptr = __gmp_reallocate_func;                                                           ^~~~~~~~~~~~~~~~~~~~~                                     mp_get_memory_functions.c:13:20: error: '__gmp_free_func' undeclared (first use in this function)                                                                    *free_func_ptr = __gmp_free_func;                                                                 ^~~~~~~~~~~~~~~                                              make[2]: *** [Makefile:1001: libisl_la-mp_get_memory_functions.lo] Error 1       make[2]: Leaving directory '/home/pi/nexmon/buildtools/isl-0.10'                 make[1]: *** [Makefile:1676: all-recursive] Error 1                              make[1]: Leaving directory '/home/pi/nexmon/buildtools/isl-0.10'                 make: *** [Makefile:701: all] Error 2

I'm using a raspberry pi 3+ with raspbian

closed 2019-04-24 by seyo-IV

kammirzazad, 2019-09-25:
@seyo-IV how did you solve this problem?

seyo-IV, 2019-09-26:
I didn't.
I get the image of Kali Linux provided by offensive security with nexmon preinstalled.

aveao, 2020-01-16:
Install libgmp *before* you try to compile.

Doing a clean compilation after installing the package (or just installing it before you try to compile in the first place) fixes this issue.