seemoo-lab/nexmon #300: Make error

seyo-IV, 2019-04-18:
root@kali:~/nexmon/patches/bcm4358/7_112_300_14_sta/nexmon/# make

OLLECTING STATISTICS read /root/nexmon/STATISTICS.md for more information
  COMPILING src/ioctl.c => obj/ioctl.o (details: log/compiler.log)
/bin/sh: 1: /root/nexmon/buildtools/gcc-arm-none-eabi-5_4-2016q2-osx/bin/arm-none-eabi-gcc: Exec format error
make: *** [Makefile:46: obj/ioctl.o] Error 2

Error Log says:
gen/nexmon.pre: No such file or directory

seyo-IV, 2019-04-21:
As mentioned I managed to install the nexmon app. No I have the problem that I can't install the nexutil and proxy. Even tho my system is writable... Strange is it worked on Oreo 8.1.1 but doesn't on 8.0.0.

On 8.1.1 I couldn't get the monitor mode working properly. So many problems...

seyo-IV, 2019-04-24:
Tried the same on a PC running Ubuntu. Same error: gen/nexmon.pre: No such file or directory

matthiasseemoo, 2019-04-24:
Which processor architecture do you have on your pc?

seyo-IV, 2019-04-25:
Can someone provide me the binary for bcm4358
7.112.200.17 so I have to just run make install firmware

seyo-IV, 2019-04-25:
I got a x86_64

closed 2019-04-25 by seyo-IV

Intelectus-mss, 2020-03-31:
error: 
   REMOUTING /vendor
mount: '/vedor' not in /proc/mounts
make: *** [Makefile:163: install-firmware] Error 1

on Nexus 6 Linegeos 16