seemoo-lab/nexmon #301: Can't install untils in any directory

seyo-IV, 2019-04-25:
I'm trying to install nexutil on a Nexus 6p Oreo stock. But I can't. I have already mounted system as rw but it didn't help.

seyo-IV, 2019-04-28:
Can someone provide the libs and bins?
Only nexutil, rawproxyreverse, rawproxy and libfakeioctl. Plz

Coding-Cyanide, 2019-04-29:
When I get home I'll provide those tools! I don't have access to my Linux ATM. Ill be home around 4:00 P.M. PST... See you then!

Cyanide

Coding-Cyanide, 2019-04-29:
Scratch that it'll probably be more like 6:00 PM PST

seyo-IV, 2019-04-30:
Well I have to wait then. But better than nothing I guess😅

Coding-Cyanide, 2019-04-30:
> Can someone provide the libs and bins?
> Only nexutil, rawproxyreverse, rawproxy and libfakeioctl. Plz

Sorry for the extremely late reply. I got extremely busy and wasn't able to use Linux at the time.
There was a 64 bit and 32 bit nexutil, I built both of them for you although I believe your phone is 64 bit. Just in case though the 3
[seyo-IV-nexmon-tools.zip](https://github.com/seemoo-lab/nexmon/files/3133503/seyo-IV-nexmon-tools.zip)
2 bit is named 'nexutil32' and the 64 bit is called 'nexutil64', rename the one you need to 'nexutil'. If the 64 bit version or the 32 bit version doesn't work, try to use the other version. If that doesn't work... Honestly, I have no clue.

In the case you didn't know:
All files in the zip that have the .'so' (shared library) extension go to /system/lib
All files in the zip that have no extension go to /system/bin (I believe this is the default) or /system/sbin or /system/xbin

If anything is inaccurate, let me know!
Good luck and enjoy!

~Cyanide

seyo-IV, 2019-05-01:
Thanks mate :)

seyo-IV, 2019-05-01:
Just a Tipp if someone else need this make: chmod +x so the tools are executable

seyo-IV, 2019-05-01:
Can you also provide the iwconfig tools I'm having trouble setting the monitor mode from the Kali shell. And I can't use iwconfig outside the Kali shell. So just to test if monitor mode does really work I need these tools. If it works I would need all the rest of the tools. I hope don't ask for too much.

Coding-Cyanide, 2019-05-01:
Sure, I will be home soon.

~Cyanide

seyo-IV, 2019-05-09:
dude u forget about me?^^

Coding-Cyanide, 2019-05-09:
Sorry, i have been extremely busy lately...ill let you know shen i have iwconfig!

seyo-IV, 2019-05-21:
@matthiasseemoo what im doing wrong? I have installed the patch and the tools. The tool were installed manualy. When i use nexutil -m2 wlan0 stays in mode mnaged. What am i missing?

matthiasseemoo, 2019-05-21:
Libnexmon.so?

seyo-IV, 2019-05-21:
no i have only the old libfakeioctl.so. Since i couldn't install the tools from the app i had to rely on other people providing me the tools.

seyo-IV, 2019-05-21:
@matthiasseemoo hi, there its me again, i have build the tools myself where are they located after i run make?

matthiasseemoo, 2019-05-21:
You can always take a look into the Makefiles of the tools. There you find
the path of the built binaries. There is also a make install command that
installs the tools on your smartphone.

For example:
https://github.com/seemoo-lab/nexmon/blob/master/utilities/dhdutil/Makefile

seyo-IV, 2019-07-04:
okay, I managed to install the tools with a Linux machine and NDK installed.

closed 2019-07-04 by seyo-IV