seemoo-lab/nexmon #302: Add support for 4.19

paultag, 2019-05-25:
The next release of Debian (raspbian testing) is using Kernel 4.19 - it would be great to get support for the 4.19 kernel so when the next stable release comes out, existing installs can keep working.

Thanks for all your hard work!

paultag, 2019-05-26:
After further tracking down the kernel version change, turns out Raspberry Pi stable's apt repos now have 4.19. It's possible to downgrade by using `rpi-update` (e.g. `sudo rpi-update a08ece3d48c3c40bf1b501772af9933249c11c5b`), but it does mean this repo won't work out of the box on fully upgraded raspberry pis running the stock image.

Thanks again for your hard work!

DrSchottky, 2019-05-30:
I added an experimental 4.19.y support to my fork (PR already sent!)

GGGunrunner, 2019-06-24:
DrSchottky, beg your pardon, how to use your fork correctly?

I'm on Pi3 with 4.19.29
Tried Re4son modules, they've installed properly, and the monitor mode _appears_ to be working (via aireplay-ng --test)......... but running wifite or other tools throws a whole bunch of deassoc errors and fails..... something is wrong somewhere, so i'd like to try your patch.

=============
thats the output when the module is installed:
airmon-ng check kill

Killing these processes:

  PID Name
  553 dhcpcd
  575 wpa_supplicant
 1238 wpa_supplicant

root@localhost:~# airmon-ng start wlan0

PHY     Interface       Driver          Chipset

phy0    wlan0           brcmfmac        Broadcom 43430

                (mac80211 monitor mode vif enabled for [phy0]wlan0 on [phy0]wlan0mon)
command failed: Unknown error 524 (-524)
                (mac80211 station mode vif disabled for [phy0]wlan0)

root@localhost:~# iwconfig 
wlan0mon  IEEE 802.11  Mode:Monitor  Frequency:2.457 GHz  Tx-Power=31 dBm   
          Retry short limit:7   RTS thr:off   Fragment thr:off
          Power Management:on

DrSchottky, 2019-06-24:
@GGGunrunner [here](https://github.com/DrSchottky/nexmon)'s my fork.
Follow the instructions about how to enable monitor mode (`iw phy etc etc`)
If you get memory error while compiling the driver follow use the fix described[here](https://github.com/seemoo-lab/nexmon/pull/292)

GGGunrunner, 2019-06-26:
Flawless victory :) Compiled, installed. Injection, aireplay tests are working. There's still an issue with wifite2, I'm debugging it and if I find anything relevant to the driver, I'll post my findings.
Thanks for your work!

jenfredwell, 2019-07-11:
I'm also struggling to downgrade the kernel to 4.14 and get bcm43430a1 working again...

@DrSchottky I have tried to clone your fork for the 4.19 kernel but have been unsuccessful in building the firmware for bcm43430a1 for that version as well.  Do I just compile from master?

Thanks!

DrSchottky, 2019-07-11:
@jenfredwell patches were removed from my fork, I need to refactor them. 
You can try bcm43455c0's 4.19 driver.
Downgrading to 4.14 should be fine as long as headers match

jenfredwell, 2019-07-12:
@DrSchottky by try bcm43455c0's driver, do you mean build that and rename it to use as the bcm43430a1 driver?

DrSchottky, 2019-07-12:
@jenfredwell build also bcm43455c0 and use its brcmfmac.ko instead of bcm43430a1's one

jenfredwell, 2019-08-07:
@DrSchottky Apologies, I forgot to follow up and say that is working!  Unrelated, but it does fill up the syslog file really quickly with 

`Aug  7 17:00:33 raspberrypi kernel: [ 1176.167709] brcmfmac: nexmon_nl_ioctl_handler: NEXMON: nexmon_nl_ioctl_handler: Exit
Aug  7 17:00:33 raspberrypi kernel: [ 1176.428878] brcmfmac: nexmon_nl_ioctl_handler: NEXMON: nexmon_nl_ioctl_handler: Enter
Aug  7 17:00:33 raspberrypi kernel: [ 1176.428892] brcmfmac: nexmon_nl_ioctl_handler: NEXMON: nexmon_nl_ioctl_handler: 0058454e 32 48
Aug  7 17:00:33 raspberrypi kernel: [ 1176.428924] brcmfmac: nexmon_nl_ioctl_handler: NEXMON: nexmon_nl_ioctl_handler: calling brcmf_fil_cmd_data_set, cmd: 263
Aug  7 17:00:33 raspberrypi kernel: [ 1176.451242] brcmfmac: nexmon_nl_ioctl_handler: NEXMON: nexmon_nl_ioctl_handler: Exit
Aug  7 17:00:34 raspberrypi kernel: [ 1176.713509] brcmfmac: nexmon_nl_ioctl_handler: NEXMON: nexmon_nl_ioctl_handler: Enter`

Is there anyway to disable or filter those?

FlexMcMurphy, 2019-12-15:
Hello,

I would also like to install Nexmon on my Pi 3B running Raspbian Kernel 4.19.66-v7+
My Wifi chip is: bcm43430a1, the driver is brcmfmac and version num: 7.45.98.38

Is there any chance of updating Nexmon to support Kernel 4.19 on the Pi3B?
Otherwise could someone please give me the steps to install the Nexmon firmware and drivers to my Pi 3B with Kernel 4.19 using this repository?

After reading around I'm thinking it might be better for me to just install the Re4son Kernel..

Any advice much appreciated :)

Flex

gwbres, 2020-05-26:
> I added an experimental 4.19.y support to my fork (PR already sent!)

I was about to add support for 4.19 but I found your message just on time

you did exactly what I needed, thank you very much!! 
**I can't believe your patch has still not been merged, seriously...**
 
I am far from being a nexmon tool fan, their makefiles are dirty, they don't support the use of an external cross-compiler, which is the context I always work with, so I have to make dirty patches.

Just captured this from my RPi0w:

`# iw phy0 info
Wiphy phy0
	[...]
	Supported interface modes:
		 * [..]
		 * monitor <----- :)
		 * [...]

airmon-ng start wlan0 >> back on track :)
`