seemoo-lab/nexmon #305: nexus 6P nexmon installation pre-compiled or instructions.

felbutss, 2019-06-27:
$50-$100 USD bounty through PayPal!!!!!!!!!!!!!!!!!!!!

I'm seriously offering this for anyone who can help me set this  device up. I'm honestly over the wasted hours and hours.

I have a Nexus 6P with....
-android 8.0.
-magisk.
-twrp.
-nethunter - full

I can't get 'Hijacker' or 'wifite to run using the internal WiFi chipset. When I try everything just starts crashing/killed.  (Same story with nexus 7 with a external adapter)

Whoever has the answer. Reply to this post or email me direct for any questions. jasonchahine85@gmail.com

Real money.

seyo-IV, 2019-07-08:
Get a nice x86_64 Linux system(Like Ubuntu 18.04). Install it on your PC or USB Stick. You can use [LiLI](https://www.linuxliveusb.com/) for that. Once started run `df -h` to see if there is enough space, cuz sometimes it's decomposed in 4GB blocks which sucks. It is not enough space for the installation!
Start the machine and open Firefox.
Download [Android NDK](https://dl.google.com/android/repository/android-ndk-r11c-linux-x86_64.zip?hl=de)
Extract the archive, open the folder(mostly that is Downloads), right-click inside the folder window and open a new terminal there, then cd(`cd android-ndk-r11c-linux-x86_64`), inside the archive move it to /opt (`sudo mv android-ndk-r11c /opt`).

Now run this:
```
export ANDROID_NDK=/opt/android-ndk-r11c
export PATH=$PATH:$ANDROID_NDK
export NDK_ROOT=/opt/android-ndk-r11c
```
Now, in the terminal window run `apt-add-repository universe`
Then run:
```
sudo dpkg --add-architecture i386
sudo apt-get update
sudo apt-get install libc6:i386 libncurses5:i386 libstdc++6:i386
```
and then run:
`sudo apt-get install git gawk qpdf adb flex bison file make`

Now run: `cd ~`
There clone the repo with: `git clone https://github.com/seemoo-lab/nexmon.git`
Now `cd nexmon` and run `source setup_env.sh` then run  `make`. Now, wait.
After its finished `cd /patches/bcm4358/7_112_300_14_sta/nexmon`. Run `make` again.

Now test the connectivity to ur smartphone, run `adb devices` if its showing error, select PTP instead of Charging in the USB Settings(jsut pull down with your finger on the display and change it there).

Now run `make install-firmware` or backup it first with `make backup-firmware`
You have to permit access of the shell on the phone, there must be a root access popup. Allow it.
Now you are done with the firmware part.

Let's go to the tools.
Run  `cd ~/nexmon/utilities` , then run `make`. Now you have to wait for a while.

Then `cd` to the tool folder for example nexutil: `cd nexutil` and run `make install`, note not every tool has the install option, just try them out. When I remember right there were, nexutil, the both proxys, dhdutil, iw, wireless tools, libfakeioctl, libnexmon, netcat.

If you get the following error: Ressource is busy, you have to modify the Makefiles in each folder and remove this line -> `shell something mount- o rw,remount /system`. So run nano Makefile, go to the line and press ctrl + k. Then ctrl +x and enter.

Hope i could help. I still don't get this running(have an open issue here :C ) maybe you have more luck than me.

felbutss, 2019-07-09:
Mate. I love you

seyo-IV, 2019-07-09:
Also there is an extra image for the Nexus 6p here https://www.offensive-security.com/kali-linux-nethunter-download/ 
Angler Oreo
;) With slot of apps preinstalled and working!

felbutss, 2019-07-11:
mate. that deserves another tip... lol

libertyordeath76, 2019-12-28:
did you guys get this working? I have a 6P running nougat 7.1 and still cant install any nexmon tools and the nexmon app keeps crashing even after I push the "grant root, permissions & check installation" button.  When I click on Nexmon tools in the app and select any tool like socat, for example, it gives me errors saying it cant install to system/bin or xbin.  I ordered a regular Nexus 6 and might be sending the 6P back because this is annoying.  I even installed Nethunter for Nexus nougat off of the offensive sec website.  It seems people are having better luck with the Nexus 5's and regular 6's from what I have been reading.  Was wondering if you guys had any luck over the last few months.

kimocoder, 2019-12-28:
I've got support for Nexus 6P :)

"git clone https://github.com/kimocoder/nexmon"

Simple.

kimocoder, 2019-12-28:
Take a look below :+1:  I don't want any money btw. Open source ftw!

https://forum.xda-developers.com/nexus-6p/development/kernel-cypher-v3-0-t4023075

libertyordeath76, 2019-12-28:
Thank you I will try it out. My main goal was to use the internal chip for monitor mode, not an external adapter. But I don't see anyone having luck with that so I might be better off just using my TP link.

kimocoder, 2019-12-29:
... I use the internal.. djises.

git clone https://githib.com/kimocoder/nexmon

Then build and transfer the utilities..

"LD_PRELOAD=libnexmon.so airodump-ng wlan0"

How hard can it be? My Makefile support arm64 (v8a), the Nexmon repo does not. Simple. God read some README's

libertyordeath76, 2020-01-08:
Kimo, I get an error saying that it cant be preloaded and it says failed to initialize wlan0.  I am new to Nethunter so your help is much appreciated because I am still learning this stuff.

kimocoder, 2020-01-08:
[image]

[image]

Ising external adapter with Hijacker
[image]

kimocoder, 2020-01-08:
wifite2 on external adapter
[image]

[image]

[image]

archer322, 2020-01-08:
https://drive.google.com/file/d/1_kgCj9SsVno1ISUq0CJCRTjD2B-Fzbxa/view?usp=drivesdk

https://www.paypal.me/FYX3R

Anyone else who needs this I'd appreciate a few bucks but if you're tight
on cash it's no biggie (^.^)

kimocoder, 2020-01-08:
Simply got to much on my hands atm

kimocoder, 2020-01-08:
Notice me in a week or two, I'll give you step by step

libertyordeath76, 2020-01-08:
Thanks Kimo, we all appreciate your hard work and help. 

Archer, what exactly did you post there?

libertyordeath76, 2020-01-08:
All of this stuff works on my Nexus 5 without problems but my 6P has been giving me nothing but problems lol.

libertyordeath76, 2020-01-20:
This is probably completely stupid and pointless information but the Anker allows you to run both a 2.4ghz and 5ghz adapters on the same cable and its recognized by the Nexus 6p. I run airodump in 2 different terminals one for each adapter at the same time and its been extremely stable. No idea what motivated me to try this 😂. Panda Pau07 and 09 both work great and so does the Tplink. 

[image]

[image]
[image]
[image]

libertyordeath76, 2020-01-20:
Kimo, I am still having trouble building nexmon.  If you have time please help me with a step by step.

kimocoder, 2020-01-20:
@libertyordeath76 which Anker is this? Tell me and I'll help you

libertyordeath76, 2020-01-20:
AK-848061054910

[image]

libertyordeath76, 2020-01-20:
I would like to try the 7-in-One model next and possibly add a 3rd adapter. My battery isnt taking as much of a hit as I thought it would from running both.

kimocoder, 2020-01-20:
Download the old NDK version 11c from
https://developer.android.com/ndk/downloads

Then, set NDK_ROOT to the NDK directory...
Do "git clone https://github.com/kimocoder/nexmon" and "cd nexmon" and do "./setup_env.sh"

kimocoder, 2020-01-20:
OOOH! Wait a bit!! uploading more Nexus fixes!

haroonalhadisk, 2020-01-21:
hi thanks for the libraries @dragarok 
does th wifi injection work ? as i'm getting   __nex_driver_io:  error
also the iwconfig shows the wlan0 as managed and not monitored. what am i doing wrong?

dragarok, 2020-01-22:
@haroonalhadisk You need to turn on wifi manually. Else it will prompt driver errror. 
Yes injection works though it says managed for wlan0. Hijacker app is having problem in saving handshake. You can test injection and airodump working in Hijacker from my repo if you follow my instructions.

haroonalhadisk, 2020-01-26:
@dragarok hi I tried again by ifconfig wlan0 down and then setting to monitor mode using iwconfig. It runs but injection seems to not work . Can you please tell me how to turn on WiFi manually.

kimocoder, 2020-01-26:
Stop using ifconfig and iwconfig, they are deprecated. Use 'ip' and 'iw' instead.. wonder how? look at my README.md at https://GitHub.com/aircrack-ng/rtl8812au

vanphilip70, 2020-01-31:
@dragarok Can you please explain further. I downloaded your compiled nexmon and I see 3 different folders in lib with the libnexmon.so file (arm64-v8a, armeabi and armeabi-v7a). 

> "Then copy the libnexmon files both 32 bit and 64 bit and put it in /system/lib and lib64" 

So I put the libnexmon.so from arm64-v8a in the /system/lib64 folder and now I have 2 left. Which one do I have to put in /system/lib? I also replaced the firmware but I just cant get internal wifi to work. Thanks in advance

Keskebeu, 2020-09-06:
I have built and run this tool on my many nexus 5 no problem. But running it in nexus 6p is a b... Injections sometime work and some time not. Can't really find out why. Switching channels works in nexus 5 but 6p does not. I am able to do monitor mode all the time. The injection seems to work when it feels like.

hh55g, 2021-07-27:
Hi Ian need to enable monitor mode on Nexus 6p any one can help