seemoo-lab/nexmon #308: Nexus 6P Android 8 stock cant activate monitor mode

seyo-IV, 2019-07-04:
I have nexmon firmware and tool installed on my nexus 6P.
When i do nexutil -m2, iwconfig say wlan0 is still managed.
When i do export LD_PRELOAD=libnexmon.so and try to use airodump, for example, it says couldn't set monitor mode for wlan0.

I did everything that was in the instructions and it still doesn't work....

felbutss, 2019-07-09:
I send that..

seyo-IV, 2019-07-12:
I finally managed to get it working. Somehow when I do nexutil -m2 wlan0 seems to be in managed mode, but it is not. when I run LD_PRELOAD=libfakeioctl.so airodump -i wlan0 it did work. But this doesn't work in the kali shell. but does in android su shell.

closed 2019-07-12 by seyo-IV

yesimxev, 2020-07-28:
As for NH terminal, you should do `export LD_PRELOAD=/system/lib64/kalilibfakeioctl.so`, but that's included in monstart-nh