seemoo-lab/nexmon #309: MPFR doesn't build on Raspbian

DrSchottky, 2019-07-06:
MPFR doesn't build on latest Raspbian since it doesn't find aclocal
```
root@raspberrypi:/home/pi/nexmon/buildtools/mpfr-3.1.4# make
CDPATH="${ZSH_VERSION+.}:" && cd . && /bin/bash /home/pi/nexmon/buildtools/mpfr-                                                                                        3.1.4/missing aclocal-1.15 -I m4
/home/pi/nexmon/buildtools/mpfr-3.1.4/missing: line 81: aclocal-1.15: command not found                                                                                        t found
WARNING: 'aclocal-1.15' is missing on your system.
         You should only need it if you modified 'acinclude.m4' or
         'configure.ac' or m4 files included by 'configure.ac'.
         The 'aclocal' program is part of the GNU Automake package:
         <http://www.gnu.org/software/automake>
         It also requires GNU Autoconf, GNU m4 and Perl in order to run:
         <http://www.gnu.org/software/autoconf>
         <http://www.gnu.org/software/m4/>
         <http://www.perl.org/>
make: *** [Makefile:425: aclocal.m4] Error 127
```
GhettoFix (tested): softlink it to the preinstalled mpfr4.

`sudo ln -s /usr/lib/arm-linux-gnueabihf/libmpfr.so.6 /usr/lib/arm-linux-gnueabihf/libmpfr.so.4
`

matthiasseemoo, 2019-07-07:
Which Raspbian version are you using? I tried it with the latest release
and it was working.

DrSchottky, 2019-07-07:
Latest release with up-to-date pkgs.
I'll check it again, if you say it worked for you maybe I messed up something.

DrSchottky, 2019-07-07:
Booted a fresh install and followed the instructions step by step: can confirm it throws that error.

matthiasseemoo, 2019-07-07:
Here is what I did on Raspbian:

`root@raspberrypi:/home/pi/mpfr-3.1.4# apt update
Get:1 http://raspbian.raspberrypi.org/raspbian buster InRelease [15.0 kB]
Get:2 http://archive.raspberrypi.org/debian buster InRelease [25.1 kB]
Get:3 http://raspbian.raspberrypi.org/raspbian buster/main armhf Packages [13.0 MB]
Get:4 http://archive.raspberrypi.org/debian buster/main armhf Packages [205 kB]
Fetched 13.2 MB in 8s (1566 kB/s)                                                                                      
Reading package lists... Done
Building dependency tree       
Reading state information... Done
All packages are up to date.
root@raspberrypi:/home/pi/mpfr-3.1.4# apt upgrade
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Calculating upgrade... Done
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
root@raspberrypi:/home/pi/mpfr-3.1.4# cd ..
root@raspberrypi:/home/pi# rm -rf mpfr-3.1.4
root@raspberrypi:/home/pi# tar xfvz mpfr-3.1.4.tar.gz 
mpfr-3.1.4/
mpfr-3.1.4/configure.ac
mpfr-3.1.4/AUTHORS
mpfr-3.1.4/aclocal.m4
mpfr-3.1.4/VERSION
mpfr-3.1.4/ltmain.sh
mpfr-3.1.4/Makefile.in
mpfr-3.1.4/Makefile.am
mpfr-3.1.4/PATCHES
mpfr-3.1.4/test-driver
mpfr-3.1.4/m4/
mpfr-3.1.4/m4/lt~obsolete.m4
mpfr-3.1.4/m4/size_max.m4
mpfr-3.1.4/m4/ltversion.m4
mpfr-3.1.4/m4/ltoptions.m4
mpfr-3.1.4/m4/libtool.m4
mpfr-3.1.4/m4/ltsugar.m4
mpfr-3.1.4/tune/
mpfr-3.1.4/tune/Makefile.in
mpfr-3.1.4/tune/Makefile.am
mpfr-3.1.4/tune/speed.c
mpfr-3.1.4/tune/tuneup.c
mpfr-3.1.4/tune/bidimensional_sample.c
mpfr-3.1.4/COPYING
mpfr-3.1.4/config.guess
mpfr-3.1.4/BUGS
mpfr-3.1.4/missing
mpfr-3.1.4/TODO
mpfr-3.1.4/src/
mpfr-3.1.4/src/fits_sint.c
mpfr-3.1.4/src/cmp2.c
mpfr-3.1.4/src/div_d.c
mpfr-3.1.4/src/csc.c
mpfr-3.1.4/src/get_f.c
mpfr-3.1.4/src/gen_inverse.h
mpfr-3.1.4/src/setmax.c
mpfr-3.1.4/src/fits_slong.c
mpfr-3.1.4/src/set_sj.c
mpfr-3.1.4/src/inp_str.c
mpfr-3.1.4/src/cmp_abs.c
mpfr-3.1.4/src/ui_div.c
mpfr-3.1.4/src/get_ld.c
mpfr-3.1.4/src/mul_d.c
mpfr-3.1.4/src/strtofr.c
mpfr-3.1.4/src/set_z_exp.c
mpfr-3.1.4/src/isinf.c
mpfr-3.1.4/src/exceptions.c
mpfr-3.1.4/src/set_d64.c
mpfr-3.1.4/src/ieee_floats.h
[... 940 lines cut ...]
/bin/bash ../libtool  --tag=CC   --mode=compile gcc -DTIME_WITH_SYS_TIME=1 -DHAVE_INTTYPES_H=1 -DHAVE_STDINT_H=1 -DHAVE_LOCALE_H=1 -DHAVE_WCHAR_H=1 -DHAVE_STDARG=1 -DHAVE_SYS_TIME_H=1 -DHAVE_STRUCT_LCONV_DECIMAL_POINT=1 -DHAVE_STRUCT_LCONV_THOUSANDS_SEP=1 -DHAVE_ALLOCA_H=1 -DHAVE_STDINT_H=1 -DHAVE_VA_COPY=1 -DHAVE_SETLOCALE=1 -DHAVE_GETTIMEOFDAY=1 -DHAVE_LONG_LONG=1 -DHAVE_INTMAX_T=1 -DMPFR_HAVE_INTMAX_MAX=1 -DMPFR_HAVE_FESETROUND=1 -DHAVE_DENORMS=1 -DHAVE_SIGNEDZ=1 -DHAVE_ROUND=1 -DHAVE_TRUNC=1 -DHAVE_FLOOR=1 -DHAVE_CEIL=1 -DHAVE_NEARBYINT=1 -DMPFR_USE_THREAD_SAFE=1 -DMPFR_USE_C11_THREAD_SAFE=1 -DHAVE_CLOCK_GETTIME=1 -DLT_OBJDIR=\".libs/\" -DHAVE_ATTRIBUTE_MODE=1 -DHAVE___GMPN_ROOTREM=1 -DHAVE___GMPN_SBPI1_DIVAPPR_Q=1 -I.     -Wall -Wmissing-prototypes -Wpointer-arith -g -O2 -MT pow_si.lo -MD -MP -MF .deps/pow_si.Tpo -c -o pow_si.lo pow_si.c
libtool: compile:  gcc -DTIME_WITH_SYS_TIME=1 -DHAVE_INTTYPES_H=1 -DHAVE_STDINT_H=1 -DHAVE_LOCALE_H=1 -DHAVE_WCHAR_H=1 -DHAVE_STDARG=1 -DHAVE_SYS_TIME_H=1 -DHAVE_STRUCT_LCONV_DECIMAL_POINT=1 -DHAVE_STRUCT_LCONV_THOUSANDS_SEP=1 -DHAVE_ALLOCA_H=1 -DHAVE_STDINT_H=1 -DHAVE_VA_COPY=1 -DHAVE_SETLOCALE=1 -DHAVE_GETTIMEOFDAY=1 -DHAVE_LONG_LONG=1 -DHAVE_INTMAX_T=1 -DMPFR_HAVE_INTMAX_MAX=1 -DMPFR_HAVE_FESETROUND=1 -DHAVE_DENORMS=1 -DHAVE_SIGNEDZ=1 -DHAVE_ROUND=1 -DHAVE_TRUNC=1 -DHAVE_FLOOR=1 -DHAVE_CEIL=1 -DHAVE_NEARBYINT=1 -DMPFR_USE_THREAD_SAFE=1 -DMPFR_USE_C11_THREAD_SAFE=1 -DHAVE_CLOCK_GETTIME=1 -DLT_OBJDIR=\".libs/\" -DHAVE_ATTRIBUTE_MODE=1 -DHAVE___GMPN_ROOTREM=1 -DHAVE___GMPN_SBPI1_DIVAPPR_Q=1 -I. -Wall -Wmissing-prototypes -Wpointer-arith -g -O2 -MT pow_si.lo -MD -MP -MF .deps/pow_si.Tpo -c pow_si.c  -fPIC -DPIC -o .libs/pow_si.o
../libtool: line 1751: warning: setlocale: LC_CTYPE: cannot change locale (UTF-8)
libtool: compile:  gcc -DTIME_WITH_SYS_TIME=1 -DHAVE_INTTYPES_H=1 -DHAVE_STDINT_H=1 -DHAVE_LOCALE_H=1 -DHAVE_WCHAR_H=1 -DHAVE_STDARG=1 -DHAVE_SYS_TIME_H=1 -DHAVE_STRUCT_LCONV_DECIMAL_POINT=1 -DHAVE_STRUCT_LCONV_THOUSANDS_SEP=1 -DHAVE_ALLOCA_H=1 -DHAVE_STDINT_H=1 -DHAVE_VA_COPY=1 -DHAVE_SETLOCALE=1 -DHAVE_GETTIMEOFDAY=1 -DHAVE_LONG_LONG=1 -DHAVE_INTMAX_T=1 -DMPFR_HAVE_INTMAX_MAX=1 -DMPFR_HAVE_FESETROUND=1 -DHAVE_DENORMS=1 -DHAVE_SIGNEDZ=1 -DHAVE_ROUND=1 -DHAVE_TRUNC=1 -DHAVE_FLOOR=1 -DHAVE_CEIL=1 -DHAVE_NEARBYINT=1 -DMPFR_USE_THREAD_SAFE=1 -DMPFR_USE_C11_THREAD_SAFE=1 -DHAVE_CLOCK_GETTIME=1 -DLT_OBJDIR=\".libs/\" -DHAVE_ATTRIBUTE_MODE=1 -DHAVE___GMPN_ROOTREM=1 -DHAVE___GMPN_SBPI1_DIVAPPR_Q=1 -I. -Wall -Wmissing-prototypes -Wpointer-arith -g -O2 -MT pow_si.lo -MD -MP -MF .deps/pow_si.Tpo -c pow_si.c -o pow_si.o >/dev/null 2>&1
...
.libs/gammaonethird.o .libs/grandom.o   -lgmp  -g -O2   -Wl,-soname -Wl,libmpfr.so.4 -o .libs/libmpfr.so.4.1.4
libtool: link: (cd ".libs" && rm -f "libmpfr.so.4" && ln -s "libmpfr.so.4.1.4" "libmpfr.so.4")
libtool: link: (cd ".libs" && rm -f "libmpfr.so" && ln -s "libmpfr.so.4.1.4" "libmpfr.so")
libtool: link: ar cru .libs/libmpfr.a  exceptions.o extract.o uceil_exp2.o uceil_log2.o ufloor_log2.o add.o add1.o add_ui.o agm.o clear.o cmp.o cmp_abs.o cmp_si.o cmp_ui.o comparisons.o div_2exp.o div_2si.o div_2ui.o div.o div_ui.o dump.o eq.o exp10.o exp2.o exp3.o exp.o frac.o frexp.o get_d.o get_exp.o get_str.o init.o inp_str.o isinteger.o isinf.o isnan.o isnum.o const_log2.o log.o modf.o  [... cut ...]
make[1]: Leaving directory '/home/pi/mpfr-3.1.4/src'
Making all in tests
make[1]: Entering directory '/home/pi/mpfr-3.1.4/tests'
make[1]: Nothing to be done for 'all'.
make[1]: Leaving directory '/home/pi/mpfr-3.1.4/tests'
Making all in tune
make[1]: Entering directory '/home/pi/mpfr-3.1.4/tune'
make[1]: Nothing to be done for 'all'.
make[1]: Leaving directory '/home/pi/mpfr-3.1.4/tune'
make[1]: Entering directory '/home/pi/mpfr-3.1.4'
make[1]: Nothing to be done for 'all-am'.
make[1]: Leaving directory '/home/pi/mpfr-3.1.4'`

DrSchottky, 2019-07-08:
Source from release archive builds fine, the one from nexmon/buildtools doesn't.
If I do
```
tar xzvf mpfr-3.1.4.tar.gz
rm -rf nexmon/buildtools/mpfr-3.1.4
mv mpfr-3.1.4/nexmon/buildtools/
cd nexmon/buildtools/mpfr-3.1.4
./configure
make
```
it works

But to be honest I don't understand why

matthiasseemoo, 2019-07-08:
Maybe git did not collect all necessary files.

DrSchottky, 2019-07-08:
I cloned the repo at least 5 times on 3 clean systems, and even
`diff -rq mpfr-3.1.4 nexmon/buildtools/mpfr-3.1.4/`
shows no differences

j0j0j0, 2019-07-27:
Maybe https://stackoverflow.com/questions/33278928/how-to-overcome-aclocal-1-15-is-missing-on-your-system-warning/33286344#33286344 helps you.

DrSchottky, 2019-07-29:
@j0j0j0 thank you, I can confirm it solves the issue.

And I add that these pkgs are required to build mpfr:

`sudo apt-get install automake autoconf libtool texinfo`

closed 2019-07-29 by DrSchottky