seemoo-lab/nexmon #316: Monitor mode scans 5GHz by default on Galaxy S7 stock 7.0

dlupy77, 2019-08-03:
Hello,
I try to run latest version of Nexmon on my Galaxy S7 Nougat, stock kernel and firmware. I use nexutil -m1 to enter monitor mode but when I use airodump-ng only 5GHz is scanned. The only way that I found to scan both 2.4 and 5GHz bands is to try to connect to any nearby WiFi with a wrong password when airodump-ng is running. I tried using airodump-ng wlan0 -b bg but with no results, only 5GHz is scanned. Does anybody have a workaround to scan both bands excepting my discovered empirical method (trial to connect nearby network with wrong password)? Does anybody knows the reason of why this happens and why -b by option doesn't have any effect in Nexmon airodump-ng binary? Thanks😁

dlupy77, 2019-08-04:
Discovered that using nexutil -k6 scans 2,4GHz band. Is there any possibility to scan both 2,4 and 5GHz simultaneously?
 I use 9.96.4 firmware, stock kernel, Android 7.0. Please give me some hints. Thanks.

matthiasseemoo, 2019-08-04:
Do you observe the same behavior with the original, non-patched firmware?

dlupy77, 2019-08-04:
The behaviour is with the unmatched firmware. Are there any chances to work if I replace original firmware with patched firmware? I keep the original firmware because I want to keep client functionality for the WiFi card and I just need WiFi scanning, I don't need injection or other functions. Thanks for your fast reply.

matthiasseemoo, 2019-08-04:
So you mean, it works with the unpatched firmware?

Does it work with the patched firmware if you do not activate monitor mode?
Normally it should. Does the patched firmware version equal the version of
your original firmware?

dlupy77, 2019-08-04:
I tried only with original (unpatched) 9.96

- monitor mode starts ok with nexutil -m1
- when run LD_PRELOAD=/path/libnexmon.so airodump-ng wlan0 only 5GHz band is scanned
- if, while scanning 5GHz, I try to connect to random WiFi network with a wrong password than access points and clients from 2.4 Gaz are visible and scanning works in both bands
- if I stop monitor mode nexutil -m0 and start it again scanning works for both bands
- if I switch off the WiFi and turn monitor mode on again scanning works again just for the 5GHz band
- if I run nexutil -k6 before nexutil -m1 scanning works just for 2,4GHz band

I didn't try nexmon patched firmware. What is the correct path where I should push the patched firmware? Are there any normal functions affected by using nexmon patched firmware for Galaxy S7 or Galaxy S7 Edge?

Thank you again for your patience.😀

matthiasseemoo, 2019-08-04:
The S7 has a wifi chip that can operate in two wifi bands in parallel.
maybe this is the reason why you can only scan in one band at a time.

-- 
Matthias Schulz
Secure Mobile Networking Lab - SEEMOO

Email: matthias.schulz@seemoo.tu-darmstadt.de
Web: http://www.seemoo.de/mschulz
Phone (new): +49 6151 16-25478    Fax: +49 6151 16-25471

Department of Computer Science
Center for Advanced Security Research Darmstadt
Technische Universität Darmstadt
Mornewegstr. 32 (Office 4.2.10, Building S4/14)
D-64293 Darmstadt, Germany

davidrozen76, 2019-08-06:
I've encountered the same problem using airodump on Galaxy S7.

I have patched and compiled that latest airodump for aarch64.

The trick is that intead of the normal method of channel switching in airodump, I have changed the function to run nexutil -

`/* Execute nexutil to really hop between channels */
                                char command[1024];
                                snprintf(command, sizeof(command), "nexutil -k%i > /dev/null", ch);
                                system(command);
 `

This way, airodump really hops between channels, both in 2.4GHz and 5GHz.

yweiss0, 2019-08-07:
Hi guys tell me please any of you managed to make it work on galaxy s7 with packet injection and get a full wpa/wpa2 handshake capture?
If yes I would like to know what should I do