seemoo-lab/nexmon #319: Unable to set monitor mode on Raspberry Pi Zero W with Stretch

JK2210, 2019-08-16:
I follow all steps from readme, but when I try to start monitor mode I get the error: `command failed: Operation not supported (-95)`.
I used this repo in July following the steps from @DrSchottky (#292).
Now with the newer Raspbian Stretch an the current repo it doesn't work.
I build `/usr/lib/arm-linux-gnueabihf/libisl.so.10`. `/usr/lib/arm-linux-gnueabihf/libmpfr.so.4`already exists on my system.

Readme:
> Install the patched firmware on your RPI3

I'm not sure, but this does also apply for Zero?

DrSchottky, 2019-08-16:
Forget about my fork: clone the official one and build brcmfmac driver from bcm43455c0 folder. I can confirm it works on 0W.
@matthiasseemoo what about placing brcmfmac in patches/common instead of a copy for each firmware?

JK2210, 2019-08-16:
I know, I used the current repo from seemoo-lab, not yours. But I mean, that your fork worked for me in June.

DrSchottky, 2019-08-16:
Probably you aren't loading the right driver. Post install steps/logs

JK2210, 2019-08-16:
```
root:/home/pi/nexmon# source setup_env.sh
root:/home/pi/nexmon# make
root:/home/pi/nexmon# cd patches/bcm43430a1/7_45_41_46/nexmon/
root:/home/pi/nexmon/patches/bcm43430a1/7_45_41_46/nexmon# make
  COLLECTING STATISTICS read /home/pi/nexmon/STATISTICS.md for more information
  COMPILING src/ioctl.c => obj/ioctl.o (details: log/compiler.log)
  COMPILING src/autostart.c => obj/autostart.o (details: log/compiler.log)
  COMPILING src/injection.c => obj/injection.o (details: log/compiler.log)
  COMPILING src/patch.c => obj/patch.o (details: log/compiler.log)
  COMPILING src/version.c => obj/version.o (details: log/compiler.log)
  COMPILING src/local_wrapper.c => obj/local_wrapper.o (details: log/compiler.log)
  COMPILING src/sendframe.c => obj/sendframe.o (details: log/compiler.log)
  COMPILING src/monitormode.c => obj/monitormode.o (details: log/compiler.log)
  COPYING UCODE /home/pi/nexmon/firmwares/bcm43430a1/7_45_41_46/ucode.bin => gen/ucode.bin
  COMPRESSING UCODE gen/ucode.bin => gen/ucode_compressed.bin
  GENERATING C FILE gen/ucode_compressed.bin => src/ucode_compressed.c
  COMPILING src/ucode_compressed.c => obj/ucode_compressed.o (details: log/compiler.log)
  GENERATING C FILE /home/pi/nexmon/firmwares/bcm43430a1/7_45_41_46/templateram.bin => src/templateram.c
  COMPILING src/templateram.c => obj/templateram.o (details: log/compiler.log)
  COMPILING /home/pi/nexmon/patches/common/wrapper.c => obj/wrapper.o (details: log/compiler.log)
  COMPILING /home/pi/nexmon/patches/common/argprintf.c => obj/argprintf.o (details: log/compiler.log)
  COMPILING /home/pi/nexmon/patches/common/helper.c => obj/helper.o (details: log/compiler.log)
  COMPILING /home/pi/nexmon/patches/common/udptunnel.c => obj/udptunnel.o (details: log/compiler.log)
  COMPILING /home/pi/nexmon/patches/common/ucode_compression_code.c => obj/ucode_compression_code.o (details: log/compiler.log)
  COMPILING /home/pi/nexmon/patches/common/objmem.c => obj/objmem.o (details: log/compiler.log)
  COMPILING /home/pi/nexmon/patches/common/securitycookie.c => obj/securitycookie.o (details: log/compiler.log)
  COMPILING /home/pi/nexmon/patches/common/radiotap.c => obj/radiotap.o (details: log/compiler.log)
  COMPILING /home/pi/nexmon/firmwares/bcm43430a1/7_45_41_46/flashpatches.c => obj/flashpatches.o (details: log/compiler.log)
  GENERATING LINKER FILE gen/nexmon.pre => gen/nexmon.ld
  GENERATING LINKER FILE gen/nexmon.pre => gen/flashpatches.ld
  GENERATING LINKER FILE gen/memory.ld
  LINKING OBJECTS => gen/patch.elf (details: log/linker.log, log/linker.err)
  GENERATING MAKE FILE gen/nexmon.pre => gen/nexmon.mk
  GENERATING MAKE FILE gen/nexmon.pre => gen/flashpatches.mk
  APPLYING FLASHPATCHES gen/flashpatches.mk => brcmfmac43430-sdio.bin (details: log/flashpatches.log)
  APPLYING PATCHES gen/nexmon.mk => brcmfmac43430-sdio.bin (details: log/patches.log)
root:/home/pi/nexmon/patches/bcm43430a1/7_45_41_46/nexmon# make install-firmware
  COLLECTING STATISTICS read /home/pi/nexmon/STATISTICS.md for more information
  COMPILING src/version.c => obj/version.o (details: log/compiler.log)
  GENERATING LINKER FILE gen/nexmon.pre => gen/nexmon.ld
  GENERATING LINKER FILE gen/nexmon.pre => gen/flashpatches.ld
  LINKING OBJECTS => gen/patch.elf (details: log/linker.log, log/linker.err)
  GENERATING MAKE FILE gen/nexmon.pre => gen/nexmon.mk
  GENERATING MAKE FILE gen/nexmon.pre => gen/flashpatches.mk
  APPLYING FLASHPATCHES gen/flashpatches.mk => brcmfmac43430-sdio.bin (details: log/flashpatches.log)
  APPLYING PATCHES gen/nexmon.mk => brcmfmac43430-sdio.bin (details: log/patches.log)
  COPYING brcmfmac43430-sdio.bin => /lib/firmware/brcm/brcmfmac43430-sdio.bin
  UNLOADING brcmfmac
  RELOADING brcmfmac
```
After that my interface wlan0 disappears.

DrSchottky, 2019-08-16:
That's because you don't compile the driver for your kernel.
After that go to /patches/bcm43455c0/7_45_189/nexmon/, build with make (w/o installing) and run insmod on brcmfmac_4.19.y-nexmon/brcmfmac.ko

JK2210, 2019-08-16:
Wait, bcm43455c0 is not on the 0w? Is bcm43143 not the right module?

DrSchottky, 2019-08-16:
No, look at the Supported Devices matrix on the homepage.
I told you to compile bcm43455c0 (and take only the driver) because there isn't a brcmfmac_4.19.y-nexmon folder under bcm43430a1 tree

JK2210, 2019-08-16:
Oh, sorry, I my mistake.
Where is this step in the readme?

Nice, it works!
Thank you!

closed 2019-08-16 by JK2210

DrSchottky, 2019-08-16:
It isn't in the readme since it's not "official".
I could send a PR with just the folder copied inside, but I asked the author about a more elegant solution.

matthiasseemoo, 2019-08-16:
Hi DrSchottky, I think that it is a good idea to have a common drivers
directory, however I would not put it into the patches/common directory but
instead create a new drivers directory in the repo's root and put the
drivers there. Then in the patches' Makefiles one can decide against which
of the common drivers to build. We should however make sure that we do not
break any compatibily by updating a driver in the common deivers directory.
Additionally, I would prefer to have the nexmon patches also in the
kernel's brcmfmac sources.

domjiang, 2020-02-28:
Hi DrSchottky and JK2210, i am trying to make the driver in patches/bcm43455c0/7_45_189/nexmon/. However, the error message says it is missing definitions.mk in the folder firmwares/bcm43455c0/7_45_189/. I checked the folder and the file is in the exact folder.

TonyU1234, 2020-03-30:
Sorry, I follow the instructions from your readme, from this thread and from https://pimylifeup.com/raspberry-pi-nexmon/
I use the kali-linux 2020 arm image for the raspberry pi zero w and the kernel 4.19.93-Re4son+. But I still get the same error when use the command 'make':
'Clock skew detected. Your build may be incomplete.
/bin/bash: /root/nexmon/build tools/gcc-arm-none-eabi-5_4-2016q2-linux-armv71/bin/arm-none-eabi-gcc: No such file or directory
make: *** [Makefile:63: obj/ioctl.o] Error 127

The directory is there but I can't go inside to take a look. And I don't know what I miss here.