seemoo-lab/nexmon #321: Pi 4 Buster 4.19.58-v7l+ Re4son

rootorben, 2019-08-20:
Hi....
i bought a Pi4 4GB RAM and installed a fresh Raspbian Setup

My kernel
Linux pi4 4.19.58-v7l+ #1245 SMP Fri Jul 12 17:31:45 BST 2019 armv7l GNU/Linux

Now my next step would be the installation of the Re4son-Kernel to this system.....
but it actually failed :(

Here is what i did....

# ** Step 1: **
# resized SD and did all the other sudo raspi-config-shit, then rebooted

# ** Step 2: **
# system update
sudo apt-get update && sudo apt-get upgrade -y && sudo apt-get dist-upgrade && sudo reboot now

# ** Step 3: **
# build deps
sudo apt install raspberrypi-kernel-headers git libgmp3-dev gawk qpdf bison flex make

# ** Step 4: **
# cloned the repo
git clone --recursive https://github.com/seemoo-lab/nexmon /home/pi/nexmon

# ** Step 5: **
# link libisl
cd ~/nexmon/buildtools/isl-0.10 && sudo -s; ./configure && make && make install && ln -s /usr/local/lib/libisl.so /usr/lib/arm-linux-gnueabihf/libisl.so.10

# ** Step 6: **
# setup aclocal-1-15
cd /home/pi; wget https://ftp.gnu.org/gnu/automake/automake-1.15.tar.gz && tar xvf automake-1.15.tar.gz && cd automake-1.15; ./configure --prefix=/opt/aclocal-1.15; make; sudo mkdir -p /opt; sudo make install; export PATH=/opt/aclocal-1.15/bin:$PATH; cd /home/pi && sudo rm -rf automake*;

# ** Step 7: **
# link mpfr
cd /home/pi/nexmon/buildtools/mpfr-3.1.4 && ./configure && make && make install && ln -s /usr/local/lib/libmpfr.so /usr/lib/arm-linux-gnueabihf/libmpfr.so.4

# ** Step 8: **
# build and install the fw
cd /home/pi/nexmon && source setup* && make; cd patches//bcm43455c0/7_45_189/nexmon; make; # **THE ERROR APPEARS RIGHT HERE** make backup-firmware; make install-firmware

# That's about all....

# The error appears while compiling the firmware-patch in BOTH folders for the Pi4
# (7_45_154 and 7_45_189)

# errorlog
root@pi4:/home/pi/nexmon/patches/bcm43455c0/7_45_189/nexmon# make

  ###########   ###########   ##########    ##########           
         ############  ############  ############  ############          
         ##            ##            ##   ##   ##  ##        ##          
         ##            ##            ##   ##   ##  ##        ##          
         ###########   ####  ######  ##   ##   ##  ##    ######          
          ###########  ####  #       ##   ##   ##  ##    #    #          
                   ##  ##    ######  ##   ##   ##  ##    #    #          
                   ##  ##    #       ##   ##   ##  ##    #    #          
         ############  ##### ######  ##   ##   ##  ##### ######          
         ###########    ###########  ##   ##   ##   ##########           

            S E C U R E   M O B I L E   N E T W O R K I N G               

                               presents:                                  

              # ###   ###  #   # # ###  ###   ###  # ###                  
              ##   # #   #  # #  ##   ##   # #   # ##   #                 
              #    # #####   #   #    #    # #   # #    #                 
              #    # #      # #  #    #    # #   # #    #                 
              #    #  #### #   # #    #    #  ###  #    #                 

                The C-based Firmware Patching Framework                   

                           !!! WARNING !!!                                
    Our software may damage your hardware and may void your hardware’s    
     warranty! You use our tools at your own risk and responsibility      

  COLLECTING STATISTICS read /home/pi/nexmon/STATISTICS.md for more information
  COMPILING src/version.c => obj/version.o (details: log/compiler.log)
  GENERATING LINKER FILE gen/nexmon.pre => gen/nexmon.ld
  GENERATING LINKER FILE gen/nexmon.pre => gen/flashpatches.ld
  LINKING OBJECTS => gen/patch.elf (details: log/linker.log, log/linker.err)
  GENERATING MAKE FILE gen/nexmon.pre => gen/nexmon.mk
  GENERATING MAKE FILE gen/nexmon.pre => gen/flashpatches.mk
  APPLYING FLASHPATCHES gen/flashpatches.mk => brcmfmac43455-sdio.bin (details: log/flashpatches.log)
  APPLYING PATCHES gen/nexmon.mk => brcmfmac43455-sdio.bin (details: log/patches.log)
  BUILDING DRIVER for kernel 4.19 brcmfmac_4.19.y-nexmon/brcmfmac.ko (details: log/driver.log)
make[1]: *** /lib/modules/4.19.58-v7l+/build: Datei oder Verzeichnis nicht gefunden.  Schluss.
make: *** [Makefile:52: brcmfmac.ko] Fehler 2

matthiasseemoo, 2019-08-21:
You need to build the brcmfmac driver for the kernel you are using, if you
do not have the proper headers for your kernel, the driver will not build.

rootorben, 2019-08-21:
What does this mean, i need the brcm for the kernel iam using?
If im on a Pi3 i need to use bcm4343-whatever-folder and on a Pi4 
i have to use one of the 2 folders i named in my post above.....

So in case im on a Pi4 to compile for a Pi4 i have to install most of this crap down here, right?
 `
pi@pi4:~ $ apt-cache search raspberrypi
libavcodec-extra58 - FFmpeg library with additional de/encoders for audio/video codecs
linux-image-4.9.0-6-rpi - Linux 4.9 for RaspberryPI
linux-image-4.9.0-6-rpi2 - Linux 4.9 for RaspberryPI2
linux-image-rpi - Linux for RaspberryPI (meta-package)
linux-image-rpi2 - Linux for RaspberryPI2 (meta-package)
libraspberrypi-bin - Miscellaneous Raspberry Pi utilities
libraspberrypi-dev - EGL/GLES/OpenVG/etc. libraries for the Raspberry Pi's VideoCore IV (headers)
libraspberrypi-doc - EGL/GLES/OpenVG/etc. libraries for the Raspberry Pi's VideoCore IV (headers)
libraspberrypi0 - EGL/GLES/OpenVG/etc. libraries for the Raspberry Pi's VideoCore IV
raspberrypi-archive-keyring - GnuPG archive keys of the raspberrypi archive
raspberrypi-artwork - Copy of the Raspberry Pi logo
raspberrypi-bootloader - Raspberry Pi bootloader
raspberrypi-kernel - Raspberry Pi bootloader
raspberrypi-kernel-headers - Header files for the Raspberry Pi Linux kernel
raspberrypi-net-mods - Network configuration for the Raspberry Pi UI
raspberrypi-sys-mods - System tweaks for the Raspberry Pi
raspberrypi-ui-mods - Config to customise the LXDE UI for the Raspberry Pi
`
And on a Raspbian Pi3 i need this down here, or wtf do u mean?
`
pi@pi3:~ $ apt-cache search kalipi
kalipi-bootloader - Kali-Pi bootloader
kalipi-kernel - Kali-Pi linux kernel
kalipi-kernel-headers - Header files for the Kali-Pi Linux kernel
kalipi-re4son-firmware - Additional Kali-Pi Pi firmware
`
I cant see any of these application-names in the deps-section by the way,
please go a little further more into detail for some dumbfuck like me ;)

greets

DrSchottky, 2019-08-21:
What's the content of your `/usr/src/` folder? You should have a folder named `linux-headers-<your_kernel_version>`
Are you sure that your kernel update ended well? Latest kernel for Raspbian repo should be 4.19.66

Oh, and if you're planning to switch to another kernel (like Re4son) after compiling nexmon be careful beacuse you can't install a kernel module compiled for another kernel version

rootorben, 2019-08-23:
sorry for my late feedback

`pi@pi4:~ $ ls -l /usr/src/
insgesamt 16
drwxr-xr-x 24 root root 4096 Aug 23 14:35 linux-headers-4.19.66+
drwxr-xr-x 24 root root 4096 Aug 23 14:34 linux-headers-4.19.66-v7+
drwxr-xr-x 24 root root 4096 Aug 23 14:34 linux-headers-4.19.66-v7l+
drwxr-xr-x  3 root root 4096 Jul 10 02:14 sense-hat
`

DrSchottky, 2019-08-23:
You need the headers matching your kernel version. Update the kernel or download the proper headers

avoz2012, 2020-12-20:
I have a raspberry 3b + build Linux kali-pi 4.19.118-Re4son-v7 + # 1 SMP Wed May 6 14:25:41 AEST 2020 armv7l GNU / Linux
Are there any options for installing Burpsuite?
Portswigger developers for ARM do not build