seemoo-lab/nexmon #323: Error while make on RPI zero W

xinanhuang, 2019-08-29:
I am getting an error following the installation guide at make, it said it cannot find -ll but I'm not quite sure what that is:

+++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
  BUILDING BUILDTOOLS
make[1]: Entering directory '/home/pi/nexmon/buildtools'
  BUILDING flash_patch_extractor
make[2]: Entering directory '/home/pi/nexmon/buildtools/flash_patch_extractor'
gcc -std=c99 -Wall -Wno-unused-result -O0 -D_BSD_SOURCE -o fpext.o -c fpext.c
In file included from /usr/include/stdio.h:27:0,
                 from fpext.c:3:
/usr/include/features.h:148:3: warning: #warning "_BSD_SOURCE and _SVID_SOURCE are deprecated, use _DEFAULT_SOURCE" [-Wcpp]
 # warning "_BSD_SOURCE and _SVID_SOURCE are deprecated, use _DEFAULT_SOURCE"
   ^~~~~~~
gcc -std=c99 -Wall -O2 -Wextra -Wno-missing-field-initializers -o darm/armv7.o -c darm/armv7.c
gcc -std=c99 -Wall -O2 -Wextra -Wno-missing-field-initializers -o darm/armv7-tbl.o -c darm/armv7-tbl.c
gcc -std=c99 -Wall -O2 -Wextra -Wno-missing-field-initializers -o darm/darm.o -c darm/darm.c
gcc -std=c99 -Wall -O2 -Wextra -Wno-missing-field-initializers -o darm/darm-tbl.o -c darm/darm-tbl.c
gcc -std=c99 -Wall -O2 -Wextra -Wno-missing-field-initializers -o darm/thumb2.o -c darm/thumb2.c
gcc -std=c99 -Wall -O2 -Wextra -Wno-missing-field-initializers -o darm/thumb2-decoder.o -c darm/thumb2-decoder.c
gcc -std=c99 -Wall -O2 -Wextra -Wno-missing-field-initializers -o darm/thumb2-tbl.o -c darm/thumb2-tbl.c
gcc -std=c99 -Wall -O2 -Wextra -Wno-missing-field-initializers -o darm/thumb.o -c darm/thumb.c
gcc -std=c99 -Wall -O2 -Wextra -Wno-missing-field-initializers -o darm/thumb-tbl.o -c darm/thumb-tbl.c
gcc -o fpext fpext.o darm/armv7.o darm/armv7-tbl.o darm/darm.o darm/darm-tbl.o darm/thumb2.o darm/thumb2-decoder.o darm/thumb2-tbl.o darm/thumb.o darm/thumb-tbl.o
make[2]: Leaving directory '/home/pi/nexmon/buildtools/flash_patch_extractor'
  BUILDING ucode_extractor
make[2]: Entering directory '/home/pi/nexmon/buildtools/ucode_extractor'
gcc -std=c99 -Wall -Wno-unused-result -O0 -D_BSD_SOURCE -o ucodeext.o -c ucodeext.c
In file included from /usr/include/stdio.h:27:0,
                 from ucodeext.c:3:
/usr/include/features.h:148:3: warning: #warning "_BSD_SOURCE and _SVID_SOURCE are deprecated, use _DEFAULT_SOURCE" [-Wcpp]
 # warning "_BSD_SOURCE and _SVID_SOURCE are deprecated, use _DEFAULT_SOURCE"
   ^~~~~~~
gcc -o ucodeext ucodeext.o
make[2]: Leaving directory '/home/pi/nexmon/buildtools/ucode_extractor'
  BUILDING b43/assembler/b43-asm.bin
make[2]: Entering directory '/home/pi/nexmon/buildtools/b43/assembler'
     DEPEND   dep/util.d
     YACC     parser.c
     LEX      scanner.c
     DEPEND   dep/scanner.d
     DEPEND   dep/parser.d
     DEPEND   dep/main.d
     DEPEND   dep/initvals.d
     DEPEND   dep/args.d
     CC       obj/args.o
     CC       obj/initvals.o
     CC       obj/main.o
main.c: In function ‘is_valid_imm’:
main.c:219:13: warning: left shift of negative value [-Wshift-negative-value]
  mask = (~0 << immediate_size) & 0xFFFF;
             ^~
     CC       obj/parser.o
parser.c: In function ‘yyparse’:
parser.c:4330:18: warning: passing argument 1 of ‘yyerror’ discards ‘const’ qualifier from pointer target type [-Wdiscarded-qualifiers]
         yyerror (yymsgp);
                  ^~~~~~
parser.y:26:13: note: expected ‘char *’ but argument is of type ‘const char *’
 extern void yyerror(char *);
             ^~~~~~~
     CC       obj/scanner.o
scanner.c: In function ‘yy_get_next_buffer’:
scanner.c:1098:18: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
   for ( n = 0; n < max_size && \
                  ^
scanner.c:2051:3: note: in expansion of macro ‘YY_INPUT’
   YY_INPUT( (&YY_CURRENT_BUFFER_LVALUE->yy_ch_buf[number_to_move]),
   ^~~~~~~~
     CC       obj/util.o
     CC       b43-asm.bin
/usr/bin/ld: cannot find -ll
collect2: error: ld returned 1 exit status
Makefile:75: recipe for target 'b43-asm.bin' failed
make[2]: *** [b43-asm.bin] Error 1
make[2]: Leaving directory '/home/pi/nexmon/buildtools/b43/assembler'
Makefile:4: recipe for target 'b43/assembler/b43-asm.bin' failed
make[1]: *** [b43/assembler/b43-asm.bin] Error 2
make[1]: Leaving directory '/home/pi/nexmon/buildtools'
Makefile:8: recipe for target 'buildtools' failed
make: *** [buildtools] Error 2

DrSchottky, 2019-08-29:
Are you using Raspbian?
Did you install the required dependencies?

xinanhuang, 2019-08-29:
yup, and yes including bison and flex

DrSchottky, 2019-08-29:
That's unusual. Which version?
Try to install `libfl-dev`

egg82, 2020-06-02:
`apt install libfl-dev` works on DietPi 6.25 (Pi Zero W)