seemoo-lab/nexmon #328: libnexmon.so for kali chroot

RoninNada, 2019-09-06:
Can anyone help in building a version of libnexmon.so to work in kali nethunter chroot?
Like how libfakeioctlkali.so is? I have tried, successfully built it but it won't run.

mirzaatifbaig, 2019-09-13:
It won't run ever, The chroot runs on top of the current or existing kernel and in order to make the patches work we need to make some changes in the kernel, mainly the bcmdhd driver in the device source needs to be able to understand what the chroot is asking for, I could tell you how to make it work without the fakeioctl but I don't think I should.Go do some reading into C and compiling kernels and you would probably figure it out in 3 months or so.

JuniorJPDJ, 2019-10-16:
> It won't run ever, The chroot runs on top of the current or existing kernel and in order to make the patches work we need to make some changes in the kernel, mainly the bcmdhd driver in the device source needs to be able to understand what the chroot is asking for, I could tell you how to make it work without the fakeioctl but I don't think I should.Go do some reading into C and compiling kernels and you would probably figure it out in 3 months or so.Also Close this issue, you're embarrassing yourself.

Imo you are talking false things without any arguments. Chroot has nothing to kernel, yes you are right that it is running on the same kernel, but it changes nothing in this example.

mirzaatifbaig, 2019-10-18:
Well, I did managed to build the library and somehow managed to show up in chroot, but it always messed up, turns out there's a way to modify the Kernel's bcmdhd drivers to allow monitor mode requests, I did bare minimum research and found out the chroot used the same kernel and in order to enable monitor mode we needed an ioctl thingy to be defined in the driver separately, after that it was able to do everything the patched firmware said it to do, battery and heating issues were encountered.I would love to hear how anyone managed to get it working in a sane more detailed way but once again, I didn't search a lot of things and to save time assumed a few.

RoninNada, 2020-01-08:
This isn't to do with the kernel. Have you seen the libfakeioctlkali.so version of libfakeioctl.so? I want to build exactly lthat but with libnexmon.so? I have built a version in the amd64 nethunter chroot but it doesn't run. Any help or ideas?