seemoo-lab/nexmon #331: RPI3B compile mpfr

paverneuil, 2019-09-20:
Hello,
I'm having a problem with the compilation of MPFR under raspberry PI 3B with the buster version. During the compilation I encounter an error. I reinstall the compilation packages but nothing is done.

Would you have a track?

```
root@raspberrypi:/home/pi/nexmon/buildtools/mpfr-3.1.4# ./configure
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking whether to enable maintainer-specific portions of Makefiles... yes
checking build system type... armv7l-unknown-linux-gnueabihf
checking host system type... armv7l-unknown-linux-gnueabihf
checking for grep that handles long lines and -e... /bin/grep
checking for egrep... /bin/grep -E
checking for a sed that does not truncate output... /bin/sed
checking for gcc... /home/pi/nexmon/buildtools/gcc-arm-none-eabi-5_4-2016q2-linux-armv7l/bin/arm-none-eabi-
checking whether the C compiler works... no
configure: error: in `/home/pi/nexmon/buildtools/mpfr-3.1.4':
configure: error: C compiler cannot create executables
See `config.log' for more details
```
Have a good day

DrSchottky, 2019-09-20:
Did you run `source setup_env.sh` before that?

maryannhorton, 2019-11-05:
It would be good to test mpfr build on Buster. I got all kinds of errors. The script above is from a part of the build process before it asks you to run "source setup_env.sh".

For me, after I installed several packages it wanted that aren't in the build doc (such as aclocal and automake) I got lots of "warnings" that seem to be fatal.  After installing them, here is the current one:

root@mcbuster:/home/mah/raspbian/nexmon/buildtools/mpfr-3.1.4# make
Making all in doc
make[1]: Entering directory '/home/mah/raspbian/nexmon/buildtools/mpfr-3.1.4/doc'
restore=: && backupdir=".am$$" && \
am__cwd=`pwd` && CDPATH="${ZSH_VERSION+.}:" && cd . && \
rm -rf $backupdir && mkdir $backupdir && \
if (/bin/bash /home/mah/raspbian/nexmon/buildtools/mpfr-3.1.4/missing makeinfo --version) >/dev/null 2>&1; then \
  for f in mpfr.info mpfr.info-[0-9] mpfr.info-[0-9][0-9] mpfr.i[0-9] mpfr.i[0-9][0-9]; do \
    if test -f $f; then mv $f $backupdir; restore=mv; else :; fi; \
  done; \
else :; fi && \
cd "$am__cwd"; \
if /bin/bash /home/mah/raspbian/nexmon/buildtools/mpfr-3.1.4/missing makeinfo  --enable-encoding -I . \
 -o mpfr.info mpfr.texi; \
then \
  rc=0; \
  CDPATH="${ZSH_VERSION+.}:" && cd .; \
else \
  rc=$?; \
  CDPATH="${ZSH_VERSION+.}:" && cd . && \
  $restore $backupdir/* `echo "./mpfr.info" | sed 's|[^/]*$||'`; \
fi; \
rm -rf $backupdir; exit $rc
/home/mah/raspbian/nexmon/buildtools/mpfr-3.1.4/missing: line 81: makeinfo: command not found
WARNING: 'makeinfo' is missing on your system.
         You should only need it if you modified a '.texi' file, or
         any other file indirectly affecting the aspect of the manual.
         You might want to install the Texinfo package:
         <http://www.gnu.org/software/texinfo/>
         The spurious makeinfo call might also be the consequence of
         using a buggy 'make' (AIX, DU, IRIX), in which case you might
         want to install GNU make:
         <http://www.gnu.org/software/make/>
make[1]: *** [Makefile:375: mpfr.info] Error 127
make[1]: Leaving directory '/home/mah/raspbian/nexmon/buildtools/mpfr-3.1.4/doc'
make: *** [Makefile:468: all-recursive] Error 1

juliatempmail, 2019-11-27:
> /home/mah/raspbian/nexmon/buildtools/mpfr-3.1.4/missing: line 81: makeinfo: command not found
> WARNING: 'makeinfo' is missing on your system.

```bash
apt install texinfo
```

sarain, 2020-02-22:
I'm also experiencing the issue with "C compiler cannot create executables" while trying to configure the build for the mpfr library on a Raspberry Pi 3 B+ running Raspbian Buster with kernel 4.19.97-v7+.  As pointed out by @DrSchottky, source setup_env.sh has been run.

In the original config.log, I see it's trying to use "/home/pi/nexmon/buildtools/gcc-arm-none-eabi-5_4-2016q2-linux-armv7l/bin/arm-none-eabi-" as the gcc path and it's returning "No such file or directory"  It looks like it should be arm-none-eabi-gcc instead.

I modified setup_env.sh, adding "gcc" to the end of the path.  When I re-ran ./configure, I got the 2nd attached config.log file.  It can now execute gcc but it fails because it can't find libmpfr.so.4 while trying to compile the conftest.c file.  This is silly because that's what I'm trying to build in the first place (see line 100 of 2nd_config.log).  I poked around a bit more but couldn't find an obvious place where libmpfr.so.4 was being referenced for the test compile.

A fix or any insight into what might be going on with this would be greatly appreciated.

Here are the two config.log files
[1st_config.log](https://github.com/seemoo-lab/nexmon/files/4238995/1st_config.log)
[2nd_config.log](https://github.com/seemoo-lab/nexmon/files/4238996/2nd_config.log)

sarain, 2020-02-22:
I figured out what was going on (at least enough to fix it).
The short answer is that you **DO NOT** want to run source setup_env.sh before building libmpfr.  If you do, it will configure $CC environment variable to point to the gcc used for building firmware.  If you try and build libmpfr while that's set, it will remember the alternative gcc path even if you clear the environment variable.

When I was still having issues after rebooting and verifying the env variables were cleared I finally went in and deleted any intermediate looking files from autoconf and configure, then tried autoreconf and ./configure again.  This time they worked and used the correct system version of gcc.

It would seem that part of the autoreconf or configure process cached the path to GCC and didn't bother to check again even with the -f flag set on autoreconf.  This took way longer than I'd planned for but hopefully this helps someone else if they find themselves in a similar situation.

This might deserve a warning in the instructions too, stating that other build activities may be affected by the environment variables set by setup_env.sh

jianqingHuang, 2021-11-04:
yes ,Don't want to run source **setup_env.sh before** building libmpfr or libisl.