seemoo-lab/nexmon #332: Nexus 6/bcm4356 firmware broken

OXiDE-dev, 2019-09-24:
I'm running lineageos 16.0 / android 9
I think everything compiled and installed fine. In any case here's the sha256sum of the firmware: `7e67b3bcb1d7324287333c457abe1394ad05565acfb86cf84f9cc1e22a1ebea1  /system/vendor/firmware/fw_bcmdhd.bin`. 

mdk3:
```
 # LD_PRELOAD=libnexmon.so mdk3 wlan0 b                                      
__nex_driver_io: error
Current MAC: DB:B9:DD:E0:39:CC on Channel  4 with SSID: r|
Packets sent:      1 - Speed:    1 packets/sec__nex_driver_io: error
__nex_driver_io: error
```
```
 # LD_PRELOAD=libfakeioctl.so mdk3 wlan0 b                               
No monitor mode supported!!!
: Invalid argument
Current MAC: 70:70:72:66:8C:F7 on Channel  2 with SSID:  
Current MAC: B2:A5:8D:19:5C:18 on Channel  4 with SSID: 2o^KfmG
Current MAC: C1:B2:4D:3C:0D:43 on Channel 14 with SSID: /xY`I(3
```
iw:
```
 # iw phy phy0 interface add mon0 type monitor
command failed: Operation not supported on transport endpoint (-95)
```
ifconfig:
```
 # ifconfig wlan0
wlan0     Link encap:Ethernet  HWaddr ec:88:92:5c:cd:ab
          UP BROADCAST NOTRAILERS PROMISC ALLMULTI  MTU:1500  Metric:1
          RX packets:1468594 errors:0 dropped:16866 overruns:0 frame:0 
          TX packets:10524 errors:0 dropped:25178 overruns:0 carrier:0 
          collisions:0 txqueuelen:1000 
          RX bytes:245295953 TX bytes:829755
```

JuniorJPDJ, 2019-10-16:
Have you set interface to monitor using `nexmon` binary? You can't use normal `iw` command to set interface to monitor.

oakes8585, 2019-12-31:
@OXiDE-dev did you ever get a working patch for the nexus 6 shamu? I have one and have been looking to get it up and running on nexmon