seemoo-lab/nexmon #333: Trying to build for Nexus 6p nougat 7.1.2

Ocelot124286, 2019-09-30:
Every time I run "make" in the patches/bcm4358/7_112_201_3_sta/nexmon directory I get

sec/ioctl.c:48:21: fatal error: bcmpcie.h: No such file or directory
compilation terminated.

Are you missing a file?

TheMMcOfficial, 2019-10-03:
Did you got all the requirements?

< Install some dependencies: sudo apt-get install git gawk qpdf adb flex bison

Only necessary for x86_64 systems, install i386 libs:

sudo dpkg --add-architecture i386
sudo apt-get update
sudo apt-get install libc6:i386 libncurses5:i386 libstdc++6:i386
Clone our repository: git clone https://github.com/seemoo-lab/nexmon.git

In the root directory of the repository: cd nexmon

Setup the build environment: source setup_env.sh
Compile some build tools and extract the ucode and flashpatches from the original firmware files: make
Go to the patches folder of your target device (e.g. bcm4339 for the Nexus 5): cd patches/bcm4339/6_37_34_43/nexmon/

Compile a patched firmware: make
Generate a backup of your original firmware file: make backup-firmware
Install the patched firmware on your smartphone: make install-firmware (make sure your smartphone is connected to your machine beforehand)

> 

>

Ocelot124286, 2019-10-04:
I've installed everything. It just says its missing bcmpcie.h, bcmpcie.h isn't missing for the nexmon oreo nexus 6p build

TheMMcOfficial, 2019-10-05:
Do you have the NDK install and setup correctly ?

Ocelot124286, 2019-10-05:
I think so, can you run me through it just in case?

TheMMcOfficial, 2019-10-05:
go there and Download the NDK 11c
https://developer.android.com/ndk/downloads/older_releases

After extract it wherever you want (I extract mine in /opt) and  run does command. After you should be able to compile nexmon if you got everything to compile it.
```
export ANDROID_NDK=/opt/android-ndk-r11c
export PATH=$PATH:$ANDROID_NDK
export NDK_ROOT=/opt/android-ndk-r11c
```

Ocelot124286, 2019-10-05:
Same error, I don't know what else to do

TheMMcOfficial, 2019-10-18:
What Linux distro do you use ?

Ocelot124286, 2019-10-18:
Kali linux- port of Debian I believe

TheMMcOfficial, 2019-10-22:
I use kali 2019.2 and everything works fine I don't know what you are doing wrong...

Ocelot124286, 2019-10-22:
I will try reinstalling Kali and I'll report back. Maybe it's because I'm using x86

kimocoder, 2019-11-16:
Everything works fine

Ocelot124286, 2019-11-16:
Every reinstall I get the same error. I might try with deepin

kimocoder, 2019-11-16:
I just compiled it for Nexus 6P and running airodump-ng on it, so I know it works

Ocelot124286, 2019-11-16:
Can you send it to me? Or email me at inotreallyabot@gmail.com

kimocoder, 2019-11-16:
What do you need?

Ocelot124286, 2019-11-16:
The patched file
FW-bcmhd.bin or something like that

kimocoder, 2019-11-16:
It's in the NetHunter repo, fresh

kimocoder, 2019-11-16:
And btw, there is the NetHunter kernel for Android 8 with hid/injection/cdrom/nexmon also.

https://gitlab.com/kimocoder/kali-nethunter-devices/commit/9c9c41c74f31995f310388c7a86a41fa70141daf#f19fbdc1b8ad6bdbf478ad0f8651b9f726fa2e9b

Ocelot124286, 2019-11-16:
For Android 7.1.2