seemoo-lab/nexmon #337: Nexus 6 errors

solo-one1, 2019-10-24:
Hello,

I am having issues after I cd nexmon/patches/bcm4356/7_35_101_sta/nexmon

when I type in the make command it gives an error after the COMPILING in red, src/ioctl.c => obj/ioctl.o (details: log/compiler.log) /root/nexmon/buildtools/gcc-arm-none-eabi-5_4-2016q2-linux-armv71 bin/../lib/gcc/arm-none-eabi/5.4.1/cc1: error while loading shared libraries: libels.so.10: cannot open shared object file or directory

Because of this error I can't continue on with the next steps. Just wondering if anyone else ran in this issue.

solo-one1, 2019-10-25:
Ok I am running Pure Nexus 7.1.2 and I continue to go at this for a few more hours and I ran into this issue. I am using ubuntu as my desktop.

mount: '/vendor' not in /proc/mounts
Makefile:162: recipe for target 'install-firmware' failed
make: *** [install-firmware] Error 1

What am I missing here?

JensMiftinger, 2020-01-29:
Hi, I'm running into the same issue with my Nexus6 7.1.2.

Have you found a fix or workaround?

KR
JensMiftinger