seemoo-lab/nexmon #341: utilities build error: nexutil

trueToastedCode, 2019-11-03:
xml.etree.ElementTree.ParseError: no element found: line 2, column 0
make[2]: Entering directory `/home/lennard/Dokumente/NexmonWorkspace/nexmon/utilities/nexutil'
[arm64-v8a] Compile        : nexutil <= nexutil.c
[arm64-v8a] Compile        : nexutil <= bcmwifi_channels.c
[arm64-v8a] Compile        : nexutil <= b64-encode.c
[arm64-v8a] Compile        : nexutil <= b64-decode.c
[arm64-v8a] Executable     : nexutil
[arm64-v8a] Install        : nexutil => libs/arm64-v8a/nexutil
[x86_64] Compile        : nexutil <= nexutil.c
[x86_64] Compile        : nexutil <= bcmwifi_channels.c
[x86_64] Compile        : nexutil <= b64-encode.c
[x86_64] Compile        : nexutil <= b64-decode.c
[x86_64] Executable     : nexutil
[x86_64] Install        : nexutil => libs/x86_64/nexutil
[armeabi-v7a] Compile thumb  : nexutil <= nexutil.c
[armeabi-v7a] Compile thumb  : nexutil <= bcmwifi_channels.c
[armeabi-v7a] Compile thumb  : nexutil <= b64-encode.c
[armeabi-v7a] Compile thumb  : nexutil <= b64-decode.c
[armeabi-v7a] Executable     : nexutil
./argp-parse.c:441: error: undefined reference to '_option_is_end'
./argp-parse.c:311: error: undefined reference to '_option_is_short'
./argp-parse.c:302: error: undefined reference to '_option_is_end'
./argp-help.c:1643: error: undefined reference to 'argp_fmtstream_set_wmargin'
./argp-help.c:1657: error: undefined reference to 'argp_fmtstream_set_lmargin'
./argp-help.c:1663: error: undefined reference to 'argp_fmtstream_puts'
./argp-help.c:1675: error: undefined reference to 'argp_fmtstream_set_wmargin'
./argp-help.c:1676: error: undefined reference to 'argp_fmtstream_set_lmargin'
./argp-help.c:1678: error: undefined reference to 'argp_fmtstream_putc'
./argp-help.c:1704: error: undefined reference to 'argp_fmtstream_putc'
./argp-help.c:1111: error: undefined reference to 'argp_fmtstream_set_lmargin'
./argp-help.c:1132: error: undefined reference to 'argp_fmtstream_set_wmargin'
./argp-help.c:1134: error: undefined reference to '_option_is_short'
./argp-help.c:1140: error: undefined reference to 'argp_fmtstream_putc'
./argp-help.c:1141: error: undefined reference to 'argp_fmtstream_putc'
./argp-help.c:1174: error: undefined reference to 'argp_fmtstream_set_wmargin'
./argp-help.c:1162: error: undefined reference to 'argp_fmtstream_puts'
./argp-help.c:1189: error: undefined reference to 'argp_fmtstream_set_lmargin'
./argp-help.c:1194: error: undefined reference to '_option_is_short'
./argp-help.c:1208: error: undefined reference to 'argp_fmtstream_point'
./argp-help.c:1216: error: undefined reference to 'argp_fmtstream_puts'
./argp-help.c:1220: error: undefined reference to 'argp_fmtstream_puts'
./argp-help.c:462: error: undefined reference to '_option_is_short'
./argp-help.c:458: error: undefined reference to '_option_is_end'
./argp-help.c:497: error: undefined reference to '_option_is_end'
./argp-help.c:1460: error: undefined reference to 'argp_fmtstream_write'
./argp-help.c:1550: error: undefined reference to 'argp_fmtstream_point'
./argp-help.c:1572: error: undefined reference to 'argp_fmtstream_point'
./argp-help.c:956: error: undefined reference to 'argp_fmtstream_point'
clang++: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [local/armeabi-v7a/nexutil] Error 1
make[2]: Leaving directory `/home/lennard/Dokumente/NexmonWorkspace/nexmon/utilities/nexutil'
make[1]: *** [Makefile:12: libs/armeabi/nexutil] Fehler 2
make[1]: Verzeichnis „/home/lennard/Dokumente/NexmonWorkspace/nexmon/utilities/nexutil“ wird verlassen
make: *** [Makefile:6: nexutil/.] Fehler 2

trueToastedCode, 2019-11-03:
ok for everyone just use the Android NDK, Revision 11c, https://developer.android.com/ndk/downloads/older_releases

closed 2019-11-03 by trueToastedCode

matthiasseemoo, 2019-11-03:
Yes, sometimes it has a purpose when something is weitten in the
instructions ;-)