seemoo-lab/nexmon #344: pi@raspberrypi:~/Documents/bigdisk/nexmon/buildtools/mpfr-3.1.4 $ sudo make ISSUE

SyncMaster740B, 2019-11-07:
pi@raspberrypi:~/Documents/bigdisk/nexmon/buildtools/mpfr-3.1.4 $ sudo make
CDPATH="${ZSH_VERSION+.}:" && cd . && /bin/bash /home/pi/Documents/bigdisk/nexmon/buildtools/mpfr-3.1.4/missing aclocal-1.15 -I m4
/home/pi/Documents/bigdisk/nexmon/buildtools/mpfr-3.1.4/missing: line 81: aclocal-1.15: command not found
WARNING: 'aclocal-1.15' is missing on your system.
         You should only need it if you modified 'acinclude.m4' or
         'configure.ac' or m4 files included by 'configure.ac'.
         The 'aclocal' program is part of the GNU Automake package:
         <http://www.gnu.org/software/automake>
         It also requires GNU Autoconf, GNU m4 and Perl in order to run:
         <http://www.gnu.org/software/autoconf>
         <http://www.gnu.org/software/m4/>
         <http://www.perl.org/>
make: *** [Makefile:425: aclocal.m4] Error 127

Can someone help me? No errors before that.

alexanderavrorin, 2019-11-07:
Just got through the same issue. You need to install libtool-bin, automake and makeinfo
```
# apt install libtool-bin automake makeinfo
```
Then run  autoreconf -f -i in the mpfr directory. After that you can run `./configure` etc.

SyncMaster740B, 2019-11-07:
Well, thanks, I get that now:

For apt install makeinfo: 
pi@raspberrypi:~ $ sudo apt install makeinfo
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package makeinfo

For make: 
root@raspberrypi:/home/pi/Documents/bigdisk/nexmon/buildtools/mpfr-3.1.4# make
/bin/bash ./config.status --recheck
running CONFIG_SHELL=/bin/bash /bin/bash ./configure --no-create --no-recursion
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking whether to enable maintainer-specific portions of Makefiles... yes
checking build system type... armv7l-unknown-linux-gnueabihf
checking host system type... armv7l-unknown-linux-gnueabihf
checking for grep that handles long lines and -e... /bin/grep
checking for egrep... /bin/grep -E
checking for a sed that does not truncate output... /bin/sed
checking for CC and CFLAGS in gmp.h... yes CC= CFLAGS=
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... configure: error: in `/home/pi/Documents/bigdisk/nexmon/buildtools/mpfr-3.1.4':
configure: error: cannot run C compiled programs.
If you meant to cross compile, use `--host'.
See `config.log' for more details
make: *** [Makefile:420: config.status] Error 1

Thanks <3

alexanderavrorin, 2019-11-07:
my bad, it should've been `texinfo`.

SyncMaster740B, 2019-11-07:
Thanks again, I get now a "sudo: unable to execute ./configure: Permission denied" when I execute "sudo ./configure" in /buildtools/isl-0.10 folder...

alexanderavrorin, 2019-11-09:
If I recall correctly, there's no need to run configure as a superuser. Restart your shell, then try again.

tweigel-dev, 2020-11-18:
This should be mentioned at README of bcm4355, isn't it ?