seemoo-lab/nexmon #353: (HELP)(bcm43455)SONY Xperia Z5 Compact build

Gorkem63, 2019-12-05:
*bcm43455)SONY Xperia Z5 Compact build need*
my Android is LineageOS 14.1.Tried to 
compile firmware but got so much errors.
For the fix this issue:
1)Some helpful guys compile FW for Z5C :)
2)Teach me from the beginning
i want so much this work

Gorkem63, 2019-12-05:
@matthiasseemoo help bro

baselsayeh, 2019-12-05:
1- Run "nexmon -V" and post the result
2- Post your firmware file
3- Have you tried to the ones that are already in this repo?

Gorkem63, 2019-12-06:
> 1- Run "nexmon -V" and post the result
> 2- Post your firmware file
> 3- Have you tried to the ones that are already in this repo?

nexmon -V command not found
thats my firmware:
https://www88.zippyshare.com/v/fg4SXeFt/file.html
(fw_bcmdhd.bin)
also there are 2 fw file:
[image]

(i ipluaded the fw_bcmdhd.bin file)

yes i tries but got many errors.

baselsayeh, 2019-12-06:
@Gorkem63 
Your firmware's version is 7.45.59.16
I will take a look at it later
Can you try 7_45_77_0_hw_23_8_2017 and see if it works?

And I don't think the build scrips work on a Mac
You need a Linux (specially Ubuntu or Debian) to compile those

Gorkem63, 2019-12-06:
> @Gorkem63
> Your firmware's version is 7.45.59.16
> I will take a look at it later
> Can you try 7_45_77_0_hw_23_8_2017 and see if it works?
> 
> And I don't think the build scrips work on Mac
> You need a Linux Machine (or VM), specially Ubuntu or Debian, to compile those firmwares

thank you bro, i will try. What time can you compile my firmware :)
i am noob of that compile works :(
wich kali version (x86 or x64 bits) can you advice me?

Gorkem63, 2019-12-06:
@baselsayeh also there isnt any way for compiling bcm43455.

baselsayeh, 2019-12-06:
@Gorkem63 
Yes, there are firmware files that can be patched on this repo.

take a look at the [Patching instructions](https://github.com/seemoo-lab/nexmon/blob/master/README.md#build-patches-for-bcm4330-bcm4339-and-bcm4358-using-a-x86-computer-running-linux-eg-ubuntu-1604 )

Gorkem63, 2019-12-06:
> @Gorkem63
> Yes, there are firmware files that can be patched on this repo.
> 
> take a look at the [Patching instructions](https://github.com/seemoo-lab/nexmon/blob/master/README.md#build-patches-for-bcm4330-bcm4339-and-bcm4358-using-a-x86-computer-running-linux-eg-ubuntu-1604)

ok bro i am on kali linux 64 bit live now, and trying to compile. If i had a problem then i ca you :)

matthiasseemoo, 2019-12-06:
Try to thorowly follow the instruction and do not diverge from them.
Please, use the proposed version of Ubuntu to get started.

Gorkem63, 2019-12-06:
> ok bro i am on kali linux 64 bit live now, and trying to compile. If i had a problem then i ca you

> > @Gorkem63
> > Yes, there are firmware files that can be patched on this repo.
> > take a look at the [Patching instructions](https://github.com/seemoo-lab/nexmon/blob/master/README.md#build-patches-for-bcm4330-bcm4339-and-bcm4358-using-a-x86-computer-running-linux-eg-ubuntu-1604)
> 
> ok bro i am on kali linux 64 bit live now, and trying to compile. If i had a problem then i will call you :)

i compiles ans pops fw_bcm43455_hw.bin file in patches folder
my fw is in  system/etc/flrmware/wlan/bcmdhd
after enter make backup-firmware it say 
cp:bad /vendor/firmware/fw_bcm43455_hw.bin no such file or directory

Gorkem63, 2019-12-06:
if i put in the ..../nexmon/fw_bcm43455_hw.bin file manually to my /system/etc/flrmware/wlan/bcmdhd dlrectory, it will ok ?
@baselsayeh @matthiasseemoo

Gorkem63, 2019-12-06:
> […](#)

i put compiled bcm manually to my system/etc/flrmware/wlan/bcmdhd then install nexmon app when i tap grant root app crash. i go to tools and

Gorkem63, 2019-12-06:
> @Gorkem63
> Yes, there are firmware files that can be patched on this repo.
> 
> take a look at the [Patching instructions](https://github.com/seemoo-lab/nexmon/blob/master/README.md#build-patches-for-bcm4330-bcm4339-and-bcm4358-using-a-x86-computer-running-linux-eg-ubuntu-1604)

i put compiled bcm manually to my system/etc/flrmware/wlan/bcmdhd then install nexmon app when i tap grant root app crash. i go to tools to binary finary install it says error cant install :(

Gorkem63, 2019-12-06:
its ok

closed 2019-12-06 by Gorkem63

baselsayeh, 2019-12-06:
> if i put in the ..../nexmon/fw_bcm43455_hw.bin file manually to my /system/etc/flrmware/wlan/bcmdhd dlrectory, it will ok ?
> @baselsayeh @matthiasseemoo

Should be, and remember to backup the firmware file first

And don't use the app, it is outdated.
Compile the binaries you wanted also from this repo (specially nexutil, wireless tools, and aircrack, available from this repo)

baselsayeh, 2019-12-06:
And don't forget to use libnexmon

Gorkem63, 2019-12-06:
> And don't forget to use libnexmon

bro ok i succesfully installed patched firmware thanks a lot :)
last question :)
how can i build latest nexmon app? i installes android ndk
@baselsayeh