seemoo-lab/nexmon #354: bcm434555 no any capture

Gorkem63, 2019-12-06:
my device sony xperia Z6 Compact i compiled the 7_45_77_0_hw_23_8_2017
firmware. its ok then i installed nexmon apk and go to toll, activate all no any errors.
(nexmon play store version)
after go terminal and nexutil -m2
no issue but when i type airodump-ng wlan0 error it say no monitor mode supported! : operation not supported on transport endpoint
ioctl(SI0CSIWMODE) failed: operation not supported on transport endpoint
error setting monitor mode on wlan0
also i got nothing packets in nexmon airodump.
[image]
[image]

Gorkem63, 2019-12-06:
@baselsayeh @matthiasseemoo

baselsayeh, 2019-12-06:
You need to compile and use libnexmon

Gorkem63, 2019-12-06:
> You need to compile and use libnexmon

compile the nexmon app right?
can you say instructions :)
i ill subscribe you dudedee :)

baselsayeh, 2019-12-06:
Compile it, then use
LD_PRELOAD=libnexmon.so airodump-ng wlan0
After setting the monitor mode
nexutil -m2
And you maybe need to disable scanning by
nexutil -c1
nexutil -d

Gorkem63, 2019-12-06:
> Compile it, then use
> LD_PRELOAD=libnexmon.so airodump-ng wlan0
> After setting the monitor mode
> nexutil -m2
> And you maybe need to disable scanning by
> nexutil -c1
> nexutil -d

by saying you compile is, fw or app?
i complieded fw already bro
just wanna know to how compile app

Gorkem63, 2019-12-06:
@matthiasseemoo @baselsayeh i need to learn compile nexmon apk for my z5c bcm43455 (i already compiled fw and installed)
with nexmon i planning mdk3, airodump (i also have nethunter kali so i prefer kali aircrack #:) )

matthiasseemoo, 2019-12-06:
We did not maintain the app since a long time. Normally, the existing
version of the app should work. If not, you need to compile it using the
exact versions of the tools provided in the readme.

Gorkem63, 2019-12-06:
fixed thanks :)

closed 2019-12-06 by Gorkem63