seemoo-lab/nexmon #355: sh: libfakeioctl.so not found

Gorkem63, 2019-12-06:
`nexutil -m`
2
`airodump-ng wlan0` or `mdk3 wlan0 b`
CANNOT LINK EXECUTABLE "sh": library 'libfakeioctl.so" not found

same 5 pieces

No monitor mode supported!!!
:operation not supported on transport endpoint
Current mac Xxxxx on channel x with ssid "randomizing ssid's so cant capture ap names" "example jdhdheh++4"
airodump like this, bur capturing working

Gorkem63, 2019-12-06:
@baselsayeh @matthiasseemoo

matthiasseemoo, 2019-12-06:
If you do not define where the dynamic linker can find the library, it
won't work.

Gorkem63, 2019-12-06:
> […](#)

how can i do ?
my device arm64 so i need to compile utilites for 64 bit ?

Gorkem63, 2019-12-06:
@matthiasseemoo

baselsayeh, 2019-12-06:
@Gorkem63 
Compile libnexmon, Put libnexmon.so from the arm64-v8a folder into /system/lib,
And put the one inside the folder ("armeabi-v7a" or "armeabi" try the both) into /system/lib32,
And if you dont have /system/lib32, then put the "armeabi-v7a" one into /system/lib and the "arm64-v8a" one into /system/lib64

baselsayeh, 2019-12-06:
@Gorkem63 
Try those ones
[libs.zip](https://github.com/seemoo-lab/nexmon/files/3934270/libs.zip)

Gorkem63, 2019-12-07:
@baselsayeh @matthiasseemoo thanks bros succesfully installed :)

Gorkem63, 2019-12-07:
done.

closed 2019-12-07 by Gorkem63