seemoo-lab/nexmon #356: How can i run nexmon in nethunter chroot

Gorkem63, 2019-12-07:
compiled and succesfully run in android but when mdk deauth attack, phone freeze .is it my compile problem?

so possible to build in chroot ?
@baselsayeh :)

baselsayeh, 2019-12-07:
@Gorkem63 
Is it the whole phone, or the Wi-Fi chip only?
If you inject too much packets, the Wi-Fi chip will run out of ram, and it will crash
try lowering the number of packets in aireplay-ng
e.g:
`aireplay-ng -0 1`
Or
`aireplay-ng -0 2`

And for the LD_PRELOAD, yes you need that, as the nexmon modified firmware's injection (And monitoring) method is not standard, and do not have the drivers in linux
However, you can export the LD_PRELOAD variable, and use airodump/aireplay
```
LD_PRELOAD=libnexmon.so
export LD_PRELOAD
airodump-ng ...

```

SamuelNigga, 2019-12-07:
@baselsayeh the whole phone is my prefer
so i wanna set up mon0 mode:
`airmon-ng start wlan0
for the use nethunter kali tools.

SamuelNigga, 2019-12-07:
i got export is not an identifier while type:
`export LD_PRELOAD`

baselsayeh, 2019-12-07:
@SamuelNigga 
I really do not know anything about kali nethunter.
And if you are using nexmon firmware, there is no need for airmon-ng, you should read [README.MD](https://github.com/seemoo-lab/nexmon/blob/master/README.md) to set it up.

SamuelNigga, 2019-12-07:
[image]

[image]

in nethunter @baselsayeh

SamuelNigga, 2019-12-07:
now got this:
[image]

baselsayeh, 2019-12-07:
It tells you you have the one with wrong architecture of libnexmon.so.
Either compile or download libnexmon.so first

As i said, i don't know anything about kali, i might not be very helpful here, sorry

SamuelNigga, 2019-12-07:
> It tells you you have the one with wrong architecture of libnexmon.so.
> Either compile or download libnexmon.so first
> 
> As i said, i don't know anything about kali, i might not be very helpful here, sorry

oww ok, nethunter's architecture is aarch64 you gave me for the android main arm64 libs so thats way it dont work in nethunter.
if i compile utilities in nethunter  chroot? is it run ? :)

baselsayeh, 2019-12-07:
@SamuelNigga 
Ok, i dont know if anything will compile under kali.
[Read this](https://github.com/seemoo-lab/nexmon/issues/355#issuecomment-562761545) steps, and you might need to replace the /system paths with your chroot pathes

SamuelNigga, 2019-12-07:
thanks @baselsayeh

closed 2019-12-07 by SamuelNigga