seemoo-lab/nexmon #357: cant add mon0 interface

SamuelNigga, 2019-12-07:
`iw phy phy0 interface add mon0 type monitor`
command failed: Operation not supported on transport endpoint (-95)

SamuelNigga, 2019-12-07:
i think the iw isnt compiled my device, thats way i got errors
@baselsayeh can you compile all  nexmon utilities for my device (arm64)  ?
:)

baselsayeh, 2019-12-07:
@SamuelNigga 
Please do read the README.md, as everything is explained there.
You do not need the mon0 interface (this is not how nexmon works), You just need:
```
LD_PRELOAD=libnexmon.so
export LD_PRELOAD
nexutil -c1
nexutil -d
nexutil -m2
airodump-ng wlan0

```

But as i said, it may or may not work under kali, as i do not know anything about it.

closed 2019-12-07 by SamuelNigga