seemoo-lab/nexmon #361: nexmon on-nexus 6 Shamu havoc-os nethunter install

oakes8585, 2019-12-09:
I remember when nexmon first came about and i was unable to get it working correctly on my nexus 6 has anyone had luck with a working patch ? Or can one please be developed for use with the app as it still after all this time does not support Nexus6 Shamu. Id love to see this as a nicely added addition for nexus 6. this phone seems to be over looked more often than not by many of great devs.

matthiasseemoo, 2019-12-09:
Why dont you work on a patch?

oakes8585, 2019-12-09:
If i had the time i would 2 jobs and a daughter with cancer has slowed me down quite a bit. I guess i could do a little at a time but i posted this because i know that there are people that have the time and it wouldent interfear with there everyday life