seemoo-lab/nexmon #362: Building APK

SamuelNigga, 2019-12-11:
i want learn the Building apk instructions.
need to build nexmon apk.
@baselsayeh @matthiasseemoo

kimocoder, 2019-12-11:
Clone the project, load it in Android Studio and "Build APK" ..

SamuelNigga, 2019-12-11:
@kimocoder so dont need to enter nexmon after patch firmware then downliad ndk, set NDK_ROOT=ndk directory after make utilities ?

kimocoder, 2019-12-11:
When building utilities you need the NDK_ROOT path yes. And used the older toolchain, described in the README.md :+1:

SamuelNigga, 2019-12-11:
@kimocoder bro my device arm64, so when i will build apk (like you said(clone nexmon, install android studio))  and  install my phone it will install utilities (aircrack) for my phone's arch right? also am i will succesfully run wireshark in builded nexmon app right ?

kimocoder, 2019-12-11:
You need to install the utilities from the app itself.
But remember.. Nexmon is a bit outdated sort og, so it may need some developer "skills" in order to make it all work, depending on device, versions etc.. the easiest for you would be to use the Nexmon APK v2.1.3 found online.

SamuelNigga, 2019-12-11:
@kimocoder ok bro, cloning project and dowloading AS, after i will open and install AS then open nexmon folder and clicking build apk.Ok thanks bro :)

closed 2019-12-11 by SamuelNigga