seemoo-lab/nexmon #365: monitor mode & wireless connection can't work in parallel?

nileshg1201, 2019-12-18:
Raspberry Pi 4 Model B Rev 1.1 running kali-linux-2019-4-rpi3-nexmon-64-img
Linux 4.19.81-Re4son-v8l+
brcmfmac (Broadcom 2711)

I have a similar issue; using the official kali linux image(containing nexmon patches) for RPi 4; SSHing into it over a mobile hotspot; after creating & bringing mon0 up, i lose the SSH connection; have to restart Pi to restore CLI access, and the mon0 interface is gone; it's like we can't access the wireless AP when it's in monitor mode; both interfaces not working in parallel.

is there a way to solve this that i dont know of?

FlexMcMurphy, 2019-12-19:
I'm not an expert on this at all but I'm thinking of installing the Re4son kernel which also contains the Nexmon patches but apparently lets you use the same Wifi card on the Pi to ssh to it and also have it in monitor mode at the same time. See this [Youtube video](https://youtu.be/U3eldMLq2cc?t=630) which in fact is worth watching from the start if you don't know anything about Re4son.

Flex

kimocoder, 2019-12-19:
I'm using both at same time on my Nexus 6P :+1: 
Works like a charm, see https://twitter.com/kimocoder

nileshg1201, 2019-12-19:
I did follow that video & [this](https://null-byte.wonderhowto.com/how-to/load-kali-linux-raspberry-pi-4-for-ultimate-miniature-hacking-station-0201737/) but soon as i start monitor mode using **monstart** i lose my SSH connection, and the Pi doesnt show in my network scan. the video says it works but its a pi zero, i have a pi 4, not sure what the issue is... @FlexMcMurphy @kimocoder i was thinking of reflashing the Re4son kernel, but there's no version for Pi 4 yet

kimocoder, 2019-12-19:
I don't know much about PI's unfortunately.. :100:

nileshg1201, 2019-12-19:
okay, will find some resolution, thanks!

closed 2019-12-19 by nileshg1201