seemoo-lab/nexmon #371: Can this patch allow 32 connections to Raspberry Pi 4

tim-moody, 2019-12-26:
The RPi 4 and other recent versions have reduced maximum wifi connections from 32 to 16, though we were actually able to stream 30 simultaneous videos on an rpi 3B. Can these patches help set the maximum back to 32?

matthiasseemoo, 2019-12-26:
Is this a firmware or a driver setting? What kind of wifi connections do
you mean? To a SoftAP that you start on the Raspberry Pi?

-- 
Matthias Schulz
Secure Mobile Networking Lab - SEEMOO

Email: matthias.schulz@seemoo.tu-darmstadt.de
Web: http://www.seemoo.de/mschulz
Phone (new): +49 6151 16-25478    Fax: +49 6151 16-25471

Department of Computer Science
Center for Advanced Security Research Darmstadt
Technische Universität Darmstadt
Mornewegstr. 32 (Office 4.2.10, Building S4/14)
D-64293 Darmstadt, Germany

tim-moody, 2019-12-26:
yes, inbound connections to hostapd. I am guessing it is a firmware setting as it seems wifi device dependent.