seemoo-lab/nexmon #373: What are

zeroby0, 2020-01-05:


jlinktu, 2020-01-05:
Core means receive units here. If you have a device with four antennas, there should also be four cores. If your device has only one antenna then there is most likely only one core. The AC86U has four cores, the Nexus 6P has two cores, and the Nexus 5 and Raspberry Pi 3B+/4B have one core each. Indexing starts at core 0. With the mask you can select from what cores you want to extract CSI, of course you are limited to the cores available on your device.

You should use the Issues section of the [nexmon_csi](https://github.com/seemoo-lab/nexmon_csi) project to ask questions specific to the CSI extractor patch.

zeroby0, 2020-01-06:
@jlinktu I see. Thanks! that clarified everything.

I opened the issue here by mistake, I had both the repositories open in adjacent tabs and created it here by accident. Thank you for the help :)

closed 2020-01-06 by zeroby0